openai/mujoco-py #2: glfw Window freezes and becomes unresponsive

mjm522, 2016-06-03:
Just trying to get to know mujoco-py wrappers. But I do face the following issue.

```
from mujoco_py import mjviewer, mjcore
from mujoco_py import mjtypes
from mujoco_py import glfw

width = 1500
height = 1500
viewer = mjviewer.MjViewer(visible=True, init_width=width, init_height=height)
viewer.start()
model = mjcore.MjModel('/home/mjm/Documents/Python/Mujoco/cartpole.xml')
viewer.set_model(model)
(data, width, height) = viewer.get_image()
#viewer.finish()
#viewer = None
```

When I do this, the "Simulate" window pops up and freezes. Now the only way to stop it is by killing it.
It is unresponsive and does not display anything. Along with the python console also crashes and would have to reopen again.

Once I kill it, this is the error that I get in the terminal:

```
XIO:  fatal IO error 0 (Success) on X server ":0"

      after 212 requests (212 known processed) with 39 events remaining.
```

I am running Ubuntu 14.04 64bit on Macbook pro with Nvdia graphics card. My environment variable $DISPLAY:=0

I got a similar error in [stackoverflow ](http://stackoverflow.com/questions/19015749/simple-glfw-freezes-thinks-like-crazy-not-stopping)

Any idea why this happens? Is there a work around to it?

Any helpful suggestion is appreciated.

tlbtlbtlb, 2016-06-03:
Do you know whether the call to get_image returns? I'm not sure what's supposed to happen when viewer.finish is never called before exit.

mjm522, 2016-06-03:
I tried with viewer.finish in the first place. The problem is caused from viewer.start() itself. 
The moment, I run that, the window pops up and then hangs.

tlbtlbtlb, 2016-06-03:
I tested the same code (just changing the path to cartpole.xml) on both OSX/Macbook and Linux 14.04/PC with nVidia graphics, and neither hangs. 

It seems most likely a problem somewhere between glfw, X, and the graphics card. Can you try the example glfw code at https://pypi.python.org/pypi/glfw/1.0.2? Can that open a window?

Also, longshot, if 1500x1500 is larger than your screen, that might cause problems.

mjm522, 2016-06-03:
That did work with no issues!! I have mentioned mujoco visualizer works with no issues in C. So I guess it is not an issue with glfw.

It is not with my resolution.  My PC resolution 2880x1800.

mjm522, 2016-06-03:
```
import gym
env = gym.make('Hopper-v1')
env.reset()
for _ in range(1000):
    env.render()
    env.step(env.action_space.sample()) # take a random action
```

This starts the hopper running for some time and again the same the problem. The "Simulate" window blackens and hangs.

tlbtlbtlb, 2016-06-04:
I assume you ran that in interactive mode. Once the loop is finished, Python is waiting on stdin, and can't respond to X server events. If you run that as a standalone python script (ie, python hoppertest.py) it should close the window when the program exits. It's a limitation of Python that it can't respond to X events while waiting for console input.

mjm522, 2016-06-05:
I got it working on my OSX. I guess this is an issue with Ubuntu. I have seen similar error in other forums regarding X server. If I get some workaround, I will definitely post it here.

avisingh599, 2016-09-09:
@mjm522 Were you able to find a workaround? 

Thanks,
Avi

avisingh599, 2016-09-09:
Never mind, adding the line `viewer.loop_once()` fixed things for me.

pulkitag, 2017-01-15:
@avisingh599 Where did you add the command `viewer.loop_once()`. Can you share the code snippet?

avisingh599, 2017-01-15:
The following snippet should be able to take images
```
import mujoco_py
import numpy as np
from scipy.misc import imsave
import time

fullpath = './example.xml'
model = mujoco_py.MjModel(fullpath)
width = 1000
height = 1000
viewer = mujoco_py.MjViewer(visible=True, init_width=width, init_height=height)
viewer.start()
viewer.set_model(model)

steps = 100
skip = 10

start = time.time()
for i in xrange(steps):
    viewer.render()
    viewer.loop_once()
    data, width, height = viewer.get_image()
    img = np.fromstring(data, dtype='uint8').reshape(height, width, 3)[::-1,:,:]
    imsave('imgs/out_' + str(i) + '.png', img)
    for j in xrange(skip):
        model.step()
end = time.time()
print(end - start)

viewer.finish()
viewer = None
```