openai/mujoco-py #4: Mouse Controls results in error.

mjm522, 2016-06-06:
Mouse actions on the open window throws error and result in hanging. For example when the rendered window is tried to be scrolled to adjust zoom, following error comes up.

Traceback (most recent call last):
  File "_ctypes/callbacks.c", line 314, in 'calling callback function'
  File "/Library/Python/2.7/site-packages/mujoco_py/mjviewer.py", line 294, in handle_scroll
    mjlib.mjv_moveCamera(MOUSE_ZOOM, 0, (-20*y_offset), byref(self.cam), width, height)
NameError: global name 'MOUSE_ZOOM' is not defined

Same is the case with left and right buttons. Is this a known issue?

Is there a work around?

I am running mujoco_py on mac osx El Captain

tlbtlbtlb, 2016-06-06:
Fixed in master branch, thanks for reporting.

closed 2016-06-06 by tlbtlbtlb

mjm522, 2016-06-06:
I reinstalled the the complete package, still I do have the same problem.

gdb, 2016-06-06:
Sorry, we did not push an updated package to Pypi. Just pushed 0.4.4 with this fix!

mjm522, 2016-06-06:
Okay, but I did clone and make and this is the screen:

writing manifest file 'MANIFEST'
```
creating mujoco-py-0.4.4
creating mujoco-py-0.4.4/mujoco_py
making hard links in mujoco-py-0.4.4...
hard linking setup.py -> mujoco-py-0.4.4
hard linking mujoco_py/__init__.py -> mujoco-py-0.4.4/mujoco_py
hard linking mujoco_py/config.py -> mujoco-py-0.4.4/mujoco_py
hard linking mujoco_py/error.py -> mujoco-py-0.4.4/mujoco_py
hard linking mujoco_py/glfw.py -> mujoco-py-0.4.4/mujoco_py
hard linking mujoco_py/mjconstants.py -> mujoco-py-0.4.4/mujoco_py
hard linking mujoco_py/mjcore.py -> mujoco-py-0.4.4/mujoco_py
hard linking mujoco_py/mjextra.py -> mujoco-py-0.4.4/mujoco_py
hard linking mujoco_py/mjlib.py -> mujoco-py-0.4.4/mujoco_py
hard linking mujoco_py/mjtypes.py -> mujoco-py-0.4.4/mujoco_py
hard linking mujoco_py/mjviewer.py -> mujoco-py-0.4.4/mujoco_py
hard linking mujoco_py/platname_targdir.py -> mujoco-py-0.4.4/mujoco_py
hard linking mujoco_py/util.py -> mujoco-py-0.4.4/mujoco_py
Creating tar archive
removing 'mujoco-py-0.4.4' (and everything under it)
```
Still I have the error.