openai/mujoco-py #10: how to change the angle of camera

jjkke88, 2016-06-23:
In initial state, the picture looks like below.
[image]
I want to change the init state of camera so that the scene looks like this.
[image]
What can I do?

jhollowayj, 2016-08-06:
http://www.mujoco.org/book/haptix.html#uiCamera has a small write up of the camera controls.

in https://github.com/openai/gym/blob/master/gym/envs/mujoco/humanoidstandup.py they have the commands:
    self.viewer.cam.trackbodyid = 1
    self.viewer.cam.distance = self.model.stat.extent * 1.0
    self.viewer.cam.lookat[2] += .8
    self.viewer.cam.elevation = -20
where self.viewer is an instance of mujoco_py.MjViewer.  

So going off of the documentation, you'll probably want to change cam.lookat to help it point straight down.  It also looks like you'll have to move the camera itself.  That (I'm guessing) might be defined in the model of your simulator, so check there for it's position

yishuangl, 2017-09-15:
I think what you really should change is self.viewer.cam.elevation

nemanja-rakicevic, 2017-10-09:
You can find some descriptions about the "abstract camera pose specification" [here](https://github.com/openai/mujoco-py/blob/c5f60322467ec8ecc0db64c3e18a4da762c27e45/mujoco_py/pxd/mjvisualize.pxd)
To make the screen like in the second image, you need to edit the `def viewer_setup(self)` function in the `reacher.py` file:
`
    def viewer_setup(self):
        self.viewer.cam.trackbodyid = 0         # 0: camera fixed, 1: camera tracking the moving object
        self.viewer.cam.distance = self.model.stat.extent * 1.0         # how much you "zoom in", model.stat.extent is the max limits of the arena
        self.viewer.cam.lookat[0] += 0.5         # x,y,z offset from the object
        self.viewer.cam.lookat[1] += 0.5
        self.viewer.cam.lookat[2] += 0.5
        self.viewer.cam.elevation = 90           # camera rotation around the axis in the plane going through the frame origin (if 0 you just see a line)
        self.viewer.cam.azimuth = 0               # camera rotation around the camera's vertical axis
`

pimdh, 2018-03-19:
You can also use the following to use one of the cameras defined in the model as default (here I assume you want to use the first defined camera):
```
from mujoco_py.generated import const
viewer.cam.type = const.CAMERA_FIXED
viewer.cam.fixedcamid = 0
```

richardrl, 2019-01-16:
How do these azimuth etc haptix camera configurations relate to the camera parameters here? http://mujoco.org/book/XMLreference.html#camera

neighthan, 2020-01-14:
In case this helps somebody else: make sure that you're not trying to change the angle on a fixed camera (i.e. where `camera.type == mujoco_py.generated.const.CAMERA_FIXED`); that doesn't seem possible, at least when I was trying. You want `camera.type == mujoco_py.generated.const.CAMERA_FREE`; make a new camera if you need it.