openai/mujoco-py #16: Adding wrapper for mj_applyFT

mjm522, 2016-09-26:
Hi,

This is a C++ helper function for mujoco: It applies cartesian force and torque (outside xfrc_applied mechanism)
```
MJAPI void mj_applyFT(const mjModel* m, mjData* d, 
                      const mjtNum* force, const mjtNum* torque, 
                      const mjtNum* point, int body, mjtNum* qfrc_target);

```
I added the wrapper in mjcore.py with the following statement.
```
mjlib.mj_applyFT(self.ptr, self.data.ptr,\
                         force.ctypes.data_as(POINTER(c_double)),\
                         torque.ctypes.data_as(POINTER(c_double)),\
                         point.ctypes.data_as(POINTER(c_double)),\
                         body_adr,\
                         self.data.qfrc_applied.ctypes.data_as(POINTER(c_double)));
```
**force** is a 3 x 1 ndarray, **torque** is a 3 x1 ndarray, **point** is a 3 x 1 ndarray and **body_adr** is the body index **qfrc_target** is the array to which we have to copy.

But it doesn't copy the values to self.data.qfrc_applied, but if I try to directly copy it, it works (i.e. `self.data.qfrc_applied = np.hstack([force, torque])).` 

I called mj_forward() to see if it gets updated, but it doesn't. 

Could you please suggest on what I am doing wrong?

closed 2016-09-26 by mjm522

arpit15, 2017-01-05:
@mjm522 were you able to make your wrapper work? Let me know what were the changes made in case it works.
Thank you

mjm522, 2017-01-05:
yeah mine works, you can fork my repo and get it.. i have even given an example in the folder :)

tlbtlbtlb, 2017-01-06:
@mjm522 If you wouldn't mind making a pull request we'd be glad to incorporate it.

mjm522, 2017-01-06:
created a PR, please do check and suggest edits if required.