openai/mujoco-py #19: How to copy value to mjcore.model.body_pos?

mjm522, 2016-10-12:
When I try to copy a value to this variable like this, 
`model.data_pos = newValue`
it says, 
`ValueError: assignment destination is read-only`. 
So I tried to using  the inbuilt function `mjlib.mju_copy(data_d.ctypes.data_as(POINTER(c_double)), data_s.ctypes.data_as(POINTER(c_double)), c_int(length))` in the following way.
`mjlib.mju_copy(model.data_pos.ctypes.data_as(POINTER(c_double)), newValue.ctypes.data_as(POINTER(c_double)), c_int(3))`
This statement does not produce any error, nevertheless the value remain unchanged. 
Could anyone please tell me how to copy a new value to it?

tlbtlbtlb, 2016-10-12:
Look for the calls to `memmove` in mjtypes.py, (most of the `mjdata` properties have setters that call it) and do something more like that.

aravind0706, 2017-11-18:
Using memmove doesn't work for `model.body_pos` or `model.geom_rgba`. It replaces the existing values with random values -1.08420217e-19, 1.82499993e+00, etc instead of the value actually stored in `val_ptr`. 

Is it because these properties were initialized with `write=False`?

tlbtlbtlb, 2017-11-19:
That is likely a mismatch between C float and double types. Make sure
you’re passing in the right numpy type, which is float64.