openai/mujoco-py #20: Crashing when calling start() method on mujoco_py.MjViewer object

febert, 2016-10-18:
Hi there,
my program is crashing when calling the start() method on the MjViewer object. 
```
self._viewer_main = mujoco_py.MjViewer(visible=True, init_width=AGENT_MUJOCO['image_width'],
            init_height=AGENT_MUJOCO['image_height'])

self._viewer_main.start()
```
The context for this code is [here](https://github.com/febert/lsdc/blob/master/python/lsdc/agent/mjc/agent_mjc.py#L104)

It only crashes when running it on a desktop machine with the nvidia 1080, but runs without erros on my laptop with a much older graphics card.

When tracking the error down in 'self._viewer_main.start()' it appears to be crashing:
[just here](https://github.com/openai/mujoco-py/blob/master/mujoco_py/mjviewer.py#L190
)

Unfortunately I'm not getting a stack trace. It just says: Process finished with exit code 139 (interrupted by signal 11: SIGSEGV)

tlbtlbtlb, 2016-10-18:
First, double-check that init_width and init_height are sensible. If they're anything other than Python integers between 100 and 1000 and divisible by 4, try with moderate values.

It wouldn't surprise me if there's a bug in Mujoco's OpenGL renderer around having multiple OpenGL windows render the same scene. In particular, it creates vertex buffers in the GPU memory, and sharing these correctly is hard to get right (and that's something that would differ between graphics cars). Since the crash is happening in C code, try running it under a C debugger, like (in Linux)
```sh
$ gdb python
(gdb) run agent_mjc.py
```
or in OSX:
```sh
$ lldb python agent_mjc.py
```

febert, 2016-10-18:
the values for init_width and init_height are 640 and 480.

The program already crashes when starting the first viewer, so it seems like multiple viewers at once cannot be the reason?

From gdb I get the following information:

```
INFO:initializing glfw@(3, 1, 2)

Program received signal SIGSEGV, Segmentation fault.
__strcmp_sse2_unaligned ()
    at ../sysdeps/x86_64/multiarch/strcmp-sse2-unaligned.S:29
29	../sysdeps/x86_64/multiarch/strcmp-sse2-unaligned.S: No such file or directory.
```

tlbtlbtlb, 2016-10-18:
Could you give a complete stack trace, with `(gdb) bt`?

febert, 2016-10-18:
Here it is:
```
#0  __strcmp_sse2_unaligned ()
    at ../sysdeps/x86_64/multiarch/strcmp-sse2-unaligned.S:29
#1  0x00007fffa7dc5ae9 in chooseFBConfig ()
   from /home/frederik/Documents/lsdc/src/3rdparty/mjpro131/bin/libglfw.so.3
#2  0x00007fffa7dc6516 in _glfwCreateContext ()
   from /home/frederik/Documents/lsdc/src/3rdparty/mjpro131/bin/libglfw.so.3
#3  0x00007fffa7dc3529 in _glfwPlatformCreateWindow ()
   from /home/frederik/Documents/lsdc/src/3rdparty/mjpro131/bin/libglfw.so.3
#4  0x00007fffa7dbb7f6 in glfwCreateWindow ()
   from /home/frederik/Documents/lsdc/src/3rdparty/mjpro131/bin/libglfw.so.3
#5  0x00007fffd71b3adc in ffi_call_unix64 ()
   from /usr/lib/x86_64-linux-gnu/libffi.so.6
#6  0x00007fffd71b340c in ffi_call ()
   from /usr/lib/x86_64-linux-gnu/libffi.so.6
#7  0x00007fffd73ca5fe in _ctypes_callproc ()
   from /home/frederik/Documents/lsdc/lsdcenv/lib/python2.7/lib-dynload/_ctypes.x86_64-linux-gnu.so
#8  0x00007fffd73cbf9e in ?? ()
   from /home/frederik/Documents/lsdc/lsdcenv/lib/python2.7/lib-dynload/_ctypes.x86_64-linux-gnu.so
#9  0x0000000000499be5 in PyEval_EvalFrameEx ()
#10 0x00000000004a090c in PyEval_EvalCodeEx ()
#11 0x000000000049ab45 in PyEval_EvalFrameEx ()
```

xiaoguai0992, 2017-06-15:
I found another problem about MjViewer.start(), I don't know if there is anything to do with your question.

```python
import mujoco_py as mj
model = mj.MjModel("humanoid.xml")
v = mj.MjViewer()
v.set_model(model)
v.start()
```

Then error occurred:
```
Traceback (most recent call last): 
  File "<stdin>", line 1, in <module>
  File "C:\Anaconda3\lib\site-packages\mujoco_py\mjviewer.py", line 180, in start _, _, refresh_rate = glfw.get_video_mode(glfw.get_primary_monitor()) 
  File "C:\Anaconda3\lib\site-packages\mujoco_py\glfw.py", line 719, in get_video_mode videomode = _glfw.glfwGetVideoMode(monitor).contents 
OSError: exception: access violation reading 0x0000000000000078
```

Do you have the same problem? I'm using win10 with python3.6.1

keven425, 2017-10-17:
I'm running into a similar issue. Here's my gdb trace:

I'm on cuda V8.0.61.

```
Program received signal SIGSEGV, Segmentation fault.
__strcmp_sse2_unaligned () at ../sysdeps/x86_64/multiarch/strcmp-sse2-unaligned.S:31
31	../sysdeps/x86_64/multiarch/strcmp-sse2-unaligned.S: No such file or directory.
(gdb) bt
#0  __strcmp_sse2_unaligned () at ../sysdeps/x86_64/multiarch/strcmp-sse2-unaligned.S:31
#1  0x0000000000415594 in chooseGLXFBConfig ()
#2  0x0000000000416e4f in _glfwChooseVisualGLX ()
#3  0x000000000041211d in _glfwPlatformCreateWindow ()
#4  0x00000000004091cc in glfwCreateWindow ()
#5  0x0000000000403e92 in main ()
```