openai/mujoco-py #26: Touch Sensors

febert, 2017-01-18:
Are touch sensors supported by mujoco-py? I am trying to create one using this xml-file:

```
<worldbody>
        <camera name="maincam" mode= "fixed" fovy="75" euler="0 0 0" pos="0 0 .75"/>
        <body name="particle" pos="0 0 0.03">
            <inertial pos="0 0 0" mass="5" diaginertia="166.667 166.667 166.667"/>
            <geom name="particle_geom" type="capsule" fromto="-0.01 0 0 0.01 0 0" size="0.05" rgba="0.9 0 0.1 0.6" />
            <joint name="ball_x" type="slide" pos="0 0 0" axis="1 0 0" />
            <joint name="ball_y" type="slide" pos="0 0 0" axis="0 1 0" />

            <sensor>
                <site name="sensorsurf" pos="0 0 0.035" size="0.05" type="ellipsoid" rgba="0.3 0.2 0.1 0.3"/>
                <touch name="touchsensor" site="sensorsurf" />
            </sensor>

        </body>

</worldbody>
```

However I'm getting the following error:
```
/site-packages/mujoco_py/mjcore.py", line 37, in __init__
    raise MjError(buf.value)
mujoco_py.mjcore.MjError: XML Error: Schema violation: unrecognized element

Element 'sensor', line 19, column 13
```
Do you have an example xml-file of how to define touch sensors?

If touch sensors are not yet supported what would need to be done to integrate them?

febert, 2017-01-18:
It is working now, the bindings indeed support touch sensors. I had to modify the xml to have the sensor defined *outside* the body.

Here is xml which works:
```
<worldbody>
    <camera name="maincam" mode= "fixed" fovy="75" euler="0 0 0" pos="0 0 .75"/>
    <body name="particle" pos="0 0 0.03">
        <inertial pos="0 0 0" mass="5" diaginertia="166.667 166.667 166.667"/>
        <geom name="particle_geom" type="capsule" fromto="-0.01 0 0 0.01 0 0" size="0.05" rgba="0.9 0 0.1 0.6" />
        <joint name="ball_x" type="slide" pos="0 0 0" axis="1 0 0" />
        <joint name="ball_y" type="slide" pos="0 0 0" axis="0 1 0" />
        <site name="sensorsurf" pos="0 0.045 0" size=".03 .03 .03" type="ellipsoid" rgba="0.3 0.2 0.1 0.3"/>

    </body>

</worldbody>

 <sensor>
     <touch name="touchsensor" site="sensorsurf" />
 </sensor>
```

closed 2017-01-28 by febert

MahanFathi, 2017-12-06:
Any idea how to access sensory data in mujoco_py/gym?

Michielskilian, 2018-12-11:
Find the file with the simulation data, for example `gym/gym/envs/mujoco/humanoid.py`  
There you can find the sensor data with (in this issue case): 
```python
self.sim.data.get_sensor('touchsensor') # Gives one value
self.sim.data.sensordata # Gives array of all sensorvalues
```

kargarisaac, 2019-07-15:
> It is working now, the bindings indeed support touch sensors. I had to modify the xml to have the sensor defined _outside_ the body.
>
> Here is xml which works:
>
> ```
> <worldbody>
>     <camera [...]

Hey,
I want to add torque sensors in joints of my robotic arm but when I add it and read data, it changes in 3 axis. I think it should change only in one axis (z on frame in joint). Am I wrong? 

I use  '<site name="lumi_torque3" zaxis="0 0 1"/>' in body and  '<torque name="lumi_joint3_torque" site="lumi_torque3" />' in sensor tag.

TigerStone93, 2021-08-09:
@Michielskilian Thank you for your answer. It's really helpful.