openai/mujoco-py #33: Change Mujoco Model Parameters online

MankowitzProjects, 2017-03-10:
Hi there,

I am not sure if this is the correct forum for this question but I would like to ask whether it is possible, using your python wrappers, to change the Mujoco model parameters dynamically between time steps. For example, in inverted pendulum, can I change the mass or length of the pendulum at each time step?

Thanks!

tlbtlbtlb, 2017-03-10:
There's no high-level API through Mujoco-py. It might be possible using Mujoco's C API. See http://mujoco.org/book/programming.html.

MankowitzProjects, 2017-03-10:
Yeah it is definitely possible using Mujoco's C API. The only issue is that I will need to recompute parameters that are saved in mjModel on the fly, by calling mj_setConst on the C side. Do you see a way for me to modify the python code to perform this call on the fly from the python side?

tlbtlbtlb, 2017-03-10:
Uncomment the definition of mj_setConst here: https://github.com/openai/mujoco-py/blob/master/mujoco_py/mjlib.py#L77 and you should be able to call it.

MankowitzProjects, 2017-03-11:
Great thanks Trevor!

Kiwoo, 2017-03-19:
Hi,

Could you explain how you solved this problem?
I am not sure how to use Mujoco's C Api with openai Gym Wrapper.

Thank you.

andrewliao11, 2017-04-11:
hi @MankowitzProjects, 
I wonder how you change the mass of certain object in mujoco. 
I just umcomment [this line](https://github.com/openai/mujoco-py/blob/master/mujoco_py/mjlib.py#L77), and then I'm at lost :(

andrewliao11, 2017-04-14:
Thanks I solved it

MankowitzProjects, 2017-04-14:
Hi Andrew,

I hadn't gotten around to implementing the model parameter changes as yet.
I am glad you solved it!

Cheers,
Dan

andrewliao11, 2017-04-25:
hi @tlbtlbtlb 
I'm still new to mujoco. Can you give us an example use of mjlib.mj_setConst?? 
It seems that the argument should be [POINTER(MJMODEL), POINTER(MJDATA), c_int]
Whereas,  I still don't know how to use it

jtoyama4, 2017-07-03:
Hi,
I am also struggling to change the model parameters such as arm's length at each episode.
Could anyone give me an example how to do it?
I am not sure what to do with mjlib.mj_setConst.
Thanks,

jtoyama4, 2017-07-03:
I just found same question in mujoco forum and it answers more specifically.

http://www.mujoco.org/forum/index.php?threads/can-the-parameters-of-a-model-be-changed-on-the-fly.3354/

Joll123, 2019-03-12:
hi @MankowitzProjects,
how to use mj_setConst on  the python side.
thanks