openai/mujoco-py #35: Run Ant-v1

andrewliao11, 2017-04-14:
Hi all, I run a3c in Ant-v1 and got this warning:
```
WARNING: Nan, Inf or huge value in QACC at DOF 0. The simulation is unstable. Time = 0.0000.
```

Are there anyone know what's the problem with this, and maybe the reason why.
(My code runs well for other envs

Erichliu00, 2017-09-03:
I met the same issue on several environments, does anyone know a solution?

andrewliao11, 2017-10-19:
my mujoco-py version is mujoco-py==0.5.7
I'm not sure if it appears in other version

kkjh0723, 2017-11-07:
@andrewliao11 I also tried Ant-v1 with your pytorch a3c code. I changed line 81 in test.py  

    state, reward, done, _ = env.step(action.numpy())

and then, the warning was disappeared

HenryZhou7, 2018-03-22:
Any updates on this issue? I am having the same trouble.

kirk86, 2018-07-26:
Same issue here with different environments. If anyone has found as solution plz kindly share!

cyrilzakka, 2018-12-28:
Seems to be an issue with MuJoCo - too much penetration or contact between two bodies will cause this error. Model needs to updated

muratcancicek, 2019-01-22:
I had a similar problem in  #340. I discovered that you should slow down stepping. I solved the issue by just delaying the next step with `time.sleep(.002)`.

`time.sleep(.002)`
`obs, r, done, _ = self.env.step(action)`

tarod13, 2019-04-14:
I have the same problem in the Hopper-v2 and Swimmer-v2 environments. I tried delaying the steps with time.sleep(), but that method didn't help.

quantumiracle, 2021-01-15:
Similar problem when using this for  a robotic control task.

HankerSia, 2021-03-05:
Similar problem when using this for grasp task with a customized mesh file. Any new solution?
I had tried the sleep solution, it seems no help...

HYDesmondLiu, 2021-09-08:
Sleep solution is not helping for me, either.

MoritzLange, 2022-08-25:
I recommend taking a look at the action you're passing to `step()`. I'm debugging some quite complex code and at some point this error occured when the action just consisted of `nan`s. So at least in my case the problem is exactly what the error message states.