openai/mujoco-py #36: OSx

gautam1858, 2017-05-15:
Only Mujoco 1.5 works in OSx so could you guys release support for 1.5 instead of 1.31 version of Mujoco

tlbtlbtlb, 2017-05-15:
I use Mujoco 1.3.1 frequently under OSX. Can you be more specific about how it doesn't work?

gautam1858, 2017-05-15:
Ok checkout this screenshot of the chat with Emo, 1.31 doesn't work with OSx so asking for help 

[image]

tlbtlbtlb, 2017-05-15:
Ah. Perhaps nobody at OpenAI has an NVMe disk.

Supporting 1.5 is likely to be a big project, since it will change the semantics of all the mujoco envs and make new results incompatible with previous results.

I wonder if there's a workaround by linking `$HOME/.mujoco` to a temporary disk or something. My superficial examination of `dtruss python -c 'import gym; env=gym.make("Hopper-v1")' doesn't show any obviously non-portable things it does in order to read the key file.

gautam1858, 2017-05-16:
OK will try

gautam1858, 2017-05-16:
Unable to workaround this and I am unable to run the Roboschool also

JuliusKunze, 2017-05-30:
Using OSX 10.12.4, I got "ERROR: Could not open disk".

[This source](http://mujoco.org/forum/index.php?threads/trying-to-run-the-mjpro131-can-not-open-disk.3438/) states: "The license manager in MuJoCo Pro 1.40 and earlier does not work with **newer** Macs that have NVMe disks. This was fixed in MuJoCo 1.50."

gautam1858, 2017-05-30:
yes thank you I was able to solve it by updating the models

JuliusKunze, 2017-05-30:
What do you mean by "updating the models"?

gautam1858, 2017-05-30:
The issue was I had trouble running the OpenAI models in mujoco for which Emo replied saying I have to update mujoco models to 1.5 version , the current OpenAI models only support Mujoco 1.3 version

transedward, 2017-06-18:
I also had a hard time using mujoco-py because my laptop is MacBook Pro 2016. Hope to see someone coming up a fix.

abhigenie92, 2017-06-24:
So there is no workaround except waiting for openai to support 1.5?

gautam1858, 2017-06-29:
Aha they have now released support for 1.5 version but some problems in installing the same

jonasschneider, 2017-06-29:
This is resolved with 1.50.1.0 -- Python 3.6 support is tracked in #52.

closed 2017-06-29 by jonasschneider

mepster, 2017-10-21:
@tlbtlbtlb Trevor, your suggestion was worth a try... I tried copying both ~/.mujoco and the working dir to a non-NVMe USB drive... unfortunately mujoco gives the same "Could not open disk" error.

A little bit cleaner way to do the dtruss is to cd into mjpro131/bin and do

```
sudo dtruss ./simulate
```

I don't know how it figures out there is an NVME drive. Looks like it is accessing paths like /System and /etc which are hard to copy over. I think I'm not up for setting up an entire USB boot disk for my mac this evening... :-) And after all that, maybe mujoco wouldn't like a USB drive either.

I have just tried asking Mujoco to fix the NVMe disk problem in a patch from 131 to 132... then one could just link to that and OpenAI wouldn't have to change anything. http://www.mujoco.org/forum/index.php?threads/error-could-not-open-disk.3441/ 

P.S. Here is the dtruss output... by the time it opens MUJOCO_LOG.TXT below, I think the error has already occurred, because it also writes "ERROR: Could not open disk" to that file.

```
$ sudo dtruss ./simulate
dtrace: system integrity protection is on, some features will not be available

SYSCALL(args) 		 = return
MuJoCo Pro library version 1.31

ERROR: Could not open disk

Press Enter to exit ...open("/dev/dtracehelper\0", 0x2, 0x7FFF5F61E6B0)		 = 3 0
ioctl(0x3, 0x80086804, 0x7FFF5F61E638)		 = 0 0
close(0x3)		 = 0 0
thread_selfid(0x3, 0x80086804, 0x7FFF5F61E638)		 = 894851 0
bsdthread_register(0x7FFFA8AC8080, 0x7FFFA8AC8070, 0x2000)		 = 1073741919 0
mprotect(0x1040CD000, 0x88, 0x1)		 = 0 0
mprotect(0x1042AA000, 0x1000, 0x0)		 = 0 0
mprotect(0x1042C0000, 0x1000, 0x0)		 = 0 0
mprotect(0x1042C1000, 0x1000, 0x0)		 = 0 0
mprotect(0x1042D7000, 0x1000, 0x0)		 = 0 0
mprotect(0x1040CF000, 0x1000, 0x1)		 = 0 0
mprotect(0x1040CD000, 0x88, 0x3)		 = 0 0
mprotect(0x1040CD000, 0x88, 0x1)		 = 0 0
ulock_wake(0x1, 0x7FFF5F61C54C, 0x0)		 = -1 Err#2
issetugid(0x1, 0x7FFF5F61C54C, 0x0)		 = 0 0
getpid(0x1, 0x7FFF5F61C54C, 0x0)		 = 88439 0
stat64("/AppleInternal/XBS/.isChrooted\0", 0x7FFF5F61C408, 0x0)		 = -1 Err#2
stat64("/AppleInternal\0", 0x7FFF5F61C4A0, 0x0)		 = -1 Err#2
csops(0x15977, 0x7, 0x7FFF5F61BF30)		 = -1 Err#22
dtrace: error on enabled probe ID 2158 (ID 561: syscall::sysctl:return): invalid kernel access in action #10 at DIF offset 40
ulock_wake(0x1, 0x7FFF5F61C4B0, 0x0)		 = -1 Err#2
csops(0x15977, 0x7, 0x7FFF5F61B810)		 = -1 Err#22
getuid(0x15977, 0x7, 0x7FFF5F61B810)		 = 0 0
getuid(0x15977, 0x7, 0x7FFF5F61B810)		 = 0 0
stat64("/System/Library/PrivateFrameworks/Heimdal.framework/Heimdal\0", 0x7FFF5F61B918, 0x7FFF5F61B810)		 = 0 0
stat64("/System/Library/Frameworks/GSS.framework/GSS\0", 0x7FFF5F61B918, 0x7FFF5F61B810)		 = 0 0
mprotect(0x104288000, 0x88, 0x1)		 = 0 0
mprotect(0x1042D8000, 0x1000, 0x0)		 = 0 0
mprotect(0x1042EE000, 0x1000, 0x0)		 = 0 0
mprotect(0x1042EF000, 0x1000, 0x0)		 = 0 0
mprotect(0x104305000, 0x1000, 0x0)		 = 0 0
mprotect(0x1040CF000, 0x1000, 0x3)		 = 0 0
mprotect(0x1040CF000, 0x1000, 0x1)		 = 0 0
mprotect(0x104288000, 0x88, 0x3)		 = 0 0
mprotect(0x104288000, 0x88, 0x1)		 = 0 0
access("/System/Library/taggedstringsoff\0", 0x4, 0x1)		 = -1 Err#2
getrlimit(0x1008, 0x7FFF5F61F5B8, 0x1)		 = 0 0
fstat64(0x1, 0x7FFF5F61F5D8, 0x1)		 = 0 0
ioctl(0x1, 0x4004667A, 0x7FFF5F61F61C)		 = 0 0
dtrace: error on enabled probe ID 2133 (ID 951: syscall::write_nocancel:return): invalid kernel access in action #12 at DIF offset 92
dtrace: error on enabled probe ID 2133 (ID 951: syscall::write_nocancel:return): invalid kernel access in action #12 at DIF offset 92
access("/etc/localtime\0", 0x4, 0x1)		 = 0 0
open_nocancel("/etc/localtime\0", 0x0, 0x0)		 = 3 0
fstat64(0x3, 0x7FFF5F61C580, 0x0)		 = 0 0
dtrace: error on enabled probe ID 2135 (ID 949: syscall::read_nocancel:return): invalid kernel access in action #12 at DIF offset 92
close_nocancel(0x3)		 = 0 0
shm_open(0x7FFFA8ABBDE7, 0x0, 0x0)		 = 3 0
mmap(0x0, 0x1000, 0x1, 0x1, 0x3, 0x0)		 = 0x10428A000 0
close_nocancel(0x3)		 = 0 0
open_nocancel("mjkey.txt\0", 0x0, 0x1B6)		 = 3 0
fstat64(0x3, 0x7FFF5F61F3A8, 0x1B6)		 = 0 0
dtrace: error on enabled probe ID 2135 (ID 949: syscall::read_nocancel:return): invalid kernel access in action #12 at DIF offset 92
dtrace: error on enabled probe ID 2135 (ID 949: syscall::read_nocancel:return): invalid kernel access in action #12 at DIF offset 92
close_nocancel(0x3)		 = 0 0
open_nocancel("MUJOCO_LOG.TXT\0", 0x20A, 0x1B6)		 = 3 0
lseek(0x3, 0x0, 0x2)		 = 767 0
fstat64(0x3, 0x7FFF5F61F418, 0x2)		 = 0 0
dtrace: error on enabled probe ID 2133 (ID 951: syscall::write_nocancel:return): invalid kernel access in action #12 at DIF offset 92
close_nocancel(0x3)		 = 0 0
dtrace: error on enabled probe ID 2133 (ID 951: syscall::write_nocancel:return): invalid kernel access in action #12 at DIF offset 92
dtrace: error on enabled probe ID 2133 (ID 951: syscall::write_nocancel:return): invalid kernel access in action #12 at DIF offset 92
fstat64(0x0, 0x7FFF5F61F618, 0x1)		 = 0 0
ioctl(0x0, 0x4004667A, 0x7FFF5F61F65C)		 = 0 0
dtrace: error on enabled probe ID 2133 (ID 951: syscall::write_nocancel:return): invalid kernel access in action #12 at DIF offset 92

```