openai/mujoco-py #40: Multiple instances with RGB fails with X server

Kaixhin, 2017-06-20:
I'm trying to run several instances of a custom env which has an RGB state via `MjViewer`, but I get errors such as the following:

```sh
XIO:  fatal IO error 11 (Resource temporarily unavailable) on X server ":0"
      after 216 requests (216 known processed) with 50 events remaining.
XIO:  fatal IO error 11 (Resource temporarily unavailable) on X server ":0"
      after 216 requests (216 known processed) with 50 events remaining.
XIO:  fatal IO error 11 (Resource temporarily unavailable) on X server ":0"
      after 218 requests (218 known processed) with 50 events remaining.
```
It works fine with one instance, but it crashes with more running in parallel. I've also tried initialising `MjViewer` with `visible=True/False` and `go_fast=True/False`, but this doesn't make a difference. For both one instance and many I start the envs through Python's `multiprocessing.Process`. At one point I saw the following error, which seems more informative: 

```sh
[xcb] Unknown sequence number while processing queue
[xcb] Most likely this is a multi-threaded client and XInitThreads has not been called
[xcb] Aborting, sorry about that.
python: ../../src/xcb_io.c:274: poll_for_event: Assertion `!xcb_xlib_threads_sequence_lost' failed.
XIO:  fatal IO error 0 (Success) on X server ":0"
      after 225 requests (225 known processed) with 0 events remaining.
```

I am running Ubuntu 14.04 with Python 3.6 and MuJoCo 1.31.

Kaixhin, 2017-06-21:
Fixed by @apaszke who recommended changing the start method.

closed 2017-06-21 by Kaixhin

ryanjulian, 2018-03-04:
I'm hitting this bug on 0.5.7. @Kaixhin can you point me specifically @apaszke's fix so I can understand the root cause? Is there an associated issue or commit where this was resolved?

Kaixhin, 2018-03-04:
Sorry that comment was badly phrased. The suggestion was to change multiprocessing's start method to `spawn`: ` mp.set_start_method('spawn')`. `fork` is the default start method on Unix.