openai/mujoco-py #43: Latest version: "fatal error: mjmodel.h: No such file or directory"

vitchyr, 2017-06-26:
If I run `pip install mujoco-py` on a new Python 3 anaconda environment, I get the following error.

```
gcc -pthread -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -I/tmp/pip-build-q027_l94/mujoco-py/mujoco_py -I/
home/vitchyr/.mujoco/mjpro150/include -I/home/vitchyr/.local/lib/python3.5/site-packages/numpy/core/include -I/home/v
itchyr/anaconda2/envs/rllab-vitchyr/include/python3.5m -c /tmp/pip-build-q027_l94/mujoco-py/mujoco_py/cymj.c -o /tmp/
pip-build-q027_l94/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build
-q027_l94/mujoco-py/mujoco_py/cymj.o -fopenmp -w
    /tmp/pip-build-q027_l94/mujoco-py/mujoco_py/cymj.c:471:21: fatal error: mjmodel.h: No such file or directory
    compilation terminated.
    error: command 'gcc' failed with exit status 1
```

However, I'm able to install older versions, i.e. `pip install mujoco-py==0.5.7` works. Is there a new dependency

jonasschneider, 2017-06-28:
I believe this is fixed with 1.50.1.0. Can you confirm?

ashwinreddy, 2017-06-28:
I ran into the same issue, but I am able to install on top of the default Python 3.6.1.
With Anaconda, even if I use the version specified, I still get the same error:
```
~ python3 -m pip install -U mujoco-py=='1.50.1.0' --no-cache-dir                                                                            1 ↵
Collecting mujoco-py==1.50.1.0
  Downloading mujoco-py-1.50.1.0.tar.gz (2.0MB)
    100% |████████████████████████████████| 2.0MB 5.3MB/s 
Requirement already up-to-date: glfw>=1.4.0 in ./anaconda3/lib/python3.6/site-packages (from mujoco-py==1.50.1.0)
Requirement already up-to-date: numpy>=1.11 in ./anaconda3/lib/python3.6/site-packages (from mujoco-py==1.50.1.0)
Requirement already up-to-date: Cython>=0.25.2 in ./anaconda3/lib/python3.6/site-packages (from mujoco-py==1.50.1.0)
Requirement already up-to-date: imageio>=2.1.2 in ./anaconda3/lib/python3.6/site-packages (from mujoco-py==1.50.1.0)
Requirement already up-to-date: pillow in ./anaconda3/lib/python3.6/site-packages (from imageio>=2.1.2->mujoco-py==1.50.1.0)
Requirement already up-to-date: olefile in ./anaconda3/lib/python3.6/site-packages (from pillow->imageio>=2.1.2->mujoco-py==1.50.1.0)
Installing collected packages: mujoco-py
  Running setup.py install for mujoco-py ... error
    Complete output from command /Users/areddy/anaconda3/bin/python3 -u -c "import setuptools, tokenize;__file__='/private/var/folders/n5/cpmk_zdx5yg0xrtdsd3hd1y40000gn/T/pip-build-q95epzie/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /var/folders/n5/cpmk_zdx5yg0xrtdsd3hd1y40000gn/T/pip-3_4bt9y_-record/install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    running build_ext
    building 'mujoco_py.cymj' extension
    creating /private/var/folders/n5/cpmk_zdx5yg0xrtdsd3hd1y40000gn/T/pip-build-q95epzie/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.7-x86_64-3.6
    creating /private/var/folders/n5/cpmk_zdx5yg0xrtdsd3hd1y40000gn/T/pip-build-q95epzie/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.7-x86_64-3.6/private
    creating /private/var/folders/n5/cpmk_zdx5yg0xrtdsd3hd1y40000gn/T/pip-build-q95epzie/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.7-x86_64-3.6/private/var
    creating /private/var/folders/n5/cpmk_zdx5yg0xrtdsd3hd1y40000gn/T/pip-build-q95epzie/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.7-x86_64-3.6/private/var/folders
    creating /private/var/folders/n5/cpmk_zdx5yg0xrtdsd3hd1y40000gn/T/pip-build-q95epzie/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.7-x86_64-3.6/private/var/folders/n5
    creating /private/var/folders/n5/cpmk_zdx5yg0xrtdsd3hd1y40000gn/T/pip-build-q95epzie/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.7-x86_64-3.6/private/var/folders/n5/cpmk_zdx5yg0xrtdsd3hd1y40000gn
    creating /private/var/folders/n5/cpmk_zdx5yg0xrtdsd3hd1y40000gn/T/pip-build-q95epzie/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.7-x86_64-3.6/private/var/folders/n5/cpmk_zdx5yg0xrtdsd3hd1y40000gn/T
    creating /private/var/folders/n5/cpmk_zdx5yg0xrtdsd3hd1y40000gn/T/pip-build-q95epzie/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.7-x86_64-3.6/private/var/folders/n5/cpmk_zdx5yg0xrtdsd3hd1y40000gn/T/pip-build-q95epzie
    creating /private/var/folders/n5/cpmk_zdx5yg0xrtdsd3hd1y40000gn/T/pip-build-q95epzie/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.7-x86_64-3.6/private/var/folders/n5/cpmk_zdx5yg0xrtdsd3hd1y40000gn/T/pip-build-q95epzie/mujoco-py
    creating /private/var/folders/n5/cpmk_zdx5yg0xrtdsd3hd1y40000gn/T/pip-build-q95epzie/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.7-x86_64-3.6/private/var/folders/n5/cpmk_zdx5yg0xrtdsd3hd1y40000gn/T/pip-build-q95epzie/mujoco-py/mujoco_py
    creating /private/var/folders/n5/cpmk_zdx5yg0xrtdsd3hd1y40000gn/T/pip-build-q95epzie/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.7-x86_64-3.6/private/var/folders/n5/cpmk_zdx5yg0xrtdsd3hd1y40000gn/T/pip-build-q95epzie/mujoco-py/mujoco_py/gl
    /usr/local/bin/gcc-7 -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -I/Users/areddy/anaconda3/include -arch x86_64 -I/Users/areddy/anaconda3/include -arch x86_64 -DONMAC -I/private/var/folders/n5/cpmk_zdx5yg0xrtdsd3hd1y40000gn/T/pip-build-q95epzie/mujoco-py/mujoco_py -I/Users/areddy/.mujoco/mjpro150/include -I/Users/areddy/anaconda3/lib/python3.6/site-packages/numpy/core/include -I/Users/areddy/anaconda3/include/python3.6m -c /private/var/folders/n5/cpmk_zdx5yg0xrtdsd3hd1y40000gn/T/pip-build-q95epzie/mujoco-py/mujoco_py/cymj.c -o /private/var/folders/n5/cpmk_zdx5yg0xrtdsd3hd1y40000gn/T/pip-build-q95epzie/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.7-x86_64-3.6/private/var/folders/n5/cpmk_zdx5yg0xrtdsd3hd1y40000gn/T/pip-build-q95epzie/mujoco-py/mujoco_py/cymj.o -fopenmp -w
    /private/var/folders/n5/cpmk_zdx5yg0xrtdsd3hd1y40000gn/T/pip-build-q95epzie/mujoco-py/mujoco_py/cymj.c:481:10: fatal error: mjmodel.h: No such file or directory
     #include "mjmodel.h"
              ^~~~~~~~~~~
    compilation terminated.
    error: command '/usr/local/bin/gcc-7' failed with exit status 1
    
[... cut ...]
Command "/Users/areddy/anaconda3/bin/python3 -u -c "import setuptools, tokenize;__file__='/private/var/folders/n5/cpmk_zdx5yg0xrtdsd3hd1y40000gn/T/pip-build-q95epzie/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /var/folders/n5/cpmk_zdx5yg0xrtdsd3hd1y40000gn/T/pip-3_4bt9y_-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /private/var/folders/n5/cpmk_zdx5yg0xrtdsd3hd1y40000gn/T/pip-build-q95epzie/mujoco-py/  ```

machinaut, 2017-06-29:
In the GCC command line, the `-I/Users/areddy/.mujoco/mjpro150/include` should be finding that header file, can you verify that it's present in that folder?

ashwinreddy, 2017-06-29:
Oops, seems like I was using an earlier version of MuJoCo. After downloading 1.50 and putting it in the folder, it seems to work now. Thanks!

jonasschneider, 2017-06-29:
Thanks for reporting -- I've clarified the requirements in the [README](https://github.com/openai/mujoco-py/blob/master/README.md#requirements) and we're tracking support for 3.6 in #52.

closed 2017-06-29 by jonasschneider