openai/mujoco-py #45: Environment changed, not working for python2 anymore

hmishra2250, 2017-06-28:
In setup.py,  the first line 

> import importlib.util

This works only with python3! Also as mentioned in the installation guide, the eggs are not available through pip. I had to install using git (which failed of course!).

hmishra2250, 2017-06-28:
I just saw .python-version file which mentions python 3.5.2. Is there someway I can use it with python2? I am stuck cuz I was reading and working old source code, but this one seems way more organized. 

Thanks

jonasschneider, 2017-06-28:
Python 2 is not supported since the (just released) 1.50.1.x series.

You can keep using the [`0.5` branch](https://github.com/openai/mujoco-py/tree/0.5); the latest release is [`0.5.7`](https://github.com/openai/mujoco-py/releases/tag/0.5.7) which you can install with `pip install mujoco-py==0.5.7`.

Sorry for the inconvenience!

closed 2017-06-28 by jonasschneider