openai/mujoco-py #47: Unknow error installing using Python3 and Anaconda3 on Linux

hmishra2250, 2017-06-28:
After coming to know that this is not available in Python2, I did fresh install of Anaconda3, set up python3 environment and tried installing.
However, it throws errors such as:

> TypeError: can't pickle Cython.Compiler.Code.FunctionState objects

and

>  Command "/home/him/anaconda3/envs/tensorflow/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-j00wxlw4/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-dstl1tts-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-j00wxlw4/mujoco-py/

Attaching hereby the full installation log file and hardware configuration for reference.
[sysconfig.txt](https://github.com/openai/mujoco-py/files/1109729/sysconfig.txt)
[install_log.txt](https://github.com/openai/mujoco-py/files/1109730/install_log.txt)

hmishra2250, 2017-06-28:
Finally, After a lot of trying, I was able to install the mujoco_py lib in python3 global as well as Anacoda environment, but not using pip, instead using `python setup.py install` command.

There are some observations for _an end user_ that I would like to make, that might seem to be helpful:

1. *DO NOT* try building mujoco_py using python2. There is some bug in cython, which got rectified in python versions 3.5+
2. Set Up exact same environment as mentioned by the authors. I even failed with python 3.6.2. I don't know why!
3. Pip/Pip3 installation can fail. It did in my case all the time. So instead use `python setup.py install` command. This holds true for both Anaconda environment as well as global environment. For global, ofcourse u will need root access.
4. For building with CPU support, make sure you have the `libosmesa6-dev` library installed.

Some _Observations/Request to the authors_:

1. Please include `patchelf==0.9` and mention `autoreconf` as requirements in your documentation for building for GPU. 0.8 version was abruptly failing due to not having required functions.
2. In Builder.py, instead of `sys.platform == 'linux'`, please check for `sys.platform.startswith("linux")`. My Laptop shows 'linux2' as its platform.
3. The GPU OpenGL library address (`/usr/local/nvidia/`) is for older model. So for new GPUs, please add a check to `/usr/lib/nvidia-*/` or something similar. My laptop had the libs at `/usr/lib/nvidia-375/`. My GPU being GTX960M.

Thanks

machinaut, 2017-06-29:
We don't list out all the required packages for linux distributions, but the Dockerfile can provide a good reference for what is needed in an otherwise-empty linux machine.

jonasschneider, 2017-06-29:
Thanks @hmishra2250, we'll be tracking support for Python 3.6 in #52. Would you mind opening a PR for the other issues you mentioned? Thanks!

episodeyang, 2017-07-03:
@hmishra2250 Thank you so much for your comment! I ran into the same problem today and here is the make file that does the job:

- [ ] use `3.5.2` with conda
- [ ] clone and install
- [ ] use Dockerfile as a reference for the dependencies.

```Makefile
get-mujoco:
	wget -O mjpro150.zip https://www.roboti.us/download/mjpro150_linux.zip
	sudo apt-get install unzip
	rm -rf ~/.mujoco
	mkdir ~/.mujoco
	unzip mjpro150.zip -d ~/.mujoco/mjpro150
	rm -rf mjpro150.zip
	echo "now put your `mjkey.txt` file into ~/.mujoco/mjkey.txt"
	sudo vim ~/.mujoco/mjkey.txt
install-mujoco-dependencies:
	sudo apt-get update -q
	DEBIAN_FRONTEND=noninteractive sudo apt-get install -y \
    curl \
    git \
    libgl1-mesa-dev \
    libgl1-mesa-glx \
    libosmesa6-dev \
    python3-pip \
    python3-numpy \
    python3-scipy \
    unzip \
    vim \
    wget \
    xpra \
    xserver-xorg-dev
	sudo apt-get clean
	sudo rm -rf /var/lib/apt/lists/*
# install patch
	sudo curl -o /usr/local/bin/patchelf https://s3-us-west-2.amazonaws.com/openai-sci-artifacts/manual-builds/patchelf_0.9_amd64.elf
	sudo chmod +x /usr/local/bin/patchelf

install-mujoco:
# installing with pip simply doesn't work.
#	sudo pip3 install -U 'mujoco-py==1.50.1.0'
	rm -rf mujoco-py
	git clone https://github.com/openai/mujoco-py.git
	bash -c "cd mujoco-py && source activate gym && python setup.py install"
```

taosheng1993, 2017-07-03:
can anyone give me a specific procedure to install the mujoco_py which can be used well. After several days trying, I am tatally lost!!!

hmishra2250, 2017-07-03:
Yeah! I have mujoco_py working with me.
So the above procedure mentioned by @episodeyang  is for Linux. It'll work for sure, the only problem u will encounter is the MjViewer class will give seg fault or some GL init error. Check [this](https://github.com/openai/mujoco-py/issues/44) issue.
So to make it work, switch to windows, install using the exact same specifications mentioned as in [this](https://github.com/openai/mujoco-py/blob/master/requirements.txt) and python==3.5.2 file, and then install mujoco_py (try with pip, if that fails, try installing using `python setup.py install` after cloning the repo to the local system). You can also refer to [this appveyor file](https://github.com/openai/mujoco-py/blob/master/.appveyor.yml) for instructions related to how to install.
In windows, you will have the advantage of the viewer working well and good. So prefer windows for now.

_TL;DR_
1. Use Windows (suggested for Viewer to work)
2. Install python==3.5.2
3. Install all [dependencies](https://github.com/openai/mujoco-py/blob/master/requirements.txt) and [dev dependencies](https://github.com/openai/mujoco-py/blob/master/requirements.dev.txt) (if required).
4. Install mujoco_py using pip command and if that fails, use the cloned repo and install using setup.py (`python setup.py install`).

Good luck!

taosheng1993, 2017-07-03:
Hi,I am still a little confused. what do you mean referring to this this appveyor file for instructions? I have no  idea how to use the yml file? plz give me some clues.

DakshMiglani, 2017-07-04:
Hey there, Your script allowed me to install it on fedora but there's a problem it still gives me an error when I import it with python.
```
fatal error: mjmodel.h: No such file or directory
 #include "mjmodel.h" 
```

machinaut, 2017-07-06:
The header files (`mjmodel.h`, etc) have to be installed separately from this python package.  Make sure you've got those: https://github.com/openai/mujoco-py#install-mujoco

vitchyr, 2017-07-09:
@DakshMiglani You might also want to check out https://github.com/openai/mujoco-py/issues/43

puneetsinghal, 2017-07-22:
Hello everyone,
I tried to use the series of commands shared by @epidoseyang with slight modifications as:
```
sudo bash ./Anaconda3-4.4.0-Linux-x86_64.sh
source ~/.bashrc
# install-python3.5.2:
	conda create --name ml python=3.5.2
	source activate ml (tried without this command but got the same results)
# get-mujoco: (Done offline)
# install-mujoco-dependencies:
	sudo apt-get update -q
	DEBIAN_FRONTEND=noninteractive sudo apt-get install -y \
	curl \
	git \
	libgl1-mesa-dev \
	libgl1-mesa-glx \
	libosmesa6-dev \
	python3-pip \
	python3-numpy \
	python3-scipy \
	unzip \
	vim \
	wget \
	xpra \
	xserver-xorg-dev
	sudo apt-get clean
	sudo rm -rf /var/lib/apt/lists/*
# install patch
	sudo curl -o /usr/local/bin/patchelf https://s3-us-west-2.amazonaws.com/openai-sci-artifacts/manual-builds/patchelf_0.9_amd64.elf
	sudo chmod +x /usr/local/bin/patchelf

# install-mujoco:
	rm -rf mujoco-py
	git clone https://github.com/openai/mujoco-py.git
	bash -c "cd mujoco-py && source activate ml && python setup.py install"
```
I used Anaconda3 with python 3.5.2 on Ubuntu 16.04. I tried to use both mj150 and mj131 version but am getting same errors at "python setup.py install" command. Can someone please help me with the possible reasons. I would really appreciate any help I can get. I have attached the text file with output messages from installation process.

After this, I still tried to run the example file "body_interaction.py" and it threw errors as:
```
Compiling /home/puneet/.conda/envs/ml/lib/python3.5/site-packages/mujoco_py-1.50.1.10-py3.5.egg/mujoco_py/cymj.pyx because it changed.
[1/1] Cythonizing /home/puneet/.conda/envs/ml/lib/python3.5/site-packages/mujoco_py-1.50.1.10-py3.5.egg/mujoco_py/cymj.pyx
running build_ext
building 'mujoco_py.cymj' extension
creating /home/puneet/.conda/envs/ml/lib/python3.5/site-packages/mujoco_py-1.50.1.10-py3.5.egg/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder
creating /home/puneet/.conda/envs/ml/lib/python3.5/site-packages/mujoco_py-1.50.1.10-py3.5.egg/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5
creating /home/puneet/.conda/envs/ml/lib/python3.5/site-packages/mujoco_py-1.50.1.10-py3.5.egg/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/home
creating /home/puneet/.conda/envs/ml/lib/python3.5/site-packages/mujoco_py-1.50.1.10-py3.5.egg/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/home/puneet
creating /home/puneet/.conda/envs/ml/lib/python3.5/site-packages/mujoco_py-1.50.1.10-py3.5.egg/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/home/puneet/.conda
creating /home/puneet/.conda/envs/ml/lib/python3.5/site-packages/mujoco_py-1.50.1.10-py3.5.egg/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/home/puneet/.conda/envs
creating /home/puneet/.conda/envs/ml/lib/python3.5/site-packages/mujoco_py-1.50.1.10-py3.5.egg/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/home/puneet/.conda/envs/ml
creating /home/puneet/.conda/envs/ml/lib/python3.5/site-packages/mujoco_py-1.50.1.10-py3.5.egg/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/home/puneet/.conda/envs/ml/lib
creating /home/puneet/.conda/envs/ml/lib/python3.5/site-packages/mujoco_py-1.50.1.10-py3.5.egg/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/home/puneet/.conda/envs/ml/lib/python3.5
creating /home/puneet/.conda/envs/ml/lib/python3.5/site-packages/mujoco_py-1.50.1.10-py3.5.egg/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/home/puneet/.conda/envs/ml/lib/python3.5/site-packages
creating /home/puneet/.conda/envs/ml/lib/python3.5/site-packages/mujoco_py-1.50.1.10-py3.5.egg/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/home/puneet/.conda/envs/ml/lib/python3.5/site-packages/mujoco_py-1.50.1.10-py3.5.egg
creating /home/puneet/.conda/envs/ml/lib/python3.5/site-packages/mujoco_py-1.50.1.10-py3.5.egg/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/home/puneet/.conda/envs/ml/lib/python3.5/site-packages/mujoco_py-1.50.1.10-py3.5.egg/mujoco_py
creating /home/puneet/.conda/envs/ml/lib/python3.5/site-packages/mujoco_py-1.50.1.10-py3.5.egg/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/home/puneet/.conda/envs/ml/lib/python3.5/site-packages/mujoco_py-1.50.1.10-py3.5.egg/mujoco_py/gl
gcc -pthread -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -I/home/puneet/.conda/envs/ml/lib/python3.5/site-packages/mujoco_py-1.50.1.10-py3.5.egg/mujoco_py -I/home/puneet/.mujoco/mjpro150/include -I/home/puneet/mujoco-py/.eggs/numpy-1.13.1-py3.5-linux-x86_64.egg/numpy/core/include -I/home/puneet/.conda/envs/ml/include/python3.5m -c /home/puneet/.conda/envs/ml/lib/python3.5/site-packages/mujoco_py-1.50.1.10-py3.5.egg/mujoco_py/cymj.c -o /home/puneet/.conda/envs/ml/lib/python3.5/site-packages/mujoco_py-1.50.1.10-py3.5.egg/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/home/puneet/.conda/envs/ml/lib/python3.5/site-packages/mujoco_py-1.50.1.10-py3.5.egg/mujoco_py/cymj.o -fopenmp -w
gcc -pthread -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -I/home/puneet/.conda/envs/ml/lib/python3.5/site-packages/mujoco_py-1.50.1.10-py3.5.egg/mujoco_py -I/home/puneet/.mujoco/mjpro150/include -I/home/puneet/mujoco-py/.eggs/numpy-1.13.1-py3.5-linux-x86_64.egg/numpy/core/include -I/home/puneet/.conda/envs/ml/includ [... cut ...] joco/mjpro150/bin -L/home/puneet/.conda/envs/ml/lib -Wl,--enable-new-dtags,-R/home/puneet/.mujoco/mjpro150/bin -lmujoco150 -lglewosmesa -lOSMesa -lGL -lpython3.5m -o /home/puneet/.conda/envs/ml/lib/python3.5/site-packages/mujoco_py-1.50.1.10-py3.5.egg/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/lib.linux-x86_64-3.5/mujoco_py/cymj.cpython-35m-x86_64-linux-gnu.so -fopenmp
Illegal instruction (core dumped)
```
[error_log.txt](https://github.com/openai/mujoco-py/files/1167129/error_log.txt)

episodeyang, 2017-07-22:
@puneetsinghal I have never seen that error before. Here is the updated script for downloading MuJoCo131.

```bash
MJC_VER=131

get-mujoco-id:
	wget https://www.roboti.us/getid/getid_linux
	chmod +x ./getid_linux
	./getid_linux
	rm ./getid_linux

get-mujoco:
	wget -O mjpro$(MJC_VER).zip https://www.roboti.us/download/mjpro$(MJC_VER)_linux.zip
	sudo apt-get install unzip
	mkdir -p ~/.mujoco
	rm -rf ~/.mujoco/mjpro$(MJC_VER)
	unzip mjpro$(MJC_VER).zip -d ~/.mujoco
	rm -rf mjpro$(MJC_VER).zip
	echo "now put your mjkey.txt file into ~/.mujoco/mjkey.txt"
	sudo vim ~/.mujoco/mjkey.txt

install-mujoco-dependencies:
	sudo apt-get update -q
	DEBIAN_FRONTEND=noninteractive sudo apt-get install -y \
    curl \
    git \
    libgl1-mesa-dev \
    libgl1-mesa-glx \
    libosmesa6-dev \
    python3-pip \
    python3-numpy \
    python3-scipy \
    unzip \
    vim \
    wget \
    xpra \
    xserver-xorg-dev
	sudo apt-get clean
	sudo rm -rf /var/lib/apt/lists/*
# install patch
	sudo curl -o /usr/local/bin/patchelf https://s3-us-west-2.amazonaws.com/openai-sci-artifacts/manual-builds/patchelf_0.9_amd64.elf
	sudo chmod +x /usr/local/bin/patchelf
install-glfw:
	bash -c "source activate gym && pip install glfw --ignore-installed"
	rm -rf glfw
	git clone https://github.com/glfw/glfw.git
	bash -c "cd glfw && cmake -DBUILD_SHARED_LIBS=ON . && sudo make install"
install-mujoco-py:
	# installing with pip simply doesn't work.
	# sudo pip3 install -U 'mujoco-py==1.50.1.0'
	rm -rf mujoco-py
	git clone https://github.com/openai/mujoco-py.git
	bash -c "source activate gym && cd mujoco-py && python setup.py install"
```

puneetsinghal, 2017-07-25:
@episodeyang: Thank you so much for the response. I tried your new makefile multiple times but unfortunately received similar errors. It starts throwing errors when it is installing mujoco_py. I am attaching the initial errors. I am totally stuck in installation of mujoco-py and would really appreciate if you/ anyone can help me further with the troubleshooting. 

```
Cloning into 'mujoco-py'...
remote: Counting objects: 963, done.
remote: Compressing objects: 100% (125/125), done.
remote: Total 963 (delta 151), reused 215 (delta 120), pack-reused 708
Receiving objects: 100% (963/963), 4.53 MiB | 5.21 MiB/s, done.
Resolving deltas: 100% (455/455), done.
Checking connectivity... done.
zip_safe flag not set; analyzing archive contents...
imageio.plugins.__pycache__.pillow_info.cpython-35: module references __file__
imageio.__pycache__.testing.cpython-35: module references __file__
imageio.core.__pycache__.util.cpython-35: module references __file__
imageio.core.__pycache__.findlib.cpython-35: module references __file__

Installed /home/puneet/mujoco-py/.eggs/imageio-2.2.0-py3.5.egg
Searching for Cython>=0.25.2
Reading https://pypi.python.org/simple/Cython/
Downloading https://pypi.python.org/packages/10/d5/753d2cb5073a9f4329d1ffed1de30b0458821780af8fdd8ba1ad5adb6f62/Cython-0.26.tar.gz#md5=2fa7ea73eb9944fdad3989adbfd22088
Best match: Cython 0.26
Processing Cython-0.26.tar.gz
Writing /tmp/easy_install-l4cxdxk0/Cython-0.26/setup.cfg
Running Cython-0.26/setup.py -q bdist_egg --dist-dir /tmp/easy_install-l4cxdxk0/Cython-0.26/egg-dist-tmp-bre9yr6o
Unable to find pgen, not compiling formal grammar.
warning: no files found matching 'Doc/*'
warning: no files found matching '*.pyx' under directory 'Cython/Debugger/Tests'
warning: no files found matching '*.pxd' under directory 'Cython/Debugger/Tests'
warning: no files found matching '*.pxd' under directory 'Cython/Utility'
creating /home/puneet/mujoco-py/.eggs/Cython-0.26-py3.5-linux-x86_64.egg
Extracting Cython-0.26-py3.5-linux-x86_64.egg to /home/puneet/mujoco-py/.eggs

Installed /home/puneet/mujoco-py/.eggs/Cython-0.26-py3.5-linux-x86_64.egg
Searching for pillow
Reading https://pypi.python.org/simple/pillow/
Downloading https://pypi.python.org/packages/55/aa/f7f983fb72710a9daa4b3374b7c160091d3f94f5c09221f9336ade9027f3/Pillow-4.2.1.tar.gz#md5=11e5e2046cf41586716a6042d89abeeb
Best match: Pillow 4.2.1
Processing Pillow-4.2.1.tar.gz
Writing /tmp/easy_install-zfnwnoam/Pillow-4.2.1/setup.cfg
Running Pillow-4.2.1/setup.py -q bdist_egg --dist-dir /tmp/easy_install-zfnwnoam/Pillow-4.2.1/egg-dist-tmp-j5yy161c
warning: no files found matching '*.sh'
no previously-included directories found matching 'docs/_static'
warning: no previously-included files found matching '.coveragerc'
warning: no previously-included files found matching '.editorconfig'
warning: no previously-included files found matching '.landscape.yaml'
warning: no previously-included files found matching '.travis'
warning: no previously-included files found matching '.travis/*'
warning: no previously-included files found matching 'appveyor.yml'
warning: no previously-included files found matching 'build_children.sh'
warning: no previously-included files found matching 'tox.ini'
warning: no previously-included files matching '.git*' found anywhere in distribution
warning: no previously-included files matching '*.pyc' found anywhere in distribution
warning: no previously-included files matching '*.so' found anywhere in distribution
Building using 4 processes
path.c: In function ‘alloc_array’:
path.c:61:15: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     if (count > (SIZE_MAX / (2 * sizeof(double))) - 1 ) {
               ^
libImaging/Resample.c: In function ‘precompute_coeffs’:
libImaging/Resample.c:151:17: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     if (outSize > INT_MAX / (kmax * sizeof(double)))
                 ^
libImaging/Draw.c: In function ‘allocate’:
libImaging/Draw.c:940:31: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
             if (outline->size > INT_MAX / sizeof(Edge)) {
                               ^
libImaging/JpegEncode.c: In function ‘ImagingJpegEncode’:
libImaging/JpegEncode.c:244:37: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         if (context->rawExifLen + 5 >  context->destination.pub.free_in_buffer){
                                     ^
libImaging/QuantOctree.c: In function ‘create_sorted_color_palette’:
libImaging/QuantOctree.c:166:19: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
    if (cube->size > LONG_MAX / sizeof(struct _ColorBucket)) {
                   ^
libImaging/QuantOctree.c: In function ‘copy_color_cube’:
libImaging/QuantOctree.c:201:20: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
    if (cube->rBits > rBits) {
                    ^
libImaging/QuantOctree.c:208:20: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
    if (cube->gBits > gBits) {
                    ^
libImaging/QuantOctree.c:215:20: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
    if (cube->bBits > bBits) {
                    ^
libImaging/QuantOctree.c:222:20: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
[... cut ...]
libImaging/QuantOctree.c: In function ‘combined_palette’:
libImaging/QuantOctree.c:297:30: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
        (nBucketsA+nBucketsB) > LONG_MAX / sizeof(struct _ColorBucket)) {
                              ^
libImaging/RankFilter.c: In function ‘ImagingRankFilter’:
libImaging/RankFilter.c:65:14: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         size > INT_MAX / (size * sizeof(FLOAT32))) {
```

machinaut, 2017-10-26:
@puneetsinghal the `illegal instruction` might mean your CPU is missing AVX instructions.

Unfortunately AVX is required for mujoco 1.50, though it's only likely to be on processors from 2010 or later.

Suman7495, 2018-02-10:
@episodeyang I followed your script to install mujoco-py on my Linux Mint, which is based on Ubuntu 16.04. Upon successful installation of mujoco-py, when I try the following code:
```
import gym
env = gym.make('Humanoid-v1')
env.reset()
env.render()
```
I get the following error:
```
running build_ext
building 'mujoco_py.cymj' extension
x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/home/suman/mujoco-py/mujoco_py -I/home/suman/.mujoco/mjpro150/include -I/home/suman/.local/lib/python3.5/site-packages/numpy/core/include -I/usr/include/python3.5m -c /home/suman/mujoco-py/mujoco_py/cymj.c -o /home/suman/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/temp.linux-x86_64-3.5/home/suman/mujoco-py/mujoco_py/cymj.o -fopenmp -w
/home/suman/mujoco-py/mujoco_py/cymj.c:569:21: fatal error: mjmodel.h: No such file or directory
compilation terminated.
Traceback (most recent call last):
  File "/usr/lib/python3.5/distutils/unixccompiler.py", line 118, in _compile
    extra_postargs)
  File "/usr/lib/python3.5/distutils/ccompiler.py", line 909, in spawn
    spawn(cmd, dry_run=self.dry_run)
  File "/usr/lib/python3.5/distutils/spawn.py", line 36, in spawn
    _spawn_posix(cmd, search_path, dry_run=dry_run)
  File "/usr/lib/python3.5/distutils/spawn.py", line 159, in _spawn_posix
    % (cmd, exit_status))
distutils.errors.DistutilsExecError: command 'x86_64-linux-gnu-gcc' failed with exit status 1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "humanoid.py", line 2, in <module>
    env = gym.make('Humanoid-v2')
  File "/home/suman/gym/gym/envs/registration.py", line 163, in make
    return registry.make(id)
  File "/home/suman/gym/gym/envs/registration.py", line 119, in make
    env = spec.make()
  File "/home/suman/gym/gym/envs/registration.py", line 85, in make
    cls = load(self._entry_point)
  File "/home/suman/gym/gym/envs/registration.py", line 14, in load
    result = entry_point.load(False)
  File "/usr/local/lib/python3.5/dist-packages/pkg_resources/__init__.py", line 2408, in load
    return self.resolve()
  File "/usr/local/lib/python3.5/dist-packages/pkg_resources/__init__.py", line 2414, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/home/suman/gym/gym/envs/mujoco/__init__.py", line 1, in <module>
    from gym.envs.mujoco.mujoco_env import MujocoEnv
  File "/home/suman/gym/gym/envs/mujoco/mujoco_env.py", line 11, in <module>
    import mujoco_py
  File "/home/suman/mujoco-py/mujoco_py/__init__.py", line 1, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/home/suman/mujoco-py/mujoco_py/builder.py", line 502, in <module>
    cymj = load_cython_ext(mjpro_path)
  File "/home/suman/mujoco-py/mujoco_py/builder.py", line 81, in load_cython_ext
    cext_so_path = builder.build()
  File "/home/suman/mujoco-py/mujoco_py/builder.py", line 238, in build
    built_so_file_path = self._build_impl()
  File "/home/suman/mujoco-py/mujoco_py/builder.py", line 261, in _build_impl
    dist.run_commands()
  File "/usr/lib/python3.5/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/usr/lib/python3.5/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/home/suman/.local/lib/python3.5/site-packages/Cython/Distutils/old_build_ext.py", line 186, in run
    _build_ext.build_ext.run(self)
  File "/usr/lib/python3.5/distutils/command/build_ext.py", line 338, in run
    self.build_extensions()
  File "/home/suman/mujoco-py/mujoco_py/builder.py", line 157, in build_extensions
    build_ext.build_extensions(self)
  File "/home/suman/.local/lib/python3.5/site-packages/Cython/Distutils/old_build_ext.py", line 194, in build_extensions
    self.build_extension(ext)
  File "/usr/lib/python3.5/distutils/command/build_ext.py", line 532, in build_extension
    depends=ext.depends)
  File "/home/suman/mujoco-py/mujoco_py/builder.py", line 127, in caching_compile
    self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
  File "/usr/lib/python3.5/distutils/unixccompiler.py", line 120, in _compile
    raise CompileError(msg)
distutils.errors.CompileError: command 'x86_64-linux-gnu-gcc' failed with exit status 1
```
Would you know how to resolve this?