openai/mujoco-py #48: Error installing version 1.50.1.0 on Mac

mirisr, 2017-06-29:
I'm easily able to install the older version even with my python3, but I get this error when trying to install the latest version:

```MacBook-Pro-3:~ Iris$ pip install -U 'mujoco-py<1.50.2,>=1.50.1'
Collecting mujoco-py<1.50.2,>=1.50.1
  Using cached mujoco-py-1.50.1.0.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/private/var/folders/q0/n8dmc77j3n3gj91k_3p90kn40000gp/T/pip-build-0p9_zA/mujoco-py/setup.py", line 2, in <module>
        import importlib.util
    ImportError: No module named util```

machinaut, 2017-06-29:
What version of python are you using?  `importlib` should be a builtin package, but I'm not sure if it's available before 3.4: https://docs.python.org/3/library/importlib.html

mirisr, 2017-06-29:
Not sure how long it's suppose to take to install, but now I tried it with `python3.6 -m pip install SomePackage  # specifically Python 3.6`

and it's hanging :(

mirisr, 2017-06-29:
Seems like pip3 works but now I get this error: 
`...
  ...  
    File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/copy.py", line 169, in deepcopy
        rv = reductor(4)
    TypeError: can't pickle Cython.Compiler.Code.FunctionState objects
    
    ----------------------------------------
Command "/Library/Frameworks/Python.framework/Versions/3.6/bin/python3.6 -u -c "import setuptools, tokenize;__file__='/private/var/folders/q0/n8dmc77j3n3gj91k_3p90kn40000gp/T/pip-build-s0lguyhk/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /var/folders/q0/n8dmc77j3n3gj91k_3p90kn40000gp/T/pip-_z24o127-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /private/var/folders/q0/n8dmc77j3n3gj91k_3p90kn40000gp/T/pip-build-s0lguyhk/mujoco-py/`

mirisr, 2017-06-29:
Turns out I need to have python 3.5

and run : ` pip3.5 install -U 'mujoco-py<1.50.2,>=1.50.1' ` instead

mirisr, 2017-06-29:
Downloaded successfully

closed 2017-06-29 by mirisr