openai/mujoco-py #49: Error instaling on Ubuntu 14.04 with Python 3.6

heiko0403, 2017-06-29:
I'm trying to install version 1.50.1.0 on Ubuntu 14.04 using Python 3.6.

I tried 
$ pip install -U 'mujoco-py<1.50.2,>=1.50.1'
but returned unknown error.

So in ./mujoco-py, i tried 
$ python setup.py install
and it looks like successfull.

However, when I import mujoco-py in IPython,
TypeError: can't pickle Cython.Compiler.Code.FunctionState objects
returned.

What should I do?

hmishra2250, 2017-06-29:
Lately I experimented with a lot of settings and installation procedure, and the summary cum observations have been given on my comment here.
I am sure my comments [here](https://github.com/openai/mujoco-py/issues/47) will help
P.S try installing without pip (using setup.py) and if it doesnt work, then switch to exact same environment as mentioned, i.e Python3.5.2 and other requirements.

heiko0403, 2017-06-29:
Thank you! 
I tried to install with Python 3.5.2, and I could install version 1.50.1.0!

but...
When I tried examples/ programs, I got 
running build_ext
Creating window glfw
Segmentation fault 

I think this is the trouble of mujoco-py.MjViewer(), but I do not know what to do.
Do you have any idea?

jonasschneider, 2017-06-29:
Thanks for reporting. We're tracking support for 3.6. in #52. Could you open a new issue with the above error? One thing to try could be to run your program under `gdb` to get a backtrace for where exactly the segfault occurs.

closed 2017-06-29 by jonasschneider