openai/mujoco-py #53: distutils.errors.CompileError: command '/usr/local/bin/gcc-7' failed with exit status 1

abhigenie92, 2017-06-30:
```
>>> import mujoco_py
Compiling /Users/abhishek/Packages/mujoco-py/mujoco_py/cymj.pyx because it changed.
[1/1] Cythonizing /Users/abhishek/Packages/mujoco-py/mujoco_py/cymj.pyx
running build_ext
building 'mujoco_py.cymj' extension
creating /Users/abhishek/Packages/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder
creating /Users/abhishek/Packages/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.7-x86_64-3.5
creating /Users/abhishek/Packages/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.7-x86_64-3.5/Users
creating /Users/abhishek/Packages/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.7-x86_64-3.5/Users/abhishek
creating /Users/abhishek/Packages/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.7-x86_64-3.5/Users/abhishek/Packages
creating /Users/abhishek/Packages/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.7-x86_64-3.5/Users/abhishek/Packages/mujoco-py
creating /Users/abhishek/Packages/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.7-x86_64-3.5/Users/abhishek/Packages/mujoco-py/mujoco_py
creating /Users/abhishek/Packages/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.7-x86_64-3.5/Users/abhishek/Packages/mujoco-py/mujoco_py/gl
/usr/local/bin/gcc-7 -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -I/anaconda/envs/python3/include -arch x86_64 -DONMAC -I/Users/abhishek/Packages/mujoco-py/mujoco_py -I/Users/abhishek/.mujoco/mjpro150/include -I/Users/abhishek/.virtualenvs/python3/lib/python3.5/site-packages/numpy/core/include -I/Users/abhishek/.virtualenvs/python3/include/python3.5m -c /Users/abhishek/Packages/mujoco-py/mujoco_py/cymj.c -o /Users/abhishek/Packages/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.7-x86_64-3.5/Users/abhishek/Packages/mujoco-py/mujoco_py/cymj.o -fopenmp -w
/Users/abhishek/Packages/mujoco-py/mujoco_py/cymj.c:476:10: fatal error: mjmodel.h: No such file or directory
 #include "mjmodel.h"
          ^~~~~~~~~~~
compilation terminated.
Traceback (most recent call last):
  File "/Users/abhishek/.virtualenvs/python3/lib/python3.5/distutils/unixccompiler.py", line 118, in _compile
    extra_postargs)
  File "/Users/abhishek/.virtualenvs/python3/lib/python3.5/distutils/ccompiler.py", line 909, in spawn
    spawn(cmd, dry_run=self.dry_run)
  File "/Users/abhishek/.virtualenvs/python3/lib/python3.5/distutils/spawn.py", line 36, in spawn
    _spawn_posix(cmd, search_path, dry_run=dry_run)
  File "/Users/abhishek/.virtualenvs/python3/lib/python3.5/distutils/spawn.py", line 159, in _spawn_posix
    % (cmd, exit_status))
distutils.errors.DistutilsExecError: command '/usr/local/bin/gcc-7' failed with exit status 1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/abhishek/Packages/mujoco-py/mujoco_py/__init__.py", line 1, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/Users/abhishek/Packages/mujoco-py/mujoco_py/builder.py", line 283, in <module>
    cymj = load_cython_ext(mjpro_path)
  File "/Users/abhishek/Packages/mujoco-py/mujoco_py/builder.py", line 54, in load_cython_ext
    cext_so_path = builder.build()
  File "/Users/abhishek/Packages/mujoco-py/mujoco_py/builder.py", line 200, in build
    so_file_path = super().build()
  File "/Users/abhishek/Packages/mujoco-py/mujoco_py/builder.py", line 131, in build
    dist.run_commands()
  File "/Users/abhishek/.virtualenvs/python3/lib/python3.5/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/Users/abhishek/.virtualenvs/python3/lib/python3.5/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/Users/abhishek/.virtualenvs/python3/lib/python3.5/site-packages/Cython/Distutils/old_build_ext.py", line 185, in run
    _build_ext.build_ext.run(self)
  File "/Users/abhishek/.virtualenvs/python3/lib/python3.5/distutils/command/build_ext.py", line 339, in run
    self.build_extensions()
  File "/Users/abhishek/Packages/mujoco-py/mujoco_py/builder.py", line 75, in build_extensions
    build_ext.build_extensions(self)
  File "/Users/abhishek/.virtualenvs/python3/lib/python3.5/site-packages/Cython/Distutils/old_build_ext.py", line 193, in build_extensions
    self.build_extension(ext)
  File "/Users/abhishek/.virtualenvs/python3/lib/python3.5/distutils/command/build_ext.py", line 533, in build_extension
    depends=ext.depends)
  File "/Users/abhishek/.virtualenvs/python3/lib/python3.5/distutils/ccompiler.py", line 574, in compile
    self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
  File "/Users/abhishek/.virtualenvs/python3/lib/python3.5/distutils/unixccompiler.py", line 120, in _compile
    raise CompileError(msg)
distutils.errors.CompileError: command '/usr/local/bin/gcc-7' failed with exit status 1
```

hmishra2250, 2017-06-30:
Can you try installation without pop, using the command `python setup.py install`, after cloning/downloading the repository.
It did work for me, even when the pip one failed.
P.S read my comment [here](https://github.com/openai/mujoco-py/issues/47)

abhigenie92, 2017-06-30:
Installing from repository, makes no difference to the error for me.

abhigenie92, 2017-07-01:
Any ideas on resolving this issue?

episodeyang, 2017-07-03:
@abhigenie92 use python3.5.2 then install from this git repo solved it for me. For detailed script you can take a look here:
https://github.com/openai/mujoco-py/issues/47#issuecomment-312551946

Duncanswilson, 2017-07-12:
@episodeyang  Just a heads up, I'm attempting to install on Ubuntu 16.04 and followed the exact commands that you linked in #47 and am still having the issue which @abhigenie92 has!

Duncanswilson, 2017-07-12:
I actually found the isssue, in `~/.mujoco/mjpro150` there was another `mjpro150/` folder with mjuoco in it. So I moved the contents of `~/.mujoco/mjpro150/mjpro150` to `~/.mujoco/mjpro150/` and deleted the empty `~/.mujoco/mjpro150/mjpro150` folder and now everything works! 

Hope this helps:sparkles:

Duncanswilson, 2017-07-12:
I also had a `ImportError: Failed to load GLFW3 shared library.` error after this which was solved by `sudo apt-get install libglfw3` :)

episodeyang, 2017-07-12:
@Duncanswilson Thanks for the comment! Yes I saw those mistakes. A more daming mistake in my script is one that I'm trying to use `mujoco1.50` with the `gym`. `gym` only work with `mujoco1.3` as far as I know :joy:

abhigenie92, 2017-11-08:
Still suck on it.
I am assuming things now support 3.6?
```
~/P/mujoco-py [master]× » python                                          (ppo_latest) 22:50:39
Python 3.6.2 |Continuum Analytics, Inc.| (default, Jul 20 2017, 13:14:59)
[GCC 4.2.1 Compatible Apple LLVM 6.0 (clang-600.0.57)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import mujoco-py
  File "<stdin>", line 1
    import mujoco-py
                 ^
SyntaxError: invalid syntax
>>> import mujoco_py
running build_ext
building 'mujoco_py.cymj' extension
/usr/local/bin/gcc-7 -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -I/anaconda/envs/ppo_latest/include -arch x86_64 -I/anaconda/envs/ppo_latest/include -arch x86_64 -DONMAC -Imujoco_py -I/Users/abhishek/Packages/mujoco-py/mujoco_py -I/Users/abhishek/.mujoco/mjpro150/include -I/anaconda/envs/ppo_latest/lib/python3.6/site-packages/numpy/core/include -I/anaconda/envs/ppo_latest/include/python3.6m -c /Users/abhishek/Packages/mujoco-py/mujoco_py/cymj.c -o /Users/abhishek/Packages/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.7-x86_64-3.6/Users/abhishek/Packages/mujoco-py/mujoco_py/cymj.o -fopenmp -w
In file included from /usr/local/Cellar/gcc/7.2.0/lib/gcc/7/gcc/x86_64-apple-darwin16.7.0/7.2.0/include-fixed/syslimits.h:7:0,
                 from /usr/local/Cellar/gcc/7.2.0/lib/gcc/7/gcc/x86_64-apple-darwin16.7.0/7.2.0/include-fixed/limits.h:34,
                 from /anaconda/envs/ppo_latest/include/python3.6m/Python.h:11,
                 from /Users/abhishek/Packages/mujoco-py/mujoco_py/cymj.c:57:
/usr/local/Cellar/gcc/7.2.0/lib/gcc/7/gcc/x86_64-apple-darwin16.7.0/7.2.0/include-fixed/limits.h:194:61: error: no include path in which to search for limits.h
 #include_next <limits.h>  /* recurse down to the real one */
                                                             ^
In file included from /Users/abhishek/Packages/mujoco-py/mujoco_py/cymj.c:57:0:
/anaconda/envs/ppo_latest/include/python3.6m/Python.h:25:10: fatal error: stdio.h: No such file or directory
 #include <stdio.h>
          ^~~~~~~~~
compilation terminated.
Traceback (most recent call last):
  File "/anaconda/envs/ppo_latest/lib/python3.6/distutils/unixccompiler.py", line 118, in _compile
    extra_postargs)
  File "/anaconda/envs/ppo_latest/lib/python3.6/distutils/ccompiler.py", line 909, in spawn
    spawn(cmd, dry_run=self.dry_run)
  File "/anaconda/envs/ppo_latest/lib/python3.6/distutils/spawn.py", line 36, in spawn
    _spawn_posix(cmd, search_path, dry_run=dry_run)
  File "/anaconda/envs/ppo_latest/lib/python3.6/distutils/spawn.py", line 159, in _spawn_posix
    % (cmd, exit_status))
distutils.errors.DistutilsExecError: command '/usr/local/bin/gcc-7' failed with exit status 1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/abhishek/Packages/mujoco-py/mujoco_py/__init__.py", line 1, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/Users/abhishek/Packages/mujoco-py/mujoco_py/builder.py", line 439, in <module>
    cymj = load_cython_ext(mjpro_path)
  File "/Users/abhishek/Packages/mujoco-py/mujoco_py/builder.py", line 80, in load_cython_ext
    cext_so_path = builder.build()
  File "/Users/abhishek/Packages/mujoco-py/mujoco_py/builder.py", line 187, in build
    built_so_file_path = self._build_impl()
  File "/Users/abhishek/Packages/mujoco-py/mujoco_py/builder.py", line 280, in _build_impl
    so_file_path = super()._build_impl()
  File "/Users/abhishek/Packages/mujoco-py/mujoco_py/builder.py", line 207, in _build_impl
    dist.run_commands()
  File "/anaconda/envs/ppo_latest/lib/python3.6/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/anaconda/envs/ppo_latest/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/Users/abhishek/Packages/mujoco-py/.eggs/Cython-0.27.3-py3.6-macosx-10.7-x86_64.egg/Cython/Distutils/old_build_ext.py", line 186, in run
    _build_ext.build_ext.run(self)
  File "/anaconda/envs/ppo_latest/lib/python3.6/distutils/command/build_ext.py", line 339, in run
    self.build_extensions()
  File "/Users/abhishek/Packages/mujoco-py/mujoco_py/builder.py", line 106, in build_extensions
    build_ext.build_extensions(self)
  File "/Users/abhishek/Packages/mujoco-py/.eggs/Cython-0.27.3-py3.6-macosx-10.7-x86_64.egg/Cython/Distutils/old_build_ext.py", line 194, in build_extensions
    self.build_extension(ext)
  File "/anaconda/envs/ppo_latest/lib/python3.6/distutils/command/build_ext.py", line 533, in build_extension
    depends=ext.depends)
  File "/anaconda/envs/ppo_latest/lib/python3.6/distutils/ccompiler.py", line 574, in compile
    self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
  File "/anaconda/envs/ppo_latest/lib/python3.6/distutils/unixccompiler.py", line 120, in _compile
    raise CompileError(msg)
distutils.errors.CompileError: command '/usr/local/bin/gcc-7' failed with exit status 1
>>> exit()
```

ElementLi, 2017-11-15:
@abhigenie92 I upgraded my gcc from 7.1 to 7.2, everything seems working now. 
`brew upgrade gcc`

danielpcox, 2018-03-25:
I was able to solve this problem by clone the repo and doing `pip install -e '.[all]'` with it.
Using Anaconda and Python 3.6.

dantuluri, 2018-08-14:
Could this be an issue with the gcc version on mac? I currently have gcc 4.2 installed and am trying to get to 6 but am having some issues with how to link it (I installed gcc@6 using homebrew).

dantuluri, 2018-08-14:
How are others not getting this problem? I reset my machine for this and am still having the same exact issue on macos 10.13.6

Zianfu96, 2018-09-02:
@TeaPearce Were you able to fix it? I have the same issue.

lutein, 2018-09-10:
@abhigenie92 Hi, I met the same error "Could not open activation key file" while the key file is actually where it should be. Have you solved this?

TeaPearce, 2018-11-22:
came back to this after a while. think it's now working after following this thread https://stackoverflow.com/questions/52509602/cant-compile-c-program-on-a-mac-after-upgrade-to-mojave

machinaut, 2018-11-28:
Closing this issue for now because the build system has changed since.

If any new users encounter something similar, please open a new issue! There's an issue template which helps us in getting the information we need.

Thanks~

closed 2018-11-28 by machinaut

ahmarz, 2020-05-11:
cd into mujoco-py then
python3 setup.py install
python

>>> import mujoco_py
Compiling /Users/q/phd/mujoco-py/mujoco_py/cymj.pyx because it depends on /Users/q/mujoco_env/lib/python3.7/site-packages/Cython-3.0a4-py3.7.egg/Cython/Includes/numpy/__init__.pxd.
[1/1] Cythonizing /Users/q/phd/mujoco-py/mujoco_py/cymj.pyx
warning: mujoco_py/generated/wrappers.pxi:4:0: Dotted filenames ('../pxd/mujoco.pxd') are deprecated. Please use the normal Python package directory layout.
running build_ext
building 'mujoco_py.cymj' extension
/usr/local/bin/gcc-9 -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -iwithsysroot/System/Library/Frameworks/System.framework/PrivateHeaders -iwithsysroot/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.7/Headers -DONMAC -Imujoco_py -I/Users/q/phd/mujoco-py/mujoco_py -I/Users/q/.mujoco/mujoco200/include -I/Users/q/mujoco_env/lib/python3.7/site-packages/numpy-1.18.4-py3.7-macosx-10.14-x86_64.egg/numpy/core/include -I/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.7/include/python3.7m -c /Users/q/phd/mujoco-py/mujoco_py/cymj.c -o /Users/q/phd/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.9_37_macextensionbuilder/temp.macosx-10.14-x86_64-3.7/Users/q/phd/mujoco-py/mujoco_py/cymj.o -fopenmp -w
gcc-9: error: unrecognized command line option '-iwithsysroot/System/Library/Frameworks/System.framework/PrivateHeaders'
gcc-9: error: unrecognized command line option '-iwithsysroot/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.7/Headers'
Traceback (most recent call last):
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.7/lib/python3.7/distutils/unixccompiler.py", line 118, in _compile
    extra_postargs)
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.7/lib/python3.7/distutils/ccompiler.py", line 909, in spawn
    spawn(cmd, dry_run=self.dry_run)
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.7/lib/python3.7/distutils/spawn.py", line 36, in spawn
    _spawn_posix(cmd, search_path, dry_run=dry_run)
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.7/lib/python3.7/distutils/spawn.py", line 159, in _spawn_posix
    % (cmd, exit_status))
distutils.errors.DistutilsExecError: command '/usr/local/bin/gcc-9' failed with exit status 1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/q/phd/mujoco-py/mujoco_py/__init__.py", line 3, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/Users/q/phd/mujoco-py/mujoco_py/builder.py", line 510, in <module>
    cymj = load_cython_ext(mujoco_path)
  File "/Users/q/phd/mujoco-py/mujoco_py/builder.py", line 105, in load_cython_ext
    cext_so_path = builder.build()
  File "/Users/q/phd/mujoco-py/mujoco_py/builder.py", line 221, in build
    built_so_file_path = self._build_impl()
  File "/Users/q/phd/mujoco-py/mujoco_py/builder.py", line 335, in _build_impl
    so_file_path = super()._build_impl()
  File "/Users/q/phd/mujoco-py/mujoco_py/builder.py", line 244, in _build_impl
    dist.run_commands()
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.7/lib/python3.7/distutils/dist.py", line 966, in run_commands
    self.run_command(cmd)
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.7/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/Users/q/mujoco_env/lib/python3.7/site-packages/Cython-3.0a4-py3.7.egg/Cython/Distutils/old_build_ext.py", line 186, in run
    _build_ext.build_ext.run(self)
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.7/lib/python3.7/distutils/command/build_ext.py", line 340, in run
    self.build_extensions()
  File "/Users/q/phd/mujoco-py/mujoco_py/builder.py", line 144, in build_extensions
    build_ext.build_extensions(self)
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.7/lib/python3.7/distutils/command/build_ext.py", line 449, in build_extensions
    self._build_extensions_serial()
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.7/lib/python3.7/distutils/command/build_ext.py", line 474, in _build_extensions_serial
    self.build_extension(ext)
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.7/lib/python3.7/distutils/command/build_ext.py", line 534, in build_extension
    depends=ext.depends)
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.7/lib/python3.7/distutils/ccompiler.py", line 574, in compile
    self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.7/lib/python3.7/distutils/unixccompiler.py", line 120, in _compile
    raise CompileError(msg)
distutils.errors.CompileError: command '/usr/local/bin/gcc-9' failed with exit status 1

I have also tried gcc@6, gcc@7, gcc@8

OS: OSX (Catalina the apple Vista)
python3.7
virtualenv

LXYYY, 2020-05-18:
> cd into mujoco-py then
> python3 setup.py install
> python
> 
> > > > import mujoco_py
> > > > Compiling /Users/q/phd/mujoco-py/mujoco_py/cymj.pyx because it depends on /Users/q/mujoco_env/lib/python3.7/site-packages/Cython-3.0a4-py3.7.egg/Cython/Includes/numpy/**init**.pxd.
> > > > [1/1] Cythonizing /Users/q/phd/mujoco-py/mujoco_py/cymj.pyx
> > > > warning: mujoco_py/generated/wrappers.pxi:4:0: Dotted filenames ('../pxd/mujoco.pxd') are deprecated. Please use the normal Python package directory layout.
> > > > running build_ext
> > > > building 'mujoco_py.cymj' extension
> > > > /usr/local/bin/gcc-9 -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -iwithsysroot/System/Library/Frameworks/System.framework/PrivateHeaders -iwithsysroot/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.7/Headers -DONMAC -Imujoco_py -I/Users/q/phd/mujoco-py/mujoco_py -I/Users/q/.mujoco/mujoco200/include -I/Users/q/mujoco_env/lib/python3.7/site-packages/numpy-1.18.4-py3.7-macosx-10.14-x86_64.egg/numpy/core/include -I/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.7/include/python3.7m -c /Users/q/phd/mujoco-py/mujoco_py/cymj.c -o /Users/q/phd/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.9_37_macextensionbuilder/temp.macosx-10.14-x86_64-3.7/Users/q/phd/mujoco-py/mujoco_py/cymj.o -fopenmp -w
> > > > gcc-9: error: unrecognized command line option '-iwithsysroot/System/Library/Frameworks/System.framework/PrivateHeaders'
> > > > gcc-9: error: unrecognized command line option '-iwithsysroot/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.7/Headers'
> > > > Traceback (most recent call last):
> > > > File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.7/lib/python3.7/distutils/unixccompiler.py", line 118, in _compile
> > > > extra_postargs)
> > > > File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.7/lib/python3.7/distutils/ccompiler.py", line 909, in spawn
> > > > spawn(cmd, dry_run=self.dry_run)
> > > > File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.7/lib/python3.7/distutils/spawn.py", line 36, in spawn
> > > > _spawn_posix(cmd, search_path, dry_run=dry_run)
> > > > File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.7/lib/python3.7/distutils/spawn.py", line 159, in _spawn_posix
> > > > % (cmd, exit_status))
> > > > distutils.errors.DistutilsExecError: command '/usr/local/bin/gcc-9' failed with exit status 1
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
> File "", line 1, in
> File "/Users/q/phd/mujoco-py/mujoco_py/**init**.py", line 3, in
> from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
> File "/Users/q/phd/mujoco-py/mujoco_py/builder.py", line 510, in
> cymj = load_cython_ext(mujoco_path)
> File "/Users/q/phd/mujoco-py/mujoco_py/builder.py", line 105, in load_cython_ext
> cext_so_path = builder.build()
> File "/Users/q/phd/mujoco-py/mujoco_py/builder.py", line 221, in build
> built_so_file_path = self._build_impl()
> File "/Users/q/phd/mujoco-py/mujoco_py/builder.py", line 335, in _build_impl
> so_file_path = super()._build_impl()
> File "/Users/q/phd/mujoco-py/mujoco_py/builder.py", line 244, in _build_impl
> dist.run_commands()
> File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.7/lib/python3.7/distutils/dist.py", line 966, in run_commands
> self.run_command(cmd)
> File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.7/lib/python3.7/distutils/dist.py", line 985, in run_command
> cmd_obj.run()
> File "/Users/q/mujoco_env/lib/python3.7/site-packages/Cython-3.0a4-py3.7.egg/Cython/Distutils/old_build_ext.py", line 186, in run
> _build_ext.build_ext.run(self)
> File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.7/lib/python3.7/distutils/command/build_ext.py", line 340, in run
> self.build_extensions()
> File "/Users/q/phd/mujoco-py/mujoco_py/builder.py", line 144, in build_extensions
> build_ext.build_extensions(self)
> File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.7/lib/python3.7/distutils/command/build_ext.py", line 449, in build_extensions
> self._build_extensions_serial()
> File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.7/lib/python3.7/distutils/command/build_ext.py", line 474, in _build_extensions_serial
> self.build_extension(ext)
> File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.7/lib/python3.7/distutils/command/build_ext.py", line 534, in build_extension
> depends=ext.depends)
> File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.7/lib/python3.7/distutils/ccompiler.py", line 574, in compile
> self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
> File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.7/lib/python3.7/distutils/unixccompiler.py", line 120, in _compile
> raise CompileError(msg)
[... cut ...]
# Add this to your .bashrc/.zshrc:
export PATH="/usr/local/opt/llvm/bin:$PATH"

export CC="/usr/local/opt/llvm/bin/clang"
export CXX="/usr/local/opt/llvm/bin/clang++"
export CXX11="/usr/local/opt/llvm/bin/clang++"
export CXX14="/usr/local/opt/llvm/bin/clang++"
export CXX17="/usr/local/opt/llvm/bin/clang++"
export CXX1X="/usr/local/opt/llvm/bin/clang++"

export LDFLAGS="-L/usr/local/opt/llvm/lib"
export CPPFLAGS="-I/usr/local/opt/llvm/include"
`

just like the troubleshooting said in README. clang is what you should use.

ForeverAT, 2023-10-04:
I get this, Do you have any clues? @LXYYY 
distutils.errors.CompileError: command '/usr/local/opt/llvm/bin/clang' failed with exit code 1