openai/mujoco-py #56: mujoco1.5.1.0 cant be installed on win10 64  with python 3.5.2 for the following reasons

taosheng1993, 2017-07-03:
cl : Command line warning D9025 : overriding '/W3' with '/w'
    cl : Command line warning D9002 : ignoring unknown option '-fopenmp'
    cymj.c
    c:\users\yt\appdata\local\temp\pip-build-st333rsf\mujoco-py\mujoco_py\cymj.c(466): fatal error C1083: Cannot open include file: 'mjmodel.h': No such file or directory
    error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio 14.0\\VC\\BIN\\x86_amd64\\cl.exe' failed with exit status 2

    ----------------------------------------
Command "c:\users\yt\appdata\local\programs\python\python35\python.exe -u -c "import setuptools, tokenize;__file__='C:\\Users\\YT\\AppData\\Local\\Temp\\pip-build-st333rsf\\mujoco-py\\setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record C:\Users\YT\AppData\Local\Temp\pip-w5zy5wak-record\install-record.txt --single-version-externally-managed --compile" failed with error code 1 in C:\Users\YT\AppData\Local\Temp\pip-build-st333rsf\mujoco-py\ 

@jonasschneider @welinder @avital @luddep @ @ @machinaut @

episodeyang, 2017-07-03:
@taosheng1993 There are some instructions in #47. My solution there is for `linux`, but some of it might help you resolve the issue.

https://github.com/openai/mujoco-py/issues/47#issuecomment-312551946

machinaut, 2017-07-06:
On windows, I use anaconda to create a python 3.5 environment, then "pip install" from there works.

I'm using windows 10 64-bit, which is pretty similar to your setup.  I think the big difference is using anaconda.

I'm not sure how your python environment manages the need for a compiler, but the first recommendation is try with an anaconda[1] environment.

[1] https://www.continuum.io/downloads#windows (download the 3.6 version but then create a 3.5 environment)

GoingMyWay, 2017-07-17:
Try to install with 
```
git clone https://github.com/openai/mujoco-py.git
cd mujoco-py &&  python setup.py install
```

https://github.com/openai/mujoco-py/issues/47

ViktorM, 2017-07-23:
@machinaut I used Anaconda and tried pip install too, but still see the `fatal error C1083: Cannot open include file: 'mjmodel.h': No such file or directory` error.

Will be thankful for any advice and help.

sushrutt12, 2017-12-07:
Check my comment [here](https://github.com/openai/mujoco-py/issues/99), I've been experimenting it for over 24 hours now, but finally worked out.