openai/mujoco-py #57: Segmentation fault in MjViewer()

heiko0403, 2017-07-03:
I installed version 1.50.1.0 on Ubuntu 14.04 using Python 3.5.

Houever, when I tried examples/ programs, I got
running build_ext
Creating window glfw
Segmentation fault.

So I tried examples/disco_fetch.py line by line on IPython, but similarly, I got segmentation fault in "viewer = MjViewer(sim)" line.

I think this is the trouble of mujoco-py.MjViewer().
I'm investigating the cause, but I share the problem for the moment.

hmishra2250, 2017-07-03:
Yeah, that's probably a bug in Linux build and I guess they are working on it. You can track the progress [here](https://github.com/openai/mujoco-py/issues/44)

touching-foots-huskie, 2017-07-07:
I also meet with this problem.
And the same time I can not pass the pytest for the following reasons:
 running build_ext
  [  [V: not found
  sh: 2: [
            : not found
  [  [V: not found
  sh: 2: [
            : not found
  [  [V: not found
  sh: 2: Syntax error: "(" unexpected
  [  [V: not found
  sh: 2: Syntax error: "(" unexpected
  ERROR: Invalid activation key

I don't why is it. where should the mjkey.txt be put? Mujoco and glfw work fine, and there is a segmentation fault in mujoco_py.MjViewer()

hmishra2250, 2017-07-07:
@touching-foots-huskie  I encountered the same problem. It's either because you are using older version of Mujoco, or the key that you are using is for different PC or OS. Remember that you need to register the _Mujoco and your PC_ in order to use it.
So if you use a key from other OS (even from same PC), it wont work. 
TL;DR
**Request for another key for different OS or PC (if in case) or upgrade your Mujoco to 1.5**

touching-foots-huskie, 2017-07-07:
I don't think it is the problem. Because the mujoco runs well in c++. This only happens on mujoco_py

taochenshh, 2017-10-12:
Any update on this problem? I am using ubuntu 14.04 and anaconda2 with python 3.5. Meet the same segmentation fault problem.
```python
GLFW error (code %d): %s 65542 b'EGL: Failed to get EGL display: Success'
Creating window glfw
Segmentation fault (core dumped)
```

aseembits93, 2018-03-01:
Hi, i tried everything but not able to get mujoco_py to work. i am using ubuntu 14.04 with a python3.5 virtualenv.

lapp0, 2022-02-20:
I also get this error. Here are some segfault logs and the python traceback.

```
catchsegv load-robo-env
hi0
Creating window glfw
Fatal Python error: Segmentation fault

Current thread 0x00007fa421372740 (most recent call first):
  File "/nix/store/db2y9n2a80nqddlhwnbz400qhlrf4g5f-python3.9-glfw-2.5.0/lib/python3.9/site-packages/glfw/__init__.py", line 1252 in create_window
  File "/nix/store/g2l2p80nq6sy6hn99ngk5y3j49ngbhlk-python3.9-mujoco-py-2.1.12.14/lib/python3.9/site-packages/mujoco_py/mjviewer.py", line 28 in __init__
  File "/nix/store/g2l2p80nq6sy6hn99ngk5y3j49ngbhlk-python3.9-mujoco-py-2.1.12.14/lib/python3.9/site-packages/mujoco_py/mjviewer.py", line 137 in __init__
  File "/nix/store/pa2w49w4l1nangbfzh3a0k1shvf1lgpw-python3.9-robosuite-1.3.0/lib/python3.9/site-packages/robosuite/renderers/mujoco/mujoco_py_renderer.py", line 59 in __init__
  File "/nix/store/pa2w49w4l1nangbfzh3a0k1shvf1lgpw-python3.9-robosuite-1.3.0/lib/python3.9/site-packages/robosuite/environments/base.py", line 294 in _reset_internal
  File "/nix/store/pa2w49w4l1nangbfzh3a0k1shvf1lgpw-python3.9-robosuite-1.3.0/lib/python3.9/site-packages/robosuite/environments/robot_env.py", line 510 in _reset_internal
  File "/nix/store/pa2w49w4l1nangbfzh3a0k1shvf1lgpw-python3.9-robosuite-1.3.0/lib/python3.9/site-packages/robosuite/environments/manipulation/stack.py", line 394 in _reset_internal
  File "/nix/store/pa2w49w4l1nangbfzh3a0k1shvf1lgpw-python3.9-robosuite-1.3.0/lib/python3.9/site-packages/robosuite/environments/base.py", line 143 in __init__
  File "/nix/store/pa2w49w4l1nangbfzh3a0k1shvf1lgpw-python3.9-robosuite-1.3.0/lib/python3.9/site-packages/robosuite/environments/robot_env.py", line 214 in __init__
  File "/nix/store/pa2w49w4l1nangbfzh3a0k1shvf1lgpw-python3.9-robosuite-1.3.0/lib/python3.9/site-packages/robosuite/environments/manipulation/manipulation_env.py", line 162 in __init__
  File "/nix/store/pa2w49w4l1nangbfzh3a0k1shvf1lgpw-python3.9-robosuite-1.3.0/lib/python3.9/site-packages/robosuite/environments/manipulation/stack.py", line 183 in __init__
  File "/nix/store/pa2w49w4l1nangbfzh3a0k1shvf1lgpw-python3.9-robosuite-1.3.0/lib/python3.9/site-packages/robosuite/environments/base.py", line 38 in make
  File "/nix/store/27p19qw5yp1g53sjk8dd56f34jlby5h0-python3.9-load-robo-env-1.0.0/lib/python3.9/site-packages/environments/load_environment.py", line 8 in get_simple_env
  File "/nix/store/27p19qw5yp1g53sjk8dd56f34jlby5h0-python3.9-load-robo-env-1.0.0/lib/python3.9/site-packages/environments/load_environment.py", line 33 in main
  File "/nix/store/27p19qw5yp1g53sjk8dd56f34jlby5h0-python3.9-load-robo-env-1.0.0/bin/.load-robo-env-wrapped", line 9 in <module>
*** signal wi
Register dump:

 RAX: 0000000000000000   RBX: 000000000000000b   RCX: 00007fa421777d9a
 RDX: 0000000000000000   RSI: 00000000016b6030   RDI: 0000000000000002
 RBP: 0000000003b9f940   R8 : 0000000000000000   R9 : 00000000016b6030
 R10: 0000000000000008   R11: 0000000000000246   R12: 0000000000000000
 R13: 0000000003c12a70   R14: 0000000000000001   R15: 0000000003c13540
 RSP: 00000000016b60a8

 RIP: 00007fa421777d9a   EFLAGS: 00000246

 CS: 0033   FS: 0000   GS: 0000

 Trap: 0000000e   Error: 00000004   OldMask: 00000000   CR2: 00000010

 FPUCW: 0000037f   FPUSW: 00000000   TAG: 00000000
 RIP: 00000000   RDP: 00000000

 ST(0) 0000 0000000000000000   ST(1) 0000 0000000000000000
 ST(2) 0000 0000000000000000   ST(3) 0000 0000000000000000
 ST(4) 0000 0000000000000000   ST(5) 0000 0000000000000000
 ST(6) 0000 0000000000000000   ST(7) 0000 0000000000000000
 mxcsr: 1f80
 XMM0:  00000000000000000000000000000000 XMM1:  00000000000000000000000000000000
 XMM2:  00000000000000000000000000000000 XMM3:  00000000000000000000000000000000
 XMM4:  00000000000000000000000000000000 XMM5:  00000000000000000000000000000000
 XMM6:  00000000000000000000000000000000 XMM7:  00000000000000000000000000000000
 XMM8:  00000000000000000000000000000000 XMM9:  00000000000000000000000000000000
 XMM10: 00000000000000000000000000000000 XMM11: 00000000000000000000000000000000
 XMM12: 00000000000000000000000000000000 XMM13: 00000000000000000000000000000000
 XMM14: 00000000000000000000000000000000 XMM15: 00000000000000000000000000000000

Backtrace:
/nix/store/jcb7fny2k03pfbdqk1hcnh12bxgax6vf-glibc-2.33-108/lib/libpthread.so.0(raise+0xca)[0x7fa421777d9a]
/nix/store/jcb7fny2k03pfbdqk1hcnh12bxgax6vf-glibc-2.33-108/lib/libpthread.so.0(+0x12ee0)[0x7fa421777ee0]
/nix/store/psmsm40h58c6awgc8lsfk2vk3kqb7pi3-llvm-13.0.0-lib/lib/libLLVM-13.so(_ZN4llvm16TargetPassConfig7addPassEPNS_4PassEb+0x1a)[0x7fa4045143ea]
/nix/store/psmsm40h58c6awgc8lsfk2vk3kqb7pi3-llvm-13.0.0-lib/lib/libLLVM-13.so(_ZN4llvm16TargetPassConfig31addRegAssignAndRewriteOptimizedEv+0xc1)[0x7fa404515051]
/nix/store/psmsm40h58c6awgc8lsfk2vk3kqb7pi3-llvm-13.0.0-lib/lib/libLLVM-13.so(_ZN4llvm16TargetPassConfig20addOptimizedRegAllocEv+0x260)[0x7fa404515d20]
/nix/store/psmsm40h58c6awgc8lsfk2vk3kqb7pi3-llvm-13.0.0-lib/lib/libLLVM-13.so(_ZN4llvm16TargetPassConfig16addMachinePassesEv+0x134)[0x7fa404516094]
/nix/store/psmsm40h58c6awgc8lsfk2vk3kqb7pi3-llvm-13.0.0-lib/lib/libLLVM-13.so(+0x10c4153)[0x7fa404223153]
/nix/store/psmsm40h58c6awgc8lsfk2vk3kqb7pi3-llvm-13.0.0-lib/lib/libLLVM-13.so(_ZN4llvm17LLVMTargetMachine17addPassesToEmitMCERNS_6legacy15PassManagerBaseERPNS_9MCContextERNS_17raw_pwrite_streamEb+0x57)[0x7fa404224967]
[... cut ...]
/nix/store/x24kxyqwqg2ln8kh9ky342kdcmhbng3h-python3-3.9.9/lib/libpython3.9.so.1.0(PyRun_SimpleFileExFlags+0x19d)[0x7fa421a0be3d]
/nix/store/x24kxyqwqg2ln8kh9ky342kdcmhbng3h-python3-3.9.9/lib/libpython3.9.so.1.0(Py_RunMain+0x87b)[0x7fa421a0d17b]
/nix/store/x24kxyqwqg2ln8kh9ky342kdcmhbng3h-python3-3.9.9/lib/libpython3.9.so.1.0(Py_BytesMain+0x60)[0x7fa421a14510]
/nix/store/jcb7fny2k03pfbdqk1hcnh12bxgax6vf-glibc-2.33-108/lib/libc.so.6(__libc_start_main+0xd0)[0x7fa42139e790]
/nix/store/x24kxyqwqg2ln8kh9ky342kdcmhbng3h-python3-3.9.9/bin/python3.9(_start+0x2a)[0x40107a]
```