openai/mujoco-py #60: Fedora support

rhofour, 2017-07-08:
I can't figure out how to get this to build on Fedora. I looked at the dockerfile and installed the closest packages I can find, then anything that looked remotely related, yet I'm still getting undefined symbols when I try and build:

/home/rofer/.mujoco/mjpro150/bin/libmujoco150.so: undefined symbol: glNormal3f

welinder, 2017-07-11:
That looks like you're missing some OpenGL library. Make sure you can compile the samples that come with Mujoco: `mjpro150/sample/makefile`

rhofour, 2017-07-15:
I have no problem compiling the samples. I tried running both test and simulate and they worked without issues. I also have what feels like every OpenGL-related lib in the Fedora standard repo installed.

machinaut, 2017-07-15:
Is there a dockerfile for fedora which reproduces this?  I can try putting one together but I haven't used fedora in docker before.

rhofour, 2017-07-15:
I haven't really used Docker before, but I could try and make a Fedora dockerfile.

machinaut, 2017-07-15:
What would help me out is what rpm/yum packages are needed to build/run mujoco -- I'm only familiar with the debian/ubuntu versions.

machinaut, 2017-07-16:
With #67 installing on Fedora (in docker) works for me.

Gist with dockerfile and makefile to run it: https://gist.github.com/machinaut/f5cf076b33752334a5618ed906baf23a

machinaut, 2017-07-16:
@rhofour if you get a chance to try out that PR #67 let me know if it doesn't fix your issue.

In my Fedora-docker I see the same error ("undefined symbol: glNormal3f") without the PR, and with it I'm able to pip install.

PR #67 (2017-07-16): Add libGL to linux-cpu libraries
Fixes build on Fedora, strangely enough.  For some reason the osmesa/glew/glfw isn't enough on Fedora, but work on debian/ubuntu.

Fixes #60
closed 2017-07-16 by welinder