openai/mujoco-py #61: Error

gautam1858, 2017-07-11:
Hi I am getting the following error with python version 3.5.2 and gcc 4.2.1

`mujoco-py/mujoco_py/cymj.c:476:10: fatal error: mjmodel.h: No such file or directory
     #include "mjmodel.h"
              ^~~~~~~~~~~
    compilation terminated.
    error: command '/usr/local/bin/gcc-7' failed with exit status 1
    
    ----------------------------------------
  Rolling back uninstall of mujoco-py
Command "/Library/Frameworks/Python.framework/Versions/3.5/bin/python3.5 -u -c "import setuptools, tokenize;__file__='/private/var/folders/pc/5yb_kn9d69v4vgbrrvjrttww0000gn/T/pip-build-qkpzb1d_/mujoco-py/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /var/folders/pc/5yb_kn9d69v4vgbrrvjrttww0000gn/T/pip-y_01snts-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /private/var/folders/pc/5yb_kn9d69v4vgbrrvjrttww0000gn/T/pip-build-qkpzb1d_/mujoco-py/`

welinder, 2017-07-11:
Where have you installed Mujoco to? Currently we require it to be located in `~/.mujoco/mjpro150`.

gautam1858, 2017-07-11:
Hi, I have Mujoco in /Users/neutrino/mujoco/mjpro150

welinder, 2017-07-11:
It should be located in `/Users/neutrino/.mujoco/mjpro150`. Note the leading dot. Closing this for now.

closed 2017-07-11 by welinder

calvinweb, 2019-03-23:
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.6/site-packages/mujoco_py-2.0.2.0-py3.6.egg/mujoco_py/__init__.py", line 3, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/usr/local/lib/python3.6/site-packages/mujoco_py-2.0.2.0-py3.6.egg/mujoco_py/builder.py", line 503, in <module>
    cymj = load_cython_ext(mujoco_path)
  File "/usr/local/lib/python3.6/site-packages/mujoco_py-2.0.2.0-py3.6.egg/mujoco_py/builder.py", line 89, in load_cython_ext
    with LockFile(lockpath):
  File "/usr/local/lib/python3.6/site-packages/lockfile/__init__.py", line 197, in __enter__
    self.acquire()
  File "/usr/local/lib/python3.6/site-packages/lockfile/linklockfile.py", line 21, in acquire
    raise LockFailed("failed to create %s" % self.unique_name)
lockfile.LockFailed: failed to create /usr/local/lib/python3.6/site-packages/mujoco_py-2.0.2.0-py3.6.egg/mujoco_py/generated/vincentde-6.local-8be85c0.37120-2558176354571308353