openai/mujoco-py #64: problem with load_model_from_path

gautam1858, 2017-07-12:
Hi whenever I run a program having `load_model_from_path`, I am getting the following error
Example :   `python3 disco_fetch.py` 
```
Traceback (most recent call last):
  File "disco_fetch.py", line 9, in <module>
    model = load_model_from_path("xmls/fetch/main.xml")
  File "mujoco_py/cymj.pyx", line 113, in mujoco_py.cymj.load_model_from_path (/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/mujoco_py-1.50.1.6-py3.5.egg/mujoco_py/cymj.c:122039)
Exception: Failed to load XML file: xmls/fetch/main.xml. mj_loadXML error: b'XML parse error at line 0, column 0:\nFailed to open file\n'
```
The program works fine for load_model_from_xml()

abhigenie92, 2017-11-12:
Hi @gautam1858 , were you able to fix this issue. I was stuck at the same.

gautam1858, 2017-11-14:
nope I was not able to

abhigenie92, 2017-11-14:
@gautam1858 Can you repopen the issue then?

SiddhardhaSaran, 2018-02-05:
That's because the path is different.Manually edit the disco_fetch file to point to the xmls/fetch/main.py file.
That means changing the line
model = load_model_from_path("give the direct path to xmls/fetch/main.py file here") in disco_fetch.py.
You may also have to change the other python files in the examples folder to run them.

lennonbest, 2018-03-29:
1.download mujoco-py:
git clone https://github.com/openai/mujoco-py.git

2.install
cd mujoco-py

sudo apt-get update -q

sudo DEBIAN_FRONTEND=noninteractive apt-get install -y curl git libgl1-mesa-dev libgl1-mesa-glx libglew-dev libosmesa6-dev python3-pip python3-numpy python3-scipy net-tools unzip vim wget xpra xserver-xorg-dev

sudo apt-get clean

rm -rf /var/lib/apt/lists/*

cp requirements.txt requirements.dev.txt mujoco_py

cd mujoco_py

sudo pip3 install -r requirements.txt

sudo pip3 install -r requirements.dev.txt

cd ..

sudo pip3 install -e .

cyrilzakka, 2018-11-19:
You can also refer to the following [issue](https://github.com/openai/mujoco-py/issues/168) for other solutions. Also keep in mind

machinaut, 2018-11-28:
@cyrilzakka does it also fail if you provide an absolute path to the XML file?