openai/mujoco-py #66: Failed building wheel for mujoco-py

pat-coady, 2017-07-13:
Here is the full log:

`pip3 install -U 'mujoco-py<1.50.2,>=1.50.1'
Collecting mujoco-py<1.50.2,>=1.50.1
  Using cached mujoco-py-1.50.1.7.tar.gz
Requirement already up-to-date: glfw>=1.4.0 in /usr/local/lib/python3.5/dist-packages (from mujoco-py<1.50.2,>=1.50.1)
Requirement already up-to-date: numpy>=1.11 in ./.local/lib/python3.5/site-packages (from mujoco-py<1.50.2,>=1.50.1)
Requirement already up-to-date: Cython>=0.25.2 in /usr/local/lib/python3.5/dist-packages (from mujoco-py<1.50.2,>=1.50.1)
Requirement already up-to-date: imageio>=2.1.2 in ./.local/lib/python3.5/site-packages (from mujoco-py<1.50.2,>=1.50.1)
Requirement already up-to-date: pillow in ./.local/lib/python3.5/site-packages (from imageio>=2.1.2->mujoco-py<1.50.2,>=1.50.1)
Requirement already up-to-date: olefile in ./.local/lib/python3.5/site-packages (from pillow->imageio>=2.1.2->mujoco-py<1.50.2,>=1.50.1)
Building wheels for collected packages: mujoco-py
  Running setup.py bdist_wheel for mujoco-py ... -
error
  Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-lfrvbntl/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmpfe_jl9xipip-wheel- --python-tag cp35:
  running bdist_wheel
  running build
  Compiling /tmp/pip-build-lfrvbntl/mujoco-py/mujoco_py/cymj.pyx because it changed.
  [1/1] Cythonizing /tmp/pip-build-lfrvbntl/mujoco-py/mujoco_py/cymj.pyx
  running build_ext
  building 'mujoco_py.cymj' extension
  creating /tmp/pip-build-lfrvbntl/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder
  creating /tmp/pip-build-lfrvbntl/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5
  creating /tmp/pip-build-lfrvbntl/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp
  creating /tmp/pip-build-lfrvbntl/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build-lfrvbntl
  creating /tmp/pip-build-lfrvbntl/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build-lfrvbntl/mujoco-py
  creating /tmp/pip-build-lfrvbntl/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build-lfrvbntl/mujoco-py/mujoco_py
  creating /tmp/pip-build-lfrvbntl/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build-lfrvbntl/mujoco-py/mujoco_py/gl
  x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/tmp/pip-build-lfrvbntl/mujoco-py/mujoco_py -I/home/pco/.mujoco/mjpro150/include -I/home/pco/.local/lib/python3.5/site-packages/numpy/core/include -I/usr/include/python3.5m -c /tmp/pip-build-lfrvbntl/mujoco-py/mujoco_py/cymj.c -o /tmp/pip-build-lfrvbntl/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build-lfrvbntl/mujoco-py/mujoco_py/cymj.o -fopenmp -w
  x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/tmp/pip-build-lfrvbntl/mujoco-py/mujoco_py -I/home/pco/.mujoco/mjpro150/include -I/home/pco/.local/lib/python3.5/site-packages/numpy/core/include -I/usr/include/python3.5m -c /tmp/pip-build-lfrvbntl/mujoco-py/mujoco_py/gl/osmesashim.c -o /tmp/pip-build-lfrvbntl/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build-lfrvbntl/mujoco-py/mujoco_py/gl/osmesashim.o -fopenmp -w
  creating /tmp/pip-build-lfrvbntl/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/lib.linux-x86_64-3.5
  creating /tmp/pip-build-lfrvbntl/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/lib.linux-x86_64-3.5/mujoco_py
  x86_64-linux-gnu-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 /tmp/pip-build-lfrvbntl/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build-lfrvbntl/mujoco-py/mujoco_py/cymj.o /tmp/pip-build-lfrvbntl/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build-lfrvbntl/mujoco-py/mujoco_py/gl/osmesashim.o -L/home/pco/.mujoco/mjpro150/bin -Wl,--enable-new-dtags,-R/home/pco/.mujoco/mjpro150/bin -lmujoco150 -lglewosmesa -lOSMesa -o /tmp/pip-build-lfrvbntl/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/lib.linux-x86_64-3.5/mujoco_py/cymj.cpython-35m-x86_64-linux-gnu.so -fopenmp
  [  [V: not found
  sh: 2: [
            : not found
  [  [V: not found
  sh: 2: [
            : not found
  [  [V: not found
  sh: 2: Syntax error: "(" unexpected
  [  [V: not found
  sh: 2: Syntax error: "(" unexpected
  ERROR: Invalid activation key
  
  Press Enter to exit ...
  
  ----------------------------------------
  Failed building wheel for mujoco-py
  Running setup.py clean for mujoco-py
Failed to build mujoco-py
Installing collected packages: mujoco-py
  Running setup.py install for mujoco-py ... \
error
    Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-lfrvbntl/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-yeem0tw6-record/install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    [    [V: not found
    sh: 2: [
                : not found
    [    [V: not found
    sh: 2: [
                : not found
    [    [V: not found
    sh: 2: Syntax error: "(" unexpected
    [    [V: not found
    sh: 2: Syntax error: "(" unexpected
    ERROR: Invalid activation key
    
    Press Enter to exit ...
    
    ----------------------------------------
Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-lfrvbntl/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-yeem0tw6-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-lfrvbntl/mujoco-py/
`
I did note "ERROR: Invalid activation key", but I am able to run ./simulate from bin directory with no problem. Also, this error comes after the strange binary characters.

cxy1997, 2017-07-15:
I've got similar problems. When I execute `pip3 install -U 'mujoco-py<1.50.2,>=1.50.1'`, it gets stuck while building wheels (12 hours and no response), and does not even raise an error.
Here is the log:
> $ pip3 install -U 'mujoco-py<1.50.2,>=1.50.1'
> Collecting mujoco-py<1.50.2,>=1.50.1
>   Downloading mujoco-py-1.50.1.7.tar.gz (100kB)
>     100% |████████████████████████████████| 102kB 145kB/s 
> Collecting glfw>=1.4.0 (from mujoco-py<1.50.2,>=1.50.1)
>   Downloading glfw-1.4.0.tar.gz
> Collecting numpy>=1.11 (from mujoco-py<1.50.2,>=1.50.1)
>   Downloading numpy-1.13.1-cp35-cp35m-manylinux1_x86_64.whl (16.9MB)
>     100% |████████████████████████████████| 16.9MB 111kB/s 
> Collecting Cython>=0.25.2 (from mujoco-py<1.50.2,>=1.50.1)
>   Downloading Cython-0.25.2-cp35-cp35m-manylinux1_x86_64.whl (7.0MB)
>     100% |████████████████████████████████| 7.0MB 262kB/s 
> Collecting imageio>=2.1.2 (from mujoco-py<1.50.2,>=1.50.1)
>   Downloading imageio-2.2.0.tar.gz (3.3MB)
>     100% |████████████████████████████████| 3.3MB 569kB/s 
> Collecting pillow (from imageio>=2.1.2->mujoco-py<1.50.2,>=1.50.1)
>   Downloading Pillow-4.2.1-cp35-cp35m-manylinux1_x86_64.whl (5.8MB)
>     100% |████████████████████████████████| 5.8MB 331kB/s 
> Collecting olefile (from pillow->imageio>=2.1.2->mujoco-py<1.50.2,>=1.50.1)
>   Downloading olefile-0.44.zip (74kB)
>     100% |████████████████████████████████| 81kB 7.4MB/s 
> Building wheels for collected packages: mujoco-py, glfw, imageio, olefile
>   Running setup.py bdist_wheel for mujoco-py ... -

The system is **Ubuntu 16.04 LST** with **Python 3.5.2**, and the _~/.mujoco_ directory is prepared with a valid _mjkey.txt_. Also, the _mujoco150/bin/simulate_ works well. 
I have built the mujoco-py with docker, but when it comes to pytest, there is always an error:

> ~/mujoco-py$ sudo make test
> cp ~/.mujoco/mjkey.txt mjkey.txt
> docker build -t mujoco_py . || rm mjkey.txt
> Sending build context to Docker daemon 7.096 MB
> Step 1 : FROM nvidia/cuda:8.0-devel-ubuntu16.04
>  ---> 0877f562ab17
> Step 2 : RUN apt-get update -q     && DEBIAN_FRONTEND=noninteractive apt-get install -y     curl     git     libgl1-mesa-dev     libgl1-mesa-glx     libosmesa6-dev     python3-pip     python3-numpy     python3-scipy     unzip     vim     wget     xpra     xserver-xorg-dev     && apt-get clean     && rm -rf /var/lib/apt/lists/*
>  ---> Using cache
>  ---> 2b4749a50ff7
> Step 3 : RUN curl -o /usr/local/bin/patchelf https://s3-us-west-2.amazonaws.com/openai-sci-artifacts/manual-builds/patchelf_0.9_amd64.elf     && chmod +x /usr/local/bin/patchelf
>  ---> Using cache
>  ---> b7c2fd58cda1
> Step 4 : ENV LANG C.UTF-8
>  ---> Using cache
>  ---> 7b0f11ab4e49
> Step 5 : RUN mkdir -p /root/.mujoco     && wget https://www.roboti.us/download/mjpro150_linux.zip -O mujoco.zip     && unzip mujoco.zip -d /root/.mujoco     && rm mujoco.zip
>  ---> Using cache
>  ---> 4f4df098548e
> Step 6 : COPY ./mjkey.txt /root/.mujoco/
>  ---> Using cache
>  ---> f5c9f36b9390
> Step 7 : ENV LD_LIBRARY_PATH /root/.mujoco/mjpro150/bin:$LD_LIBRARY_PATH
>  ---> Using cache
>  ---> 2c9d2da25b5a
> Step 8 : COPY vendor/Xdummy /usr/local/bin/Xdummy
>  ---> Using cache
>  ---> 5247ede6f4ef
> Step 9 : RUN chmod +x /usr/local/bin/Xdummy
>  ---> Using cache
>  ---> 3f7af51cfb0f
> Step 10 : COPY ./vendor/10_nvidia.json /usr/share/glvnd/egl_vendor.d/10_nvidia.json
>  ---> Using cache
>  ---> b8f82a136b2f
> Step 11 : WORKDIR /mujoco_py
>  ---> Using cache
>  ---> 90e0644cb116
> Step 12 : COPY ./requirements.txt /mujoco_py/
>  ---> Using cache
>  ---> 91f23bf029af
> Step 13 : COPY ./requirements.dev.txt /mujoco_py/
>  ---> Using cache
>  ---> ca46cf983eef
> Step 14 : RUN pip3 install -r requirements.txt
>  ---> Using cache
>  ---> e44ec11bd9e7
> Step 15 : RUN pip3 install -r requirements.dev.txt
>  ---> Using cache
>  ---> ac39aabbfa97
> Step 16 : ENTRYPOINT /mujoco_py/vendor/Xdummy-entrypoint
>  ---> Using cache
>  ---> 7a9d18179b08
> Step 17 : CMD pytest
>  ---> Using cache
>  ---> 1d9e218e3f51
> Step 18 : COPY . /mujoco_py
>  ---> Using cache
>  ---> fb31e6b4e59d
> Step 19 : RUN python3 setup.py install
>  ---> Using cache
>  ---> 0edb701324e1
> Successfully built 0edb701324e1
> \# run it interactive mode so we can abort with CTRL+C
> docker run --rm -i mujoco_py pytest
> ============================= test session starts ==============================
> platform linux -- Python 3.5.2, pytest-3.1.3, py-1.4.34, pluggy-0.4.0
> rootdir: /mujoco_py, inifile:
> plugins: instafail-0.3.0
> Makefile:26: recipe for target 'test' failed
> make: *** [test] Error 1

I think these might be the same problem.

machinaut, 2017-07-15:
Looks like there _might_ be a problem where `pip` is caching an older version of `mujoco-py` which satisfies your version requirements.

If you uninstall it and add `--no-cache` pip will try to download a more recent versions that meets the requirements, which I'd expect to have fixes for at least some of those issues.

machinaut, 2017-07-15:
@pat-coady can you verify that your activation key isn't expired and the file matches the one in mjpro150/bin (or whatever key ./simulate was using)?  Possibly a key for a different version of mujoco?

That's different than the error message when the key file can't be found, so my thought is it found the file but for some reason the key failed to validate (possibly for a different OS/machine, out of date/expired, for a different version of mujoco, or possibly for other reasons).

machinaut, 2017-07-15:
@cxy1997 I think your issue is a different thing -- would you mind filing a separate issue for that?

pat-coady, 2017-07-16:
@machinaut - The diff on the mjkey.txt files came up clean. The key is only a few days old and was issued for the machine I'm running on. I ended up installing mujoco-py ver0.5.7 (and mjpro131) as a workaround, and that is working fine.

Regardin @cxy1997 post - could be slightly related to problem I'm seeing. My install also froze exactly like they described. I gave it 10 minutes. But the spinning bar was frozen. When I pressed ENTER I got the rest of the log file that I cut and pasted above.

Thanks for looking into this.

cxy1997, 2017-07-16:
@machinaut and @pat-coady  - In fact I encountered another problem, and when dealing with it I figured out an alternative way to install _mujoco-py_. Here's the story.

After I tried to install it with `sudo make`, failed and then restarted _Ubuntu 16.04_, the keyboard and mouse just wouldn't work (only in Ubuntu). 

I found that some of the dependencies in [`Dockerfile`](https://github.com/openai/mujoco-py/blob/master/Dockerfile) and [`Makefile`](https://github.com/openai/mujoco-py/blob/master/Makefile) are unnecessary and may conflict with the drivers. So I manually installed it without using **docker**, it works well on a raw Ubuntu 16.04 (I couldb't fix the drivers and had to reinstall Ubuntu):
```
git clone https://github.com/openai/mujoco-py.git
cd mujoco-py
sudo apt-get update
sudo apt-get install libgl1-mesa-dev libgl1-mesa-glx libosmesa6-dev python3-pip python3-numpy python3-scipy
pip3 install -r requirements.txt
sudo python3 setup.py install
```

pat-coady, 2017-07-16:
@cxy1997 - I also lost my keyboard and mouse by blindly installing all the dependencies listed in docker file (my fault, not mujoco-py problem). Fortunately I was able to fix that without a complete OS reinstall. I tried your procedure and it seems like some progress. But I still can't get a working installation. I get to the point where it is "Cythonizing" things the first time I import mujoco-py. But then I go around and around with permission denied problems, then rerunning with sudo, then gcc errors. I started to flail again and gave up.

Fortunately as able to revert to 0.5.7 again and am running for now.

cxy1997, 2017-07-16:
@pat-coady - The 'Cythonizing' thing you mentioned is common when `mujoco_py` is imported for the first time. You can play with the examples with `sudo`, like this:
`sudo python3 examples/body_interaction.py`
Ather that, you should be able to `import mujoco_py` without any problem.

pat-coady, 2017-07-16:
@cxy1997 - Yes, I had gotten past the Cythonize (although I had to run 'sudo python' - otherwise permission problems). Then I ran into another hurdle (it is all a blur now). Then I finally went to train my agent and got a 'mujoco-py' not installed sort of error. Along with all of this, my jupyter notebook installation stopped working. Reinstalling Ubuntu now as too many things seem messed up at this point.

I'm going to wait and see if the straightforward pip approach gets repaired. And stick with 0.5.7 version for now.

pat-coady, 2017-07-16:
Ok, here is a new log file. Slightly different error now:

sudo -H pip3 install -U 'mujoco-py<1.50.2,>=1.50.1'
[sudo] password for pco: 
Collecting mujoco-py<1.50.2,>=1.50.1
  Using cached mujoco-py-1.50.1.7.tar.gz
Requirement already up-to-date: glfw>=1.4.0 in /usr/local/lib/python3.5/dist-packages (from mujoco-py<1.50.2,>=1.50.1)
Requirement already up-to-date: numpy>=1.11 in /usr/local/lib/python3.5/dist-packages (from mujoco-py<1.50.2,>=1.50.1)
Requirement already up-to-date: Cython>=0.25.2 in /usr/local/lib/python3.5/dist-packages (from mujoco-py<1.50.2,>=1.50.1)
Requirement already up-to-date: imageio>=2.1.2 in /usr/local/lib/python3.5/dist-packages (from mujoco-py<1.50.2,>=1.50.1)
Requirement already up-to-date: pillow in /usr/local/lib/python3.5/dist-packages (from imageio>=2.1.2->mujoco-py<1.50.2,>=1.50.1)
Requirement already up-to-date: olefile in /usr/local/lib/python3.5/dist-packages (from pillow->imageio>=2.1.2->mujoco-py<1.50.2,>=1.50.1)
Building wheels for collected packages: mujoco-py
  Running setup.py bdist_wheel for mujoco-py ... error
  Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-f9gjmsg2/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmp1btvwa7opip-wheel- --python-tag cp35:
  running bdist_wheel
  running build
  Compiling /tmp/pip-build-f9gjmsg2/mujoco-py/mujoco_py/cymj.pyx because it changed.
  [1/1] Cythonizing /tmp/pip-build-f9gjmsg2/mujoco-py/mujoco_py/cymj.pyx
  running build_ext
  building 'mujoco_py.cymj' extension
  creating /tmp/pip-build-f9gjmsg2/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder
  creating /tmp/pip-build-f9gjmsg2/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5
  creating /tmp/pip-build-f9gjmsg2/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp
  creating /tmp/pip-build-f9gjmsg2/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build-f9gjmsg2
  creating /tmp/pip-build-f9gjmsg2/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build-f9gjmsg2/mujoco-py
  creating /tmp/pip-build-f9gjmsg2/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build-f9gjmsg2/mujoco-py/mujoco_py
  creating /tmp/pip-build-f9gjmsg2/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build-f9gjmsg2/mujoco-py/mujoco_py/gl
  x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/tmp/pip-build-f9gjmsg2/mujoco-py/mujoco_py -I/root/.mujoco/mjpro150/include -I/usr/local/lib/python3.5/dist-packages/numpy/core/include -I/usr/include/python3.5m -c /tmp/pip-build-f9gjmsg2/mujoco-py/mujoco_py/cymj.c -o /tmp/pip-build-f9gjmsg2/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build-f9gjmsg2/mujoco-py/mujoco_py/cymj.o -fopenmp -w
  /tmp/pip-build-f9gjmsg2/mujoco-py/mujoco_py/cymj.c:471:21: fatal error: mjmodel.h: No such file or directory
  compilation terminated.
  error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
  
  ----------------------------------------
  Failed building wheel for mujoco-py
  Running setup.py clean for mujoco-py
Failed to build mujoco-py
Installing collected packages: mujoco-py
  Found existing installation: mujoco-py 0.5.7
    Uninstalling mujoco-py-0.5.7:
      Successfully uninstalled mujoco-py-0.5.7
  Running setup.py install for mujoco-py ... error
    Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-f9gjmsg2/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-yojvx9qt-record/install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    running build_ext
    building 'mujoco_py.cymj' extension
    x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/tmp/pip-build-f9gjmsg2/mujoco-py/mujoco_py -I/root/.mujoco/mjpro150/include -I/usr/local/lib/python3.5/dist-packages/numpy/core/include -I/usr/include/python3.5m -c /tmp/pip-build-f9gjmsg2/mujoco-py/mujoco_py/cymj.c -o /tmp/pip-build-f9gjmsg2/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build-f9gjmsg2/mujoco-py/mujoco_py/cymj.o -fopenmp -w
    /tmp/pip-build-f9gjmsg2/mujoco-py/mujoco_py/cymj.c:471:21: fatal error: mjmodel.h: No such file or directory
    compilation terminated.
    error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
    
    ----------------------------------------
  Rolling back uninstall of mujoco-py
Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-f9gjmsg2/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-yojvx9qt-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-f9gjmsg2/mujoco-py/

machinaut, 2017-07-16:
In that last one, the compiler failed with `fatal error: mjmodel.h: No such file or directory`.

It's looking for that file with the include flag `-I/root/.mujoco/mjpro150/include` -- are the header files present there, and are they readable by the user/process which is doing the pip installation?

pat-coady, 2017-07-17:
Ok, seems we are getting close. No, my header files are in ~/.mujoco/mj150/include:

pco@pco-ThinkPad-T540p:~/.mujoco/mjpro150/include$ ls
glfw3.h  mjdata.h  mjmodel.h  mjrender.h  mjvisualize.h  mjxmacro.h  mujoco.h

Seems odd they would be expected in /root/.mujoco?

pat-coady, 2017-07-17:
@machinaut - I put a copy of the .mujoco directory in /root. It got past that error, but then complained about glfw3. I installed libglfw3 libglfw3-dev. And then I ended up right back where I started with the error log I first posted.

machinaut, 2017-07-17:
`mujoco-py` looks for the `.mujoco/mjpro150` in the users home directory.  If you're running on a linux system as the root user, the home directory is `/root`.

@pat-coady same as first error means it's back to `ERROR: Invalid activation key`?

Can you confirm you're able to run the mujoco 1.50 samples using the key provided (e.g. `simulate`, which is in `mjpro150/bin`)?

pat-coady, 2017-07-17:
@machinaut - yes, back to original error with strange binary characters and the Invalid activation key.

I wonder why none of these issues occur when installing 0.5.7? 

```
pco@pco-ThinkPad-T540p:~/.mujoco/mjpro150/bin$ ./simulate
MuJoCo Pro library version 1.50
```

This opens blue window with 'Drag-and-drop model file here.'

I wonder why I am able to install and run mujoco-py 0.5.7 with no problems (albeit with mjpro131)? I didn't have to install any dependencies are anything else, "it just worked". 

Thanks a lot, Pat

pat-coady, 2017-07-17:
@machinaut - I've got a very boring filesystem. Just exactly what you get from a fresh Ubuntu install except with 2 other directories off my user home: Projects and Programs (which only contains pycharm). Even my .bashrc is untouched from Ubuntu install except for adding mujoco to LD_LIBRARY_PATH (I've tried pointed this to 131 and 150, but doesn't make a difference).

Also, not much installed, usual machine learning stuff: scipy, sklearn, numpy, matplotlib, seaborn, tensorflow, google chrome, sublime, pycharm, pandas. Nothing fancy.

Very odd. 

I'll probably be creating building a fresh setup on AWS in a couple and will likely put mujoco-py on that. I'll document every step I do in case that fails.

taegyunkim, 2017-07-29:
Had the exact same issue with @pat-coady and just ended up with installing mjpro131 with mujoco-py==0.5.7

I also have pretty boring filesystem.

taegyunk@taegyunk-xubuntu$> pip3 install -U 'mujoco-py<1.50.2,>=1.50.1'                                                                                                                                                
Collecting mujoco-py<1.50.2,>=1.50.1
  Using cached mujoco-py-1.50.1.11.tar.gz
Requirement already up-to-date: glfw>=1.4.0 in /usr/local/lib/python3.5/dist-packages (from mujoco-py<1.50.2,>=1.50.1)
Requirement already up-to-date: numpy>=1.11 in /usr/local/lib/python3.5/dist-packages (from mujoco-py<1.50.2,>=1.50.1)
Requirement already up-to-date: Cython>=0.25.2 in /usr/local/lib/python3.5/dist-packages (from mujoco-py<1.50.2,>=1.50.1)
Requirement already up-to-date: imageio>=2.1.2 in /usr/local/lib/python3.5/dist-packages (from mujoco-py<1.50.2,>=1.50.1)
Requirement already up-to-date: pillow in /usr/local/lib/python3.5/dist-packages (from imageio>=2.1.2->mujoco-py<1.50.2,>=1.50.1)
Requirement already up-to-date: olefile in /usr/local/lib/python3.5/dist-packages (from pillow->imageio>=2.1.2->mujoco-py<1.50.2,>=1.50.1)
Building wheels for collected packages: mujoco-py
  Running setup.py bdist_wheel for mujoco-py ... |

error
  Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/taegyunk/pip-build-9jt58lck/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/taegyunk/tmpdgzj1rv_pip-wheel- --python-tag cp35:
  running bdist_wheel
  running build
  Compiling /tmp/taegyunk/pip-build-9jt58lck/mujoco-py/mujoco_py/cymj.pyx because it changed.
  [1/1] Cythonizing /tmp/taegyunk/pip-build-9jt58lck/mujoco-py/mujoco_py/cymj.pyx
  running build_ext
  building 'mujoco_py.cymj' extension
  creating /tmp/taegyunk/pip-build-9jt58lck/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder
  creating /tmp/taegyunk/pip-build-9jt58lck/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5
  creating /tmp/taegyunk/pip-build-9jt58lck/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp
  creating /tmp/taegyunk/pip-build-9jt58lck/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/taegyunk
  creating /tmp/taegyunk/pip-build-9jt58lck/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/taegyunk/pip-build-9jt58lck
  creating /tmp/taegyunk/pip-build-9jt58lck/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/taegyunk/pip-build-9jt58lck/mujoco-py
  creating /tmp/taegyunk/pip-build-9jt58lck/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/taegyunk/pip-build-9jt58lck/mujoco-py/mujoco_py
  creating /tmp/taegyunk/pip-build-9jt58lck/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/taegyunk/pip-build-9jt58lck/mujoco-py/mujoco_py/gl
  x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/tmp/taegyunk/pip-build-9jt58lck/mujoco-py/mujoco_py -I/home/taegyunk/.mujoco/mjpro150/include -I/usr/local/lib/python3.5/dist-packages/numpy/core/include -I/usr/include/python3.5m -c /tmp/taegyunk/pip-build-9jt58lck/mujoco-py/mujoco_py/cymj.c -o /tmp/taegyunk/pip-build-9jt58lck/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/taegyunk/pip-build-9jt58lck/mujoco-py/mujoco_py/cymj.o -fopenmp -w
  x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/tmp/taegyunk/pip-build-9jt58lck/mujoco-py/mujoco_py -I/home/taegyunk/.mujoco/mjpro150/include -I/usr/local/lib/python3.5/dist-packages/numpy/core/include -I/usr/include/python3.5m -c /tmp/taegyunk/pip-build-9jt58lck/mujoco-py/mujoco_py/gl/osmesashim.c -o /tmp/taegyunk/pip-build-9jt58lck/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/taegyunk/pip-build-9jt58lck/mujoco-py/mujoco_py/gl/osmesashim.o -fopenmp -w
  creating /tmp/taegyunk/pip-build-9jt58lck/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/lib.linux-x86_64-3.5
  creating /tmp/taegyunk/pip-build-9jt58lck/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/lib.linux-x86_64-3.5/mujoco_py
  x86_64-linux-gnu-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 /tmp/taegyunk/pip-build-9jt58lck/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/taegyunk/pip-build-9jt58lck/mujoco-py/mujoco_py/cymj.o /tmp/taegyunk/pip-build-9jt58lck/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/taegyunk/pip-build-9jt58lck/mujoco-py/mujoco_py/gl/osmesashim.o -L/home/taegyunk/.mujoco/mjpro150/bin -Wl,--enable-new-dtags,-R/home/taegyunk/.mujoco/mjpro150/bin -lmujoco150 -lglewosmesa -lOSMesa -lGL -o /tmp/taegyunk/pip-build-9jt58lck/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/lib.linux-x86_64-3.5/mujoco_py/cymj.cpython-35m-x86 [... cut ...]
    Press Enter to exit ...
    
    ----------------------------------------
Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/taegyunk/pip-build-9jt58lck/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/taegyunk/pip-79b6a9px-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/taegyunk/pip-build-9jt58lck/mujoco-py/

DoktorMike, 2017-08-01:
I'm also running this install and it's failing. My install is in /home/myusername/.mujoco/mjpro150 and I'm running Ubuntu 16.04.

➜  ~ pip3 install mujoco-py --no-cache 
Collecting mujoco-py
  Downloading mujoco-py-1.50.1.11.tar.gz (104kB)
    100% |████████████████████████████████| 112kB 2.2MB/s 
Requirement already satisfied: glfw>=1.4.0 in /usr/local/lib/python3.5/dist-packages (from mujoco-py)
Requirement already satisfied: numpy>=1.11 in /usr/local/lib/python3.5/dist-packages (from mujoco-py)
Requirement already satisfied: Cython>=0.25.2 in /usr/local/lib/python3.5/dist-packages (from mujoco-py)
Requirement already satisfied: imageio>=2.1.2 in /usr/local/lib/python3.5/dist-packages (from mujoco-py)
Requirement already satisfied: pillow in /usr/lib/python3/dist-packages (from imageio>=2.1.2->mujoco-py)
Installing collected packages: mujoco-py
  Running setup.py install for mujoco-py ... \
|
error
    Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-wf46s7j1/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-7bmql548-record/install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    Compiling /tmp/pip-build-wf46s7j1/mujoco-py/mujoco_py/cymj.pyx because it changed.
    [1/1] Cythonizing /tmp/pip-build-wf46s7j1/mujoco-py/mujoco_py/cymj.pyx
    running build_ext
    building 'mujoco_py.cymj' extension
    creating /tmp/pip-build-wf46s7j1/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder
    creating /tmp/pip-build-wf46s7j1/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5
    creating /tmp/pip-build-wf46s7j1/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp
    creating /tmp/pip-build-wf46s7j1/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build-wf46s7j1
    creating /tmp/pip-build-wf46s7j1/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build-wf46s7j1/mujoco-py
    creating /tmp/pip-build-wf46s7j1/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build-wf46s7j1/mujoco-py/mujoco_py
    creating /tmp/pip-build-wf46s7j1/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build-wf46s7j1/mujoco-py/mujoco_py/gl
    x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/tmp/pip-build-wf46s7j1/mujoco-py/mujoco_py -I/home/michael/.mujoco/mjpro150/include -I/usr/local/lib/python3.5/dist-packages/numpy/core/include -I/usr/include/python3.5m -c /tmp/pip-build-wf46s7j1/mujoco-py/mujoco_py/cymj.c -o /tmp/pip-build-wf46s7j1/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build-wf46s7j1/mujoco-py/mujoco_py/cymj.o -fopenmp -w
    x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/tmp/pip-build-wf46s7j1/mujoco-py/mujoco_py -I/home/michael/.mujoco/mjpro150/include -I/usr/local/lib/python3.5/dist-packages/numpy/core/include -I/usr/include/python3.5m -c /tmp/pip-build-wf46s7j1/mujoco-py/mujoco_py/gl/osmesashim.c -o /tmp/pip-build-wf46s7j1/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build-wf46s7j1/mujoco-py/mujoco_py/gl/osmesashim.o -fopenmp -w
    creating /tmp/pip-build-wf46s7j1/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/lib.linux-x86_64-3.5
    creating /tmp/pip-build-wf46s7j1/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/lib.linux-x86_64-3.5/mujoco_py
    x86_64-linux-gnu-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 /tmp/pip-build-wf46s7j1/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build-wf46s7j1/mujoco-py/mujoco_py/cymj.o /tmp/pip-build-wf46s7j1/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build-wf46s7j1/mujoco-py/mujoco_py/gl/osmesashim.o -L/home/michael/.mujoco/mjpro150/bin -Wl,--enable-new-dtags,-R/home/michael/.mujoco/mjpro150/bin -lmujoco150 -lglewosmesa -lOSMesa -lGL -o /tmp/pip-build-wf46s7j1/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/lib.linux-x86_64-3.5/mujoco_py/cymj.cpython-35m-x86_64-linux-gnu.so -fopenmp
    [    [V: not found
    sh: 2: [
                : not found
    [    [V: not found
    sh: 2: [
                : not found
    [    [V: not found
    sh: 2: Syntax error: "(" unexpected
    [    [V: not found
    sh: 2: Syntax error: "(" unexpected
    ERROR: Invalid activation key
    
    Press Enter to exit ...
    
    ----------------------------------------
Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-wf46s7j1/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-7bmql548-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-wf46s7j1/mujoco-py/

gntoni, 2017-08-10:
Any advances on this? I'm having the same error on ubuntu 14.04 using a conda environment.

I tried to run setup() build --verbose, but the output is the same:
running build
[[V: not found
sh: 2: [
        : not found
[[V: not found
sh: 2: [
        : not found
[[V: not found
sh: 2: Syntax error: "(" unexpected
[[V: not found
sh: 2: Syntax error: "(" unexpected
ERROR: Invalid activation key

ikeyasu, 2017-08-15:
I'm also having the same error on ubuntu 16.10 using a conda env.
I confirmed the activation key is correct. ("./simulate" works. )

```
$ python3 setup.py -v  build
running build
running build_ext
building 'mujoco_py.cymj' extension
gcc -pthread -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -I/home/ikeyasu/Downloads/mujoco-py/mujoco_py -I/home/ikeyasu/.mujoco/mjpro150/include -I/home/ikeyasu/anaconda3/envs/chainerrl351/lib/python3.5/site-packages/numpy/core/include -I/home/ikeyasu/anaconda3/envs/chainerrl351/include/python3.5m -c /home/ikeyasu/Downloads/mujoco-py/mujoco_py/cymj.c -o /home/ikeyasu/Downloads/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/home/ikeyasu/Downloads/mujoco-py/mujoco_py/cymj.o -fopenmp -w
gcc -pthread -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -I/home/ikeyasu/Downloads/mujoco-py/mujoco_py -I/home/ikeyasu/.mujoco/mjpro150/include -I/home/ikeyasu/anaconda3/envs/chainerrl351/lib/python3.5/site-packages/numpy/core/include -I/home/ikeyasu/anaconda3/envs/chainerrl351/include/python3.5m -c /home/ikeyasu/Downloads/mujoco-py/mujoco_py/gl/osmesashim.c -o /home/ikeyasu/Downloads/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/home/ikeyasu/Downloads/mujoco-py/mujoco_py/gl/osmesashim.o -fopenmp -w
creating /home/ikeyasu/Downloads/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/lib.linux-x86_64-3.5
creating /home/ikeyasu/Downloads/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/lib.linux-x86_64-3.5/mujoco_py
gcc -pthread -shared -L/home/ikeyasu/anaconda3/envs/chainerrl351/lib -Wl,-rpath=/home/ikeyasu/anaconda3/envs/chainerrl351/lib,--no-as-needed /home/ikeyasu/Downloads/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/home/ikeyasu/Downloads/mujoco-py/mujoco_py/cymj.o /home/ikeyasu/Downloads/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/home/ikeyasu/Downloads/mujoco-py/mujoco_py/gl/osmesashim.o -L/home/ikeyasu/.mujoco/mjpro150/bin -L/home/ikeyasu/anaconda3/envs/chainerrl351/lib -Wl,--enable-new-dtags,-R/home/ikeyasu/.mujoco/mjpro150/bin -lmujoco150 -lglewosmesa -lOSMesa -lGL -lpython3.5m -o /home/ikeyasu/Downloads/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/lib.linux-x86_64-3.5/mujoco_py/cymj.cpython-35m-x86_64-linux-gnu.so -fopenmp
[[V: not found
sh: 2: [
        : not found
[[V: not found
sh: 2: [
        : not found
[[V: not found
sh: 2: Syntax error: "(" unexpected
[[V: not found
sh: 2: Syntax error: "(" unexpected
ERROR: Invalid activation key
```

lonehades, 2017-08-16:
It's same problem on my ubuntu 16.04.3 LTS using virtualenv.
'./simulate' is working well.

...
Running command /home/ben/study/enn/.venv/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-lsu65531/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-5txgc6gw-record/install-record.txt --single-version-externally-managed --compile --install-headers /home/ben/study/enn/.venv/include/site/python3.5/mujoco-py
    running install
    running build
    [    [V: not found
    sh: 2: [
                : not found
    [    [V: not found
    sh: 2: [
                : not found
    [    [V: not found
    sh: 2: Syntax error: "(" unexpected
    [    [V: not found
    sh: 2: Syntax error: "(" unexpected
    ERROR: Invalid activation key

ChaoYue0307, 2017-08-24:
I have same problem too, on Ubuntu 17.04
pip3 install -U 'mujoco-py<1.50.2,>=1.50.1'
Collecting mujoco-py<1.50.2,>=1.50.1
  Using cached mujoco-py-1.50.1.18.tar.gz
Collecting Cython>=0.25.2 (from mujoco-py<1.50.2,>=1.50.1)
  Using cached Cython-0.26-cp35-cp35m-manylinux1_x86_64.whl
Collecting glfw>=1.4.0 (from mujoco-py<1.50.2,>=1.50.1)
Collecting imageio>=2.1.2 (from mujoco-py<1.50.2,>=1.50.1)
Collecting numpy>=1.11 (from mujoco-py<1.50.2,>=1.50.1)
  Using cached numpy-1.13.1-cp35-cp35m-manylinux1_x86_64.whl
Collecting pillow (from imageio>=2.1.2->mujoco-py<1.50.2,>=1.50.1)
  Using cached Pillow-4.2.1-cp35-cp35m-manylinux1_x86_64.whl
Collecting olefile (from pillow->imageio>=2.1.2->mujoco-py<1.50.2,>=1.50.1)
Building wheels for collected packages: mujoco-py
  Running setup.py bdist_wheel for mujoco-py ... \

error
  Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-npqsffv_/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmph9m49sz5pip-wheel- --python-tag cp35:
  running bdist_wheel
  running build
  Compiling /tmp/pip-build-npqsffv_/mujoco-py/mujoco_py/cymj.pyx because it depends on /home/chaoyue/.local/lib/python3.5/site-packages/Cython/Includes/numpy/__init__.pxd.
  [1/1] Cythonizing /tmp/pip-build-npqsffv_/mujoco-py/mujoco_py/cymj.pyx
  running build_ext
  building 'mujoco_py.cymj' extension
  creating /tmp/pip-build-npqsffv_/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder
  creating /tmp/pip-build-npqsffv_/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5
  creating /tmp/pip-build-npqsffv_/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp
  creating /tmp/pip-build-npqsffv_/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build-npqsffv_
  creating /tmp/pip-build-npqsffv_/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build-npqsffv_/mujoco-py
  creating /tmp/pip-build-npqsffv_/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build-npqsffv_/mujoco-py/mujoco_py
  creating /tmp/pip-build-npqsffv_/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build-npqsffv_/mujoco-py/mujoco_py/gl
  x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fdebug-prefix-map=/build/python3.5-7CCmgg/python3.5-3.5.3=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/tmp/pip-build-npqsffv_/mujoco-py/mujoco_py -I/home/chaoyue/.mujoco/mjpro150/include -I/home/chaoyue/.local/lib/python3.5/site-packages/numpy/core/include -I/usr/include/python3.5m -c /tmp/pip-build-npqsffv_/mujoco-py/mujoco_py/cymj.c -o /tmp/pip-build-npqsffv_/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build-npqsffv_/mujoco-py/mujoco_py/cymj.o -fopenmp -w
  x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fdebug-prefix-map=/build/python3.5-7CCmgg/python3.5-3.5.3=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/tmp/pip-build-npqsffv_/mujoco-py/mujoco_py -I/home/chaoyue/.mujoco/mjpro150/include -I/home/chaoyue/.local/lib/python3.5/site-packages/numpy/core/include -I/usr/include/python3.5m -c /tmp/pip-build-npqsffv_/mujoco-py/mujoco_py/gl/osmesashim.c -o /tmp/pip-build-npqsffv_/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build-npqsffv_/mujoco-py/mujoco_py/gl/osmesashim.o -fopenmp -w
  creating /tmp/pip-build-npqsffv_/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/lib.linux-x86_64-3.5
  creating /tmp/pip-build-npqsffv_/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/lib.linux-x86_64-3.5/mujoco_py
  x86_64-linux-gnu-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fdebug-prefix-map=/build/python3.5-7CCmgg/python3.5-3.5.3=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 /tmp/pip-build-npqsffv_/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build-npqsffv_/mujoco-py/mujoco_py/cymj.o /tmp/pip-build-npqsffv_/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build-npqsffv_/mujoco-py/mujoco_py/gl/osmesashim.o -L/home/chaoyue/.mujoco/mjpro150/bin -Wl,--enable-new-dtags,-R/home/chaoyue/.mujoco/mjpro150/bin -lmujoco150 -lglewosmesa -lOSMesa -lGL -o /tmp/pip-build-npqsffv_/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/lib.linux-x86_64-3.5/mujoco_py/cymj.cpython-35m-x86_64-linux-gnu.so -fopenmp
  [  [V: not found
  sh: 2: [
            : not found
  [  [V: not foundpip3 install -U 'mujoco-py<1.50.2,>=1.50.1'
Collecting mujoco-py<1.50.2,>=1.50.1
  Using cached mujoco-py-1.50.1.18.tar.gz
Collecting Cython>=0.25.2 (from mujoco-py<1.50.2,>=1.50.1)
  Using cached Cython-0.26-cp35-cp35m-manylinux1_x86_64.whl
Collecting glfw>=1.4.0 (from mujoco-py<1.50.2,>=1.50.1)
Collecting imageio>=2.1.2 (from mujoco-py<1.50.2,>=1.50.1)
Collecting numpy>=1.11 (from mujoco-py<1.50.2,>=1.50.1)
  Using cached numpy-1.13.1-cp35-cp35m-manylinux1_x86_64.whl
[... cut ...]
    Press Enter to exit ...
    
    ----------------------------------------
Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-npqsffv_/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-qi6j4_9l-record/install-record.txt --single-version-externally-managed --compile --user --prefix=" failed with error code 1 in /tmp/pip-build-npqsffv_/mujoco-py/

ThyrixYang, 2017-08-29:
Same error with: "ERROR: Invalid activation key"

Kukanani, 2017-09-04:
I can also confirm this behavior on Ubuntu 16.04. Any other ideas of how we can help troubleshoot this, @machinaut ?

neveryoubetray, 2017-09-07:
Try cloning the repo instead: 
git clone https://github.com/openai/mujoco-py.git
go to the directory mujoco-py
python3 setup.py install
as suggested by:
https://github.com/openai/mujoco-py/issues/47
should then be able to 
>>python3
>>import mujoco_py
which will compile the first time.
But remember, this version doesn't work with gym, so gym.make('environment') won't work! 
But the examples should run (eg. examples/tosser.py)
Use an older version if you want to use with gym.

fishchian, 2017-09-08:
I also have the same issue, and it is so annoying.

When read the email send from mujoco.pro again, I found some inconsistency between their suggestion and the installation suggestion in openAI.mujoco-py

see below:
You have successfully registered for a free trial of MuJoCo Pro. Your license text and activation key are attached to this email. Please download and unzip the software, copy the attached file 'mjkey.txt' to the 'bin' subdirectory, and then you can run the samples. See also instructions in README.txt in the 'doc' subdirectory.

I tried as this but the error is then:  
   ERROR: Could not open activation key file /home/kqian/.mujoco/mjkey.txt

made me mad...........wtf....
anyone who has fixed this issue is welcome to give us some advice, will be appreciated, thanks

rhiga2, 2017-09-11:
@fishchian I had similar issues. I ended up copying mjkey.txt into ~/.mujoco and ~/.mujoco/mjpro150/bin. Not sure why that worked for me.

fishchian, 2017-09-11:
thanks for mentioning. I also copied the key into  ~/.mujoco and
~/.mujoco/mjpro150/bin. Still, doesn't work for me.

#Failed building wheels//

thank you for sharing...

fishchian, 2017-10-03:
Mujoco 0.5.7 works well for me. it is compatible with Python2 and python3

ViktorM, 2017-10-26:
Are there any updates on the progress with current issues? Installation of 0.5.7 works well and out of box as it should be, but looks like it's impossible to install and make 1.5.x working on quit standard configuration - Ubuntu 16.04. I suppose all these issues can be considered to be solved only when just "pip install mujoco-py" or "pip install -e ." work.

sritee, 2017-10-29:
@fishchian  how do you know the commands for the older version? The documentation seems to be for 1.5

fishchian, 2017-10-29:
@sritee  Both would probably work
https://github.com/openai/mujoco-py/tree/0.5
: pip install mujoco-py

https://github.com/openai/mujoco-py
: Python 2 has been desupported since 1.50.1.0. Python 2 users can stay on the 0.5 branch. The latest release there is 0.5.7 which can be installed with pip install mujoco-py==0.5.7.

xinleipan, 2017-12-26:
I also encountered similar problem but was able to solve it by switching to mjpro131. You should first download mjpro131 from the official site and then move it to `~/.mujoco/mjpro131`, also put your key in `~/.mujoco`. Then you should install mujoco_py from the branch 0.5.

cathywu, 2017-12-27:
Are there any updates on this issue? I am getting this issue after confirming that I can run mujoco in the same folder/terminal where I am attempting to install `mujoco_py`.

```
$ /home/XXX/.mujoco/mjpro150/bin/simulate
MuJoCo Pro library version 1.50
```

And the following results from various install commands, `pip install -U 'mujoco-py<1.50.2,>=1.50.1'` and `pip install mujoco_py`:
```
.txt --single-version-externally-managed --compile:
    running install
    running build
    [    [V: not found
    sh: 2: [
                : not found
    [    [V: not found
    sh: 2: [
                : not found
    [    [V: not found
    sh: 2: Syntax error: "(" unexpected
    [    [V: not found
    sh: 2: Syntax error: "(" unexpected
    ERROR: Invalid activation key

    Press Enter to exit ...

    ----------------------------------------
Command "/home/XXX/anaconda3/envs/multiagent/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-buf19ljm/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-9iq1k1i6-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-buf19ljm/mujoco-py/
```

xinleipan, 2017-12-27:
@cathywu I think you may try mjpro131 instead of mjpro150...

nazir-ust, 2018-01-28:
Was able to solve this by combining [this](https://github.com/openai/mujoco-py/issues/66#issuecomment-315597679) and [this](https://github.com/openai/mujoco-py/issues/110#issuecomment-345267275). Able to load examples and import smoothly. Thanks.

CrazyHeex, 2018-02-06:
git clone the mujoco-py to where you wish to.
pip -e ./mujoco-py
done
solved

gandalf1819, 2019-04-09:
I'm trying to build a docker image of universe. I'm running the following command:

`docker build -t universe .`

I have mujoco installed at ~/.mujoco/mujoco200. I have the license key at ~/.mujoco/.

I'm getting the following error:

Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-da8i39jm/mujoco-py/setup.py", line 45, in <module>
        tests_require=read_requirements_file('requirements.dev.txt'),
      File "/usr/lib/python3.5/distutils/core.py", line 148, in setup
        dist.run_commands()
      File "/usr/lib/python3.5/distutils/dist.py", line 955, in run_commands
        self.run_command(cmd)
      File "/usr/lib/python3.5/distutils/dist.py", line 974, in run_command
        cmd_obj.run()
      File "/usr/lib/python3/dist-packages/setuptools/command/install.py", line 61, in run
        return orig.install.run(self)
      File "/usr/lib/python3.5/distutils/command/install.py", line 583, in run
        self.run_command('build')
      File "/usr/lib/python3.5/distutils/cmd.py", line 313, in run_command
        self.distribution.run_command(command)
      File "/usr/lib/python3.5/distutils/dist.py", line 974, in run_command
        cmd_obj.run()
      File "/tmp/pip-install-da8i39jm/mujoco-py/setup.py", line 29, in run
        import mujoco_py  # noqa: force build
      File "/tmp/pip-install-da8i39jm/mujoco-py/mujoco_py/__init__.py", line 3, in <module>
        from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
      File "/tmp/pip-install-da8i39jm/mujoco-py/mujoco_py/builder.py", line 502, in <module>
        mujoco_path, key_path = discover_mujoco()
      File "/tmp/pip-install-da8i39jm/mujoco-py/mujoco_py/utils.py", line 93, in discover_mujoco
        raise Exception(message)
    Exception:
    You appear to be missing MuJoCo.  We expected to find the file here: /root/.mujoco/mujoco200
    
    This package only provides python bindings, the library must be installed separately.
    
    Please follow the instructions on the README to install MuJoCo
    
        https://github.com/openai/mujoco-py#install-mujoco
    
    Which can be downloaded from the website
    
        https://www.roboti.us/index.html
    
    
    ----------------------------------------
Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-da8i39jm/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-vl7solc7/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-install-da8i39jm/mujoco-py/
The command '/bin/sh -c pip install gym[all]' returned a non-zero code: 1

Shashank001122, 2019-10-03:
> I'm trying to build a docker image of universe. I'm running the following command:
> 
> `docker build -t universe .`
> 
> I have mujoco installed at ~/.mujoco/mujoco200. I have the license key at ~/.mujoco/.
> 
> I'm getting the following error:
> 
> Traceback (most recent call last):
> File "", line 1, in
> File "/tmp/pip-install-da8i39jm/mujoco-py/setup.py", line 45, in
> tests_require=read_requirements_file('requirements.dev.txt'),
> File "/usr/lib/python3.5/distutils/core.py", line 148, in setup
> dist.run_commands()
> File "/usr/lib/python3.5/distutils/dist.py", line 955, in run_commands
> self.run_command(cmd)
> File "/usr/lib/python3.5/distutils/dist.py", line 974, in run_command
> cmd_obj.run()
> File "/usr/lib/python3/dist-packages/setuptools/command/install.py", line 61, in run
> return orig.install.run(self)
> File "/usr/lib/python3.5/distutils/command/install.py", line 583, in run
> self.run_command('build')
> File "/usr/lib/python3.5/distutils/cmd.py", line 313, in run_command
> self.distribution.run_command(command)
> File "/usr/lib/python3.5/distutils/dist.py", line 974, in run_command
> cmd_obj.run()
> File "/tmp/pip-install-da8i39jm/mujoco-py/setup.py", line 29, in run
> import mujoco_py # noqa: force build
> File "/tmp/pip-install-da8i39jm/mujoco-py/mujoco_py/**init**.py", line 3, in
> from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
> File "/tmp/pip-install-da8i39jm/mujoco-py/mujoco_py/builder.py", line 502, in
> mujoco_path, key_path = discover_mujoco()
> File "/tmp/pip-install-da8i39jm/mujoco-py/mujoco_py/utils.py", line 93, in discover_mujoco
> raise Exception(message)
> Exception:
> You appear to be missing MuJoCo. We expected to find the file here: /root/.mujoco/mujoco200
> 
> ```
> This package only provides python bindings, the library must be installed separately.
> 
> Please follow the instructions on the README to install MuJoCo
> 
>     https://github.com/openai/mujoco-py#install-mujoco
> 
> Which can be downloaded from the website
> 
>     https://www.roboti.us/index.html
> 
> 
> ----------------------------------------
> ```
> 
> Command "/usr/bin/python3 -u -c "import setuptools, tokenize;**file**='/tmp/pip-install-da8i39jm/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(**file**);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, **file**, 'exec'))" install --record /tmp/pip-record-vl7solc7/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-install-da8i39jm/mujoco-py/
> The command '/bin/sh -c pip install gym[all]' returned a non-zero code: 1

Any luck with this! I am stuck

blackjacket1403, 2025-01-11:
https://github.com/openai/mujoco-py/releases/tag/0.5.7

Refer to this and use this version of mujoco-py, this worked for me with successful builds.