openai/mujoco-py #68: windows: anaconda: error running: The filename or extension is too long

rzorzorzo, 2017-07-16:
hello,

On windows 10, I was finaly able to install mujoco-py. However when trying to run an example I am getting the following:

> tensorflow-gpu) C:\test\mujoco-py\mujoco-py-master\examples>python tosser.py
running build_ext
building 'mujoco_py.cymj' extension
creating C:\Users\test\Anaconda3\envs\tensorflow-gpu\lib\site-packages\mujoco_py-1.50.1.7-py3.5.egg\mujoco_py\generated\_pyxbld_WindowsExtensionBuilder\temp.win-amd64-3.5\Release\Users\test\Anaconda3\envs\tensorflow-gpu\lib\site-packages\mujoco_py-1.50.1.7-py3.5.egg
Traceback (most recent call last):
  File "C:\Users\test\Anaconda3\envs\tensorflow-gpu\lib\distutils\dir_util.py", line 70, in mkpath
    os.mkdir(head, mode)
FileNotFoundError: [WinError 206] The filename or extension is too long: 'C:\\Users\\test\\Anaconda3\\envs\\tensorflow-gpu\\lib\\site-packages\\mujoco_py-1.50.1.7-py3.5.egg\\mujoco_py\\generated\\_pyxbld_WindowsExtensionBuilder\\temp.win-amd64-3.5\\Release\\Users\\test\\Anaconda3\\envs\\tensorflow-gpu\\lib\\site-packages\\mujoco_py-1.50.1.7-py3.5.egg'

machinaut, 2017-07-17:
The crux of the error seems to be the resulting path for what it's building is too long for the operating system.

I'm confused why there seems to be a python within a python.  "Users\test\Anaconda3\envs\tensorflow-gpu\lib\site-packages\mujoco_py-1.50.1.7-py3.5" appears twice in the path, which seems to point to something being incorrect.

What does just trying to import the module do:  `python -c "import mujoco_py"`

ViktorM, 2017-07-24:
Now I have the same error too. And after python -c "import mujoco_py" too:
```
python -c "import mujoco_py"
running build_ext
building 'mujoco_py.cymj' extension
creating C:\Users\vmakoviychuk\AppData\Local\Continuum\Anaconda3\envs\Isaac\lib\site-packages\mujoco_py-1.50.1.10-py3.5.egg\mujoco_py\generated\_pyxbld_WindowsExtensionBuilder\temp.win-amd64-3.5\Release\Users\vmakoviychuk\AppData\Local\Continuum\Anaconda3\envs
Traceback (most recent call last):
  File "C:\Users\vmakoviychuk\AppData\Local\Continuum\Anaconda3\envs\Isaac\lib\distutils\dir_util.py", line 70, in mkpath
    os.mkdir(head, mode)
FileNotFoundError: [WinError 206] The filename or extension is too long: 'C:\\Users\\vmakoviychuk\\AppData\\Local\\Continuum\\Anaconda3\\envs\\Isaac\\lib\\site-packages\\mujoco_py-1.50.1.10-py3.5.egg\\mujoco_py\\generated\\_pyxbld_WindowsExtensionBuilder\\temp.win-amd64-3.5\\Release\\Users\\vmakoviychuk\\AppData\\Local\\Continuum\\Anaconda3\\envs'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\vmakoviychuk\AppData\Local\Continuum\Anaconda3\envs\Isaac\lib\site-packages\mujoco_py-1.50.1.10-py3.5.egg\mujoco_py\__init__.py", line 1, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "C:\Users\vmakoviychuk\AppData\Local\Continuum\Anaconda3\envs\Isaac\lib\site-packages\mujoco_py-1.50.1.10-py3.5.egg\mujoco_py\builder.py", line 296, in <module>
    cymj = load_cython_ext(mjpro_path)
  File "C:\Users\vmakoviychuk\AppData\Local\Continuum\Anaconda3\envs\Isaac\lib\site-packages\mujoco_py-1.50.1.10-py3.5.egg\mujoco_py\builder.py", line 56, in load_cython_ext
    cext_so_path = builder.build()
  File "C:\Users\vmakoviychuk\AppData\Local\Continuum\Anaconda3\envs\Isaac\lib\site-packages\mujoco_py-1.50.1.10-py3.5.egg\mujoco_py\builder.py", line 119, in build
    built_so_file_path = self._build_impl()
  File "C:\Users\vmakoviychuk\AppData\Local\Continuum\Anaconda3\envs\Isaac\lib\site-packages\mujoco_py-1.50.1.10-py3.5.egg\mujoco_py\builder.py", line 139, in _build_impl
    dist.run_commands()
  File "C:\Users\vmakoviychuk\AppData\Local\Continuum\Anaconda3\envs\Isaac\lib\distutils\dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "C:\Users\vmakoviychuk\AppData\Local\Continuum\Anaconda3\envs\Isaac\lib\distutils\dist.py", line 974, in run_command
    cmd_obj.run()
  File "C:\Users\vmakoviychuk\AppData\Local\Continuum\Anaconda3\envs\Isaac\lib\site-packages\Cython\Distutils\old_build_ext.py", line 185, in run
    _build_ext.build_ext.run(self)
  File "C:\Users\vmakoviychuk\AppData\Local\Continuum\Anaconda3\envs\Isaac\lib\distutils\command\build_ext.py", line 338, in run
    self.build_extensions()
  File "C:\Users\vmakoviychuk\AppData\Local\Continuum\Anaconda3\envs\Isaac\lib\site-packages\mujoco_py-1.50.1.10-py3.5.egg\mujoco_py\builder.py", line 77, in build_extensions
    build_ext.build_extensions(self)
  File "C:\Users\vmakoviychuk\AppData\Local\Continuum\Anaconda3\envs\Isaac\lib\site-packages\Cython\Distutils\old_build_ext.py", line 193, in build_extensions
    self.build_extension(ext)
  File "C:\Users\vmakoviychuk\AppData\Local\Continuum\Anaconda3\envs\Isaac\lib\distutils\command\build_ext.py", line 532, in build_extension
    depends=ext.depends)
  File "C:\Users\vmakoviychuk\AppData\Local\Continuum\Anaconda3\envs\Isaac\lib\distutils\_msvccompiler.py", line 308, in compile
    sources, depends, extra_postargs)
  File "C:\Users\vmakoviychuk\AppData\Local\Continuum\Anaconda3\envs\Isaac\lib\distutils\ccompiler.py", line 348, in _setup_compile
    self.mkpath(os.path.dirname(obj))
  File "C:\Users\vmakoviychuk\AppData\Local\Continuum\Anaconda3\envs\Isaac\lib\distutils\ccompiler.py", line 915, in mkpath
    mkpath(name, mode, dry_run=self.dry_run)
  File "C:\Users\vmakoviychuk\AppData\Local\Continuum\Anaconda3\envs\Isaac\lib\distutils\dir_util.py", line 74, in mkpath
    "could not create '%s': %s" % (head, exc.args[-1]))
distutils.errors.DistutilsFileError: could not create 'C:\Users\vmakoviychuk\AppData\Local\Continuum\Anaconda3\envs\Isaac\lib\site-packages\mujoco_py-1.50.1.10-py3.5.egg\mujoco_py\generated\_pyxbld_WindowsExtensionBuilder\temp.win-amd64-3.5\Release\Users\vmakoviychuk\AppData\Local\Continuum\Anaconda3\envs': The filename or extension is too long
```

nikkou, 2017-08-28:
I've spent this evening trying to set up Windows 10 + Anaconda + Python 3.5 environment + Tensorflow + Mujoco_py and can't resolve this final error as well. I tried to set up registry keys and group policies to enable long paths in Windows according to various guides but it didn't work at all.

Jay-Lewis, 2017-10-08:
I have the same issue. Windows 7 + Python 3.5.2 environment . Has anyone found a solution or resolution to this problem? thanks!

SystemSigma, 2017-10-28:
`C:\Users\Andrea>python
Python 3.5.2 (v3.5.2:4def2a2901a5, Jun 25 2016, 22:18:55) [MSC v.1900 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import mujoco_py
Compiling C:\Users\Andrea\AppData\Local\Programs\Python\Python35\lib\site-packages\mujoco_py-1.50.1.24-py3.5.egg\mujoco_py\cymj.pyx because it changed.
[1/1] Cythonizing C:\Users\Andrea\AppData\Local\Programs\Python\Python35\lib\site-packages\mujoco_py-1.50.1.24-py3.5.egg\mujoco_py\cymj.pyx
running build_ext
building 'mujoco_py.cymj' extension
creating C:\Users\Andrea\AppData\Local\Programs\Python\Python35\lib\site-packages\mujoco_py-1.50.1.24-py3.5.egg\mujoco_py\generated\_pyxbld_WindowsExtensionBuilder
creating C:\Users\Andrea\AppData\Local\Programs\Python\Python35\lib\site-packages\mujoco_py-1.50.1.24-py3.5.egg\mujoco_py\generated\_pyxbld_WindowsExtensionBuilder\temp.win-amd64-3.5
creating C:\Users\Andrea\AppData\Local\Programs\Python\Python35\lib\site-packages\mujoco_py-1.50.1.24-py3.5.egg\mujoco_py\generated\_pyxbld_WindowsExtensionBuilder\temp.win-amd64-3.5\Release
creating C:\Users\Andrea\AppData\Local\Programs\Python\Python35\lib\site-packages\mujoco_py-1.50.1.24-py3.5.egg\mujoco_py\generated\_pyxbld_WindowsExtensionBuilder\temp.win-amd64-3.5\Release\Users
creating C:\Users\Andrea\AppData\Local\Programs\Python\Python35\lib\site-packages\mujoco_py-1.50.1.24-py3.5.egg\mujoco_py\generated\_pyxbld_WindowsExtensionBuilder\temp.win-amd64-3.5\Release\Users\Andrea
creating C:\Users\Andrea\AppData\Local\Programs\Python\Python35\lib\site-packages\mujoco_py-1.50.1.24-py3.5.egg\mujoco_py\generated\_pyxbld_WindowsExtensionBuilder\temp.win-amd64-3.5\Release\Users\Andrea\AppData
creating C:\Users\Andrea\AppData\Local\Programs\Python\Python35\lib\site-packages\mujoco_py-1.50.1.24-py3.5.egg\mujoco_py\generated\_pyxbld_WindowsExtensionBuilder\temp.win-amd64-3.5\Release\Users\Andrea\AppData\Local
creating C:\Users\Andrea\AppData\Local\Programs\Python\Python35\lib\site-packages\mujoco_py-1.50.1.24-py3.5.egg\mujoco_py\generated\_pyxbld_WindowsExtensionBuilder\temp.win-amd64-3.5\Release\Users\Andrea\AppData\Local\Programs
creating C:\Users\Andrea\AppData\Local\Programs\Python\Python35\lib\site-packages\mujoco_py-1.50.1.24-py3.5.egg\mujoco_py\generated\_pyxbld_WindowsExtensionBuilder\temp.win-amd64-3.5\Release\Users\Andrea\AppData\Local\Programs\Python
creating C:\Users\Andrea\AppData\Local\Programs\Python\Python35\lib\site-packages\mujoco_py-1.50.1.24-py3.5.egg\mujoco_py\generated\_pyxbld_WindowsExtensionBuilder\temp.win-amd64-3.5\Release\Users\Andrea\AppData\Local\Programs\Python\Python35
creating C:\Users\Andrea\AppData\Local\Programs\Python\Python35\lib\site-packages\mujoco_py-1.50.1.24-py3.5.egg\mujoco_py\generated\_pyxbld_WindowsExtensionBuilder\temp.win-amd64-3.5\Release\Users\Andrea\AppData\Local\Programs\Python\Python35\lib
creating C:\Users\Andrea\AppData\Local\Programs\Python\Python35\lib\site-packages\mujoco_py-1.50.1.24-py3.5.egg\mujoco_py\generated\_pyxbld_WindowsExtensionBuilder\temp.win-amd64-3.5\Release\Users\Andrea\AppData\Local\Programs\Python\Python35\lib\site-packages
Traceback (most recent call last):
  File "C:\Users\Andrea\AppData\Local\Programs\Python\Python35\lib\distutils\dir_util.py", line 70, in mkpath
    os.mkdir(head, mode)
FileNotFoundError: [WinError 206] Nome del file o estensione troppo lunga: 'C:\\Users\\Andrea\\AppData\\Local\\Programs\\Python\\Python35\\lib\\site-packages\\mujoco_py-1.50.1.24-py3.5.egg\\mujoco_py\\generated\\_pyxbld_WindowsExtensionBuilder\\temp.win-amd64-3.5\\Release\\Users\\Andrea\\AppData\\Local\\Programs\\Python\\Python35\\lib\\site-packages'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\Andrea\AppData\Local\Programs\Python\Python35\lib\site-packages\mujoco_py-1.50.1.24-py3.5.egg\mujoco_py\__init__.py", line 1, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "C:\Users\Andrea\AppData\Local\Programs\Python\Python35\lib\site-packages\mujoco_py-1.50.1.24-py3.5.egg\mujoco_py\builder.py", line 439, in <module>
    cymj = load_cython_ext(mjpro_path)
  File "C:\Users\Andrea\AppData\Local\Programs\Python\Python35\lib\site-packages\mujoco_py-1.50.1.24-py3.5.egg\mujoco_py\builder.py", line 80, in load_cython_ext
    cext_so_path = builder.build()
  File "C:\Users\Andrea\AppData\Local\Programs\Python\Python35\lib\site-packages\mujoco_py-1.50.1.24-py3.5.egg\mujoco_py\builder.py", line 187, in build
    built_so_file_path = self._build_impl()
  File "C:\Users\Andrea\AppData\Local\Programs\Python\Python35\lib\site-packages\mujoco_py-1.50.1.24-py3.5.egg\mujoco_py\builder.py", line 207, in _build_impl
    dist.run_commands()
  File "C:\Users\Andrea\AppData\Local\Programs\Python\Python35\lib\distutils\dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "C:\Users\Andrea\AppData\Local\Programs\Python\Python35\lib\distutils\dist.py", line 974, in run_command
    cmd_obj.run()
  File "c:\users\andrea\desktop\rl\mujoco-py\.eggs\cython-0.27.2-py3.5-win-amd64.egg\Cython\Distutils\old_build_ext.py", line 186, in run
    _build_ext.build_ext.run(self)
  File "C:\Users\Andrea\AppData\Local\Programs\Python\Python35\lib\distutils\command\build_ext.py", line 338, in run
    self.build_extensions()
[... cut ...]
  File "C:\Users\Andrea\AppData\Local\Programs\Python\Python35\lib\distutils\dir_util.py", line 74, in mkpath
    "could not create '%s': %s" % (head, exc.args[-1]))
distutils.errors.DistutilsFileError: could not create 'C:\Users\Andrea\AppData\Local\Programs\Python\Python35\lib\site-packages\mujoco_py-1.50.1.24-py3.5.egg\mujoco_py\generated\_pyxbld_WindowsExtensionBuilder\temp.win-amd64-3.5\Release\Users\Andrea\AppData\Local\Programs\Python\Python35\lib\site-packages': Nome del file o estensione troppo lunga`

Same problem here, but without Anaconda.. Any news?

SystemSigma, 2017-10-28:
Same problem here:
Windows 10 + Python 3.5.2 64bit... Any news? [WinError 206]
```C:\Users\Andrea>python
Python 3.5.2 (v3.5.2:4def2a2901a5, Jun 25 2016, 22:18:55) [MSC v.1900 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import mujoco_py
running build_ext
building 'mujoco_py.cymj' extension
creating C:\Users\Andrea\AppData\Local\Programs\Python\Python35\lib\site-packages\mujoco_py-1.50.1.24-py3.5.egg\mujoco_py\generated\_pyxbld_WindowsExtensionBuilder\temp.win-amd64-3.5\Release\Users\Andrea\AppData\Local\Programs\Python\Python35\lib\site-packages
Traceback (most recent call last):
  File "C:\Users\Andrea\AppData\Local\Programs\Python\Python35\lib\distutils\dir_util.py", line 70, in mkpath
    os.mkdir(head, mode)
FileNotFoundError: [WinError 206] Nome del file o estensione troppo lunga: 'C:\\Users\\Andrea\\AppData\\Local\\Programs\\Python\\Python35\\lib\\site-packages\\mujoco_py-1.50.1.24-py3.5.egg\\mujoco_py\\generated\\_pyxbld_WindowsExtensionBuilder\\temp.win-amd64-3.5\\Release\\Users\\Andrea\\AppData\\Local\\Programs\\Python\\Python35\\lib\\site-packages'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\Andrea\AppData\Local\Programs\Python\Python35\lib\site-packages\mujoco_py-1.50.1.24-py3.5.egg\mujoco_py\__init__.py", line 1, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "C:\Users\Andrea\AppData\Local\Programs\Python\Python35\lib\site-packages\mujoco_py-1.50.1.24-py3.5.egg\mujoco_py\builder.py", line 439, in <module>
    cymj = load_cython_ext(mjpro_path)
  File "C:\Users\Andrea\AppData\Local\Programs\Python\Python35\lib\site-packages\mujoco_py-1.50.1.24-py3.5.egg\mujoco_py\builder.py", line 80, in load_cython_ext
    cext_so_path = builder.build()
  File "C:\Users\Andrea\AppData\Local\Programs\Python\Python35\lib\site-packages\mujoco_py-1.50.1.24-py3.5.egg\mujoco_py\builder.py", line 187, in build
    built_so_file_path = self._build_impl()
  File "C:\Users\Andrea\AppData\Local\Programs\Python\Python35\lib\site-packages\mujoco_py-1.50.1.24-py3.5.egg\mujoco_py\builder.py", line 207, in _build_impl
    dist.run_commands()
  File "C:\Users\Andrea\AppData\Local\Programs\Python\Python35\lib\distutils\dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "C:\Users\Andrea\AppData\Local\Programs\Python\Python35\lib\distutils\dist.py", line 974, in run_command
    cmd_obj.run()
  File "c:\users\andrea\desktop\rl\mujoco-py\.eggs\cython-0.27.2-py3.5-win-amd64.egg\Cython\Distutils\old_build_ext.py", line 186, in run
    _build_ext.build_ext.run(self)
  File "C:\Users\Andrea\AppData\Local\Programs\Python\Python35\lib\distutils\command\build_ext.py", line 338, in run
    self.build_extensions()
  File "C:\Users\Andrea\AppData\Local\Programs\Python\Python35\lib\site-packages\mujoco_py-1.50.1.24-py3.5.egg\mujoco_py\builder.py", line 106, in build_extensions
    build_ext.build_extensions(self)
  File "c:\users\andrea\desktop\rl\mujoco-py\.eggs\cython-0.27.2-py3.5-win-amd64.egg\Cython\Distutils\old_build_ext.py", line 194, in build_extensions
    self.build_extension(ext)
  File "C:\Users\Andrea\AppData\Local\Programs\Python\Python35\lib\distutils\command\build_ext.py", line 532, in build_extension
    depends=ext.depends)
  File "C:\Users\Andrea\AppData\Local\Programs\Python\Python35\lib\distutils\_msvccompiler.py", line 308, in compile
    sources, depends, extra_postargs)
  File "C:\Users\Andrea\AppData\Local\Programs\Python\Python35\lib\distutils\ccompiler.py", line 348, in _setup_compile
    self.mkpath(os.path.dirname(obj))
  File "C:\Users\Andrea\AppData\Local\Programs\Python\Python35\lib\distutils\ccompiler.py", line 915, in mkpath
    mkpath(name, mode, dry_run=self.dry_run)
  File "C:\Users\Andrea\AppData\Local\Programs\Python\Python35\lib\distutils\dir_util.py", line 74, in mkpath
    "could not create '%s': %s" % (head, exc.args[-1]))
distutils.errors.DistutilsFileError: could not create 'C:\Users\Andrea\AppData\Local\Programs\Python\Python35\lib\site-packages\mujoco_py-1.50.1.24-py3.5.egg\mujoco_py\generated\_pyxbld_WindowsExtensionBuilder\temp.win-amd64-3.5\Release\Users\Andrea\AppData\Local\Programs\Python\Python35\lib\site-packages': Nome del file o estensione troppo lunga```

SystemSigma, 2017-10-29:
The problem seems to be solved with the Python installation dir in 
C:\Python35

I can succesfully import mujoco_py from terminal but not on pycharm.

PabTorre, 2019-01-04:
Did some digging around this bug... 
Tracked it down to builder.py line228... 
Up to that build build.build_base has a directory that looks ok. 
(len of the str up to that point is 149 letters in my case...) 
But from there it goes to dist.run_commands() where it dives into Cython. 
On Cython's side... (with my very limited understanding of that side...) it looks like tempfile.TemporaryDirectory on cymj.pyx is duplicating the directory structure and going over the 260 limit. 

I haven't been able to fix it, but this information may be useful for someone else looking into this bug. 

cheers.

nishtha82, 2021-06-21:
Hello! I am facing the same issue with Windows 10 python 3.7 I tired @PabTorre solution (thanks for posting!) but its not working for me! Can someone help please!!

djh-1997, 2022-04-09:
> Hello! I am facing the same issue with Windows 10 python 3.7 I tired @PabTorre solution (thanks for posting!) but its not working for me! Can someone help please!!

You can copy the mujoco_py file(mine is under C:\Users\19240\.mujoco\mujoco-py) to the site-packages(mine is C:\Users\19240\Anaconda3\envs\mujoco\Lib\site-packages), it is useful in my computer, I hope it can help you.