openai/mujoco-py #70: 'import mujoco_py' returns an error

shehroze37, 2017-07-17:
I have installed everything but unfortunately when I import mujoco_py, I get the following error. 

running build_ext
building 'mujoco_py.cymj' extension
gcc -pthread -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -I/home/ubuntu/metis/mujoco-py/mujoco_py -I/home/ubuntu/.mujoco/mjpro150/include -I/home/ubuntu/anaconda2/envs/rllab3/lib/python3.5/site-packages/numpy/core/include -I/home/ubuntu/anaconda2/envs/rllab3/include/python3.5m -c /home/ubuntu/metis/mujoco-py/mujoco_py/cymj.c -o /home/ubuntu/metis/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/home/ubuntu/metis/mujoco-py/mujoco_py/cymj.o -fopenmp -w
gcc -pthread -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -I/home/ubuntu/metis/mujoco-py/mujoco_py -I/home/ubuntu/.mujoco/mjpro150/include -I/home/ubuntu/anaconda2/envs/rllab3/lib/python3.5/site-packages/numpy/core/include -I/home/ubuntu/anaconda2/envs/rllab3/include/python3.5m -c /home/ubuntu/metis/mujoco-py/mujoco_py/gl/osmesashim.c -o /home/ubuntu/metis/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/home/ubuntu/metis/mujoco-py/mujoco_py/gl/osmesashim.o -fopenmp -w
gcc -pthread -shared -L/home/ubuntu/anaconda2/envs/rllab3/lib -Wl,-rpath=/home/ubuntu/anaconda2/envs/rllab3/lib,--no-as-needed /home/ubuntu/metis/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/home/ubuntu/metis/mujoco-py/mujoco_py/cymj.o /home/ubuntu/metis/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/home/ubuntu/metis/mujoco-py/mujoco_py/gl/osmesashim.o -L/home/ubuntu/.mujoco/mjpro150/bin -L/home/ubuntu/anaconda2/envs/rllab3/lib -Wl,--enable-new-dtags,-R/home/ubuntu/.mujoco/mjpro150/bin -lmujoco150 -lglewosmesa -lOSMesa -lGL -lpython3.5m -o /home/ubuntu/metis/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/lib.linux-x86_64-3.5/mujoco_py/cymj.cpython-35m-x86_64-linux-gnu.so -fopenmp
/usr/bin/ld: cannot find -lGL
collect2: error: ld returned 1 exit status
Traceback (most recent call last):
  File "/home/ubuntu/anaconda2/envs/rllab3/lib/python3.5/distutils/unixccompiler.py", line 196, in link
    self.spawn(linker + ld_args)
  File "/home/ubuntu/anaconda2/envs/rllab3/lib/python3.5/distutils/ccompiler.py", line 909, in spawn
    spawn(cmd, dry_run=self.dry_run)
  File "/home/ubuntu/anaconda2/envs/rllab3/lib/python3.5/distutils/spawn.py", line 36, in spawn
    _spawn_posix(cmd, search_path, dry_run=dry_run)
  File "/home/ubuntu/anaconda2/envs/rllab3/lib/python3.5/distutils/spawn.py", line 159, in _spawn_posix
    % (cmd, exit_status))
distutils.errors.DistutilsExecError: command 'gcc' failed with exit status 1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/ubuntu/metis/mujoco-py/mujoco_py/__init__.py", line 1, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/home/ubuntu/metis/mujoco-py/mujoco_py/builder.py", line 296, in <module>
    cymj = load_cython_ext(mjpro_path)
  File "/home/ubuntu/metis/mujoco-py/mujoco_py/builder.py", line 56, in load_cython_ext
    cext_so_path = builder.build()
  File "/home/ubuntu/metis/mujoco-py/mujoco_py/builder.py", line 119, in build
    built_so_file_path = self._build_impl()
  File "/home/ubuntu/metis/mujoco-py/mujoco_py/builder.py", line 139, in _build_impl
    dist.run_commands()
  File "/home/ubuntu/anaconda2/envs/rllab3/lib/python3.5/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/home/ubuntu/anaconda2/envs/rllab3/lib/python3.5/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/home/ubuntu/anaconda2/envs/rllab3/lib/python3.5/site-packages/Cython/Distutils/old_build_ext.py", line 185, in run
    _build_ext.build_ext.run(self)
  File "/home/ubuntu/anaconda2/envs/rllab3/lib/python3.5/distutils/command/build_ext.py", line 338, in run
    self.build_extensions()
  File "/home/ubuntu/metis/mujoco-py/mujoco_py/builder.py", line 77, in build_extensions
    build_ext.build_extensions(self)
  File "/home/ubuntu/anaconda2/envs/rllab3/lib/python3.5/site-packages/Cython/Distutils/old_build_ext.py", line 193, in build_extensions
    self.build_extension(ext)
  File "/home/ubuntu/anaconda2/envs/rllab3/lib/python3.5/distutils/command/build_ext.py", line 557, in build_extension
    target_lang=language)
  File "/home/ubuntu/anaconda2/envs/rllab3/lib/python3.5/distutils/ccompiler.py", line 717, in link_shared_object
    extra_preargs, extra_postargs, build_temp, target_lang)
  File "/home/ubuntu/anaconda2/envs/rllab3/lib/python3.5/distutils/unixccompiler.py", line 198, in link
    raise LinkError(msg)
distutils.errors.LinkError: command 'gcc' failed with exit status 1

Any ideas of whats going wrong here?

Cheers!

machinaut, 2017-07-21:
Your GL library was not found during build: `/usr/bin/ld: cannot find -lGL`.  Do you have an opengl library (e.g. mesa) installed?

xiachongkun, 2017-08-08:
I also have the same mistake!! I looked for many solutions ,but they did't work....

xuemei-ye, 2018-04-15:
I meeted the same problem：
[image]

But the libGL.so is there：
[image]

I don't kown why this happens, Is it the wrong soft link or version error?  maybe you can give me some advice, thanks!

jacky-liang, 2018-10-01:
I'm having the same issue...did any of you find a fix to this problem? 

I'm on Ubuntu 16.04, Python 3.7

I verify that I have  libgl1-mesa-dev libosmesa6-dev libgl1-mesa-glx installed, and that mujoco itself works.

Peter-zds, 2021-05-13:
I have tried `sudo ln -s /usr/lib/x86_64-linux-gnu/libGL.so.1 /usr/lib/x86_64-linux-gnu/libGL.so` in the readme.md and it works