openai/mujoco-py #71: Error while installing with pip

shehroze37, 2017-07-17:
I get the following error while installing with pip as shown in the readme

Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-ojvarseg/mujoco-py/setup.py", line 52, in <module>
        cmdclass={'build': Build},
      File "/usr/lib/python3.4/distutils/core.py", line 148, in setup
        dist.run_commands()
      File "/usr/lib/python3.4/distutils/dist.py", line 955, in run_commands
        self.run_command(cmd)
      File "/usr/lib/python3.4/distutils/dist.py", line 974, in run_command
        cmd_obj.run()
      File "/usr/local/lib/python3.4/dist-packages/setuptools/command/install.py", line 61, in run
        return orig.install.run(self)
      File "/usr/lib/python3.4/distutils/command/install.py", line 583, in run
        self.run_command('build')
      File "/usr/lib/python3.4/distutils/cmd.py", line 313, in run_command
        self.distribution.run_command(command)
      File "/usr/lib/python3.4/distutils/dist.py", line 974, in run_command
        cmd_obj.run()
      File "/tmp/pip-build-ojvarseg/mujoco-py/setup.py", line 18, in run
        builder = importlib.util.module_from_spec(spec)
    AttributeError: 'module' object has no attribute 'module_from_spec'

hmishra2250, 2017-07-20:
It seems from your log that you are using python 3.4. But as mentioned in [Readme](https://github.com/openai/mujoco-py/blob/master/README.md), only python 3.5.2 is supported, so you might need to upgrade your python.
Read the instructions carefully.

inksci, 2017-10-13:
@shehroze37 you can use 'pip install ' within conda, where python 3.5.2 is installed.

kishanvarma, 2018-02-13:
Did this problem get solved ?

kirk86, 2018-03-14:
We need support for python 3.6.x asap!

Derekabc, 2019-04-15:
I am using Python=3.6, Gym =0.10.5 and mujoco-py=0.5.7, it fails. After degrading to gym=0.9.1 or 0.10, it does not work?  Any body solved this?