openai/mujoco-py #74: Install failed on OS X, Anaconda 3.5.2

yo-shavit, 2017-07-20:
Hi there,
Below are logs and error messages I get when attempting the pip installation of mujoco-py.
I'm running OS X 10.10.5, in a conda env with python 3.5.2

```
`$ pip install mujoco-py
Collecting mujoco-py
  Using cached mujoco-py-1.50.1.9.tar.gz
Requirement already satisfied: glfw>=1.4.0 in /Users/yonadav/anaconda/envs/tensorflow3.5/lib/python3.5/site-packages (from mujoco-py)
Requirement already satisfied: numpy>=1.11 in /Users/yonadav/anaconda/envs/tensorflow3.5/lib/python3.5/site-packages (from mujoco-py)
Requirement already satisfied: Cython>=0.25.2 in /Users/yonadav/anaconda/envs/tensorflow3.5/lib/python3.5/site-packages (from mujoco-py)
Requirement already satisfied: imageio>=2.1.2 in /Users/yonadav/anaconda/envs/tensorflow3.5/lib/python3.5/site-packages (from mujoco-py)
Requirement already satisfied: pillow in /Users/yonadav/anaconda/envs/tensorflow3.5/lib/python3.5/site-packages (from imageio>=2.1.2->mujoco-py)
Requirement already satisfied: olefile in /Users/yonadav/anaconda/envs/tensorflow3.5/lib/python3.5/site-packages (from pillow->imageio>=2.1.2->mujoco-py)
Building wheels for collected packages: mujoco-py
  Running setup.py bdist_wheel for mujoco-py ... error
  Complete output from command /Users/yonadav/anaconda/envs/tensorflow3.5/bin/python -u -c "import setuptools, tokenize;__file__='/private/var/folders/0c/fvvc6btj6058m1ns1f0ls6gm0000gn/T/pip-build-gs22rky5/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /var/folders/0c/fvvc6btj6058m1ns1f0ls6gm0000gn/T/tmptoa_mos3pip-wheel- --python-tag cp35:
  running bdist_wheel
  running build
  Compiling /private/var/folders/0c/fvvc6btj6058m1ns1f0ls6gm0000gn/T/pip-build-gs22rky5/mujoco-py/mujoco_py/cymj.pyx because it changed.
  [1/1] Cythonizing /private/var/folders/0c/fvvc6btj6058m1ns1f0ls6gm0000gn/T/pip-build-gs22rky5/mujoco-py/mujoco_py/cymj.pyx
  running build_ext
  building 'mujoco_py.cymj' extension
  creating /private/var/folders/0c/fvvc6btj6058m1ns1f0ls6gm0000gn/T/pip-build-gs22rky5/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder
  creating /private/var/folders/0c/fvvc6btj6058m1ns1f0ls6gm0000gn/T/pip-build-gs22rky5/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-x86_64-3.5
  creating /private/var/folders/0c/fvvc6btj6058m1ns1f0ls6gm0000gn/T/pip-build-gs22rky5/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-x86_64-3.5/private
  creating /private/var/folders/0c/fvvc6btj6058m1ns1f0ls6gm0000gn/T/pip-build-gs22rky5/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-x86_64-3.5/private/var
  creating /private/var/folders/0c/fvvc6btj6058m1ns1f0ls6gm0000gn/T/pip-build-gs22rky5/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-x86_64-3.5/private/var/folders
  creating /private/var/folders/0c/fvvc6btj6058m1ns1f0ls6gm0000gn/T/pip-build-gs22rky5/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-x86_64-3.5/private/var/folders/0c
  creating /private/var/folders/0c/fvvc6btj6058m1ns1f0ls6gm0000gn/T/pip-build-gs22rky5/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-x86_64-3.5/private/var/folders/0c/fvvc6btj6058m1ns1f0ls6gm0000gn
  creating /private/var/folders/0c/fvvc6btj6058m1ns1f0ls6gm0000gn/T/pip-build-gs22rky5/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-x86_64-3.5/private/var/folders/0c/fvvc6btj6058m1ns1f0ls6gm0000gn/T
  creating /private/var/folders/0c/fvvc6btj6058m1ns1f0ls6gm0000gn/T/pip-build-gs22rky5/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-x86_64-3.5/private/var/folders/0c/fvvc6btj6058m1ns1f0ls6gm0000gn/T/pip-build-gs22rky5
  creating /private/var/folders/0c/fvvc6btj6058m1ns1f0ls6gm0000gn/T/pip-build-gs22rky5/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-x86_64-3.5/private/var/folders/0c/fvvc6btj6058m1ns1f0ls6gm0000gn/T/pip-build-gs22rky5/mujoco-py
  creating /private/var/folders/0c/fvvc6btj6058m1ns1f0ls6gm0000gn/T/pip-build-gs22rky5/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-x86_64-3.5/private/var/folders/0c/fvvc6btj6058m1ns1f0ls6gm0000gn/T/pip-build-gs22rky5/mujoco-py/mujoco_py
  creating /private/var/folders/0c/fvvc6btj6058m1ns1f0ls6gm0000gn/T/pip-build-gs22rky5/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-x86_64-3.5/private/var/folders/0c/fvvc6btj6058m1ns1f0ls6gm0000gn/T/pip-build-gs22rky5/mujoco-py/mujoco_py/gl
  /usr/local/bin/gcc-6 -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -I/Users/yonadav/anaconda/envs/tensorflow3.5/include -arch x86_64 -DONMAC -I/private/var/folders/0c/fvvc6btj6058m1ns1f0ls6gm0000gn/T/pip-build-gs22rky5/mujoco-py/mujoco_py -I/Users/yonadav/.mujoco/mjpro150/include -I/Users/yonadav/anaconda/envs/tensorflow3.5/lib/python3.5/site-packages/numpy/core/include -I/Users/yonadav/anaconda/envs/tensorflow3.5/include/python3.5m -c /private/var/folders/0c/fvvc6btj6058m1ns1f0ls6gm0000gn/T/pip-build-gs22rky5/mujoco-py/mujoco_py/cymj.c -o /private/var/folders/0c/fvvc6btj6058m1ns1f0ls6gm0000gn/T/pip-build-gs22rky5/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-x86_64-3.5/private/var/folders/0c/fvvc6btj6058m1ns1f0ls6gm0000gn/T/pip-build-gs22rky5/mujoco-py/mujoco_py/cymj.o -fopenmp -w
  /usr/local/bin/gcc-6 -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -I/Users/yonadav/anaconda/envs/tensorflow3.5/include -arch x86_64 -DONMAC -I/private/var/folders/0c/fvvc6btj6058m1ns1f0ls6gm0000gn/T/pip-build-gs22rky5/mujoco-py/mujoco_py -I/Users/yonadav/.mujoco/mjpro150/include -I/Users/yonadav/anaconda/envs/tensorflow3.5/lib/python3.5/site-packages/numpy/core/include -I/Users/yonadav/anaconda/envs/tensorflow3.5/include/python3.5m -c /private/var/folders/0c/fvvc6btj6058m1ns1f0ls6gm0000gn/T/pip-build-gs22rky5/mujoco-py/mujoco_py/gl/dummyshim.c -o /private/var/folders/0c/fvvc6btj6058m1ns1f0ls6gm0000gn/T/pip-build-gs22rky5/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-x86_64-3.5/private/var/folders/0c/fvvc6btj6058m1ns1f0ls6gm0000gn/T/pip-build-gs22rky5/mujoco-py/mujoco_py/gl/dummyshim.o -fopenmp -w
  creating /private/var/folders/0c/fvvc6btj6058m1ns1f0ls6gm0000gn/T/pip-build-gs22rky5/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/lib.macosx-10.6-x86_64-3.5
  creating /private/var/folders/0c/fvvc6btj6058m1ns1f0ls6gm0000gn/T/pip-build-gs22rky5/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/lib.macosx-10.6-x86_64-3.5/mujoco_py
  /usr/local/bin/gcc-6 -bundle -undefined dynamic_lookup -L/Users/yonadav/anaconda/envs/tensorflow3.5/lib -arch x86_64 /private/var/folders/0c/fvvc6btj6058m1ns1f0ls6gm0000gn/T/pip-build-gs22rky5/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-x86_64-3.5/private/var/folders/0c/fvvc6btj6058m1ns1f0ls6gm0000gn/T/pip-build-gs22rky5/mujoco-py/mujoco_py/cymj.o /private/var/folders/0c/fvvc6btj6058m1ns1f0ls6gm0000gn/T/pip-build-gs22rky5/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-x86_64-3.5/private/var/folders/0c/fvvc6btj6058m1ns1f0ls6gm0000gn/T/pip-build-gs22rky5/mujoco-py/mujoco_py/gl/dummyshim.o -L/Users/yonadav/.mujoco/mjpro150/bin -L/Users/yonadav/anaconda/envs/tensorflow3.5/lib -L/Users/yonadav/.mujoco/mjpro150/bin -lmujoco150 -lglfw.3 -o /private/var/folders/0c/fvvc6btj6058m1ns1f0ls6gm0000gn/T/pip-build-gs22rky5/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/lib.macosx-10.6-x86_64-3.5/mujoco_py/cymj.cpython-35m-darwin.so -fopenmp
  Traceback (most recent call last):
    File "<string>", line 1, in <module>
    File "/private/var/folders/0c/fvvc6btj6058m1ns1f0ls6gm0000gn/T/pip-build-gs22rky5/mujoco-py/setup.py", line 52, in <module>
      cmdclass={'build': Build},
    File "/Users/yonadav/anaconda/envs/tensorflow3.5/lib/python3.5/distutils/core.py", line 148, in setup
      dist.run_commands()
    File "/Users/yonadav/anaconda/envs/tensorflow3.5/lib/python3.5/distutils/dist.py", line 955, in run_commands
      self.run_command(cmd)
    File "/Users/yonadav/anaconda/envs/tensorflow3.5/lib/python3.5/distutils/dist.py", line 974, in run_command
      cmd_obj.run()
    File "/Users/yonadav/anaconda/envs/tensorflow3.5/lib/python3.5/site-packages/wheel/bdist_wheel.py", line 179, in run
      self.run_command('build')
[... cut ...]
  File "/Users/yonadav/anaconda/envs/tensorflow3.5/lib/python3.5/imp.py", line 342, in load_dynamic
    return _load(spec)
ImportError: dlopen(/Users/yonadav/.mujoco/mujoco-py/mujoco_py/generated/cymj_macextensionbuilder.so, 2): Symbol not found: _GOMP_parallel
  Referenced from: /Users/yonadav/.mujoco/mujoco-py/mujoco_py/generated/cymj_macextensionbuilder.so
  Expected in: flat namespace
 in /Users/yonadav/.mujoco/mujoco-py/mujoco_py/generated/cymj_macextensionbuilder.so
```

Could someone help me work through this bug? Thanks for the help!

machinaut, 2017-07-25:
This is my gcc version, though I'm not sure it's what's wrong in this case:

```
$ gcc --version
gcc (Homebrew GCC 6.4.0 --without-multilib) 6.4.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

If you want to try it (and use homebrew) you can get this version with `brew install gcc@6 --without-multilib`

jmeyers44, 2017-08-08:
I was able to successfully install on OSX 10.12.3 Python 3.5.2 using @machinaut 's gcc version after previously failing installation using `Apple LLVM version 8.0.0 (clang-800.0.42.1)`

mepster, 2017-10-21:
Hmm...

```
$ brew install gcc@6 --without-multilib
```
doesn't seem to install a gcc without multilib.

```
$ brew install gcc@6 --without-multilib
Updating Homebrew...
==> Auto-updated Homebrew!
Updated 1 tap (homebrew/core).
==> New Formulae
swift-protobuf
==> Updated Formulae
youtube-dl

Warning: gcc@6: this formula has no --without-multilib option so it will be ignored!
==> Downloading https://homebrew.bintray.com/bottles/gcc@6-6.4.0.sierra.bottle.tar.gz
[...]
```

```
$ /usr/local/bin/gcc-6 --version
gcc-6 (Homebrew GCC 6.4.0) 6.4.0
```
Has anyone else got this to work? There are some posts on the interwebs about this not working in homebrew... but looks like @machinaut got it to work?

Thanks all.

mepster, 2017-10-21:
I did finally get it to work using gcc@5, specifically:

```
$ brew install cmake boost boost-python sdl2 swig wget
$ pip install 'gym[all]'
$ brew install gcc@5
```

(Thanks to @JoeNyland in this thread https://github.com/dmlc/xgboost/issues/1501 )

Then put this in your .bash_profile (easy to undo later):
```
alias gcc='gcc-5'
alias cc='gcc-5'
alias g++='g++-5'
alias c++='c++-5'
```

Then this works:

```
pip install 'mujoco-py<1.50.2,>=1.50.1'
```

(Don't forget to undo .bash_profile changes.)

Now unfortunately I have hit the "No module named 'mujoco_py.mjlib'" issue! https://github.com/openai/mujoco-py/issues/80 :-(

All of the above was with Python 3.5.4 and mujoco150. I can't use mujoco131 because I have a new Mac and it fails with "ERROR: Could not open disk" as here: https://github.com/openai/mujoco-py/issues/36 https://github.com/openai/gym/issues/638

mixuala, 2017-11-06:
It took me awhile, but the above comment helped me get it working for OSX. This is what I did:

**openAI/gym install on OSX 10.12.6,  with mujoco**
```
# create conda env with python v3.5.2
# https://github.com/openai/mujoco-py/issues/47#issuecomment-312551946
conda create --name gym python=3.5.2
source activate gym
mkdir -p ~/openai && cd ~/openai

# install gym
# https://github.com/openai/mujoco-py/issues/74#issuecomment-338410266
brew install cmake boost boost-python sdl2 swig wget

# see: https://github.com/openai/gym/issues/164#issuecomment-263785233
export MACOSX_DEPLOYMENT_TARGET=10.12
# pip install 'gym[all]'
git clone https://github.com/openai/gym.git
cd ./gym
pip install -e '.[all]'
brew install gcc@5

## use gcc5
# add to ~/.bash_profile
alias gcc='gcc-5'
alias cc='gcc-5'
alias g++='g++-5'
alias c++='c++-5'
# source ~/.bash_profile

## install mujoco v1.50 binaries
cd ~/openai
mkdir ~/.mujoco
# NOTE: remember to copy mujoco license key to ~/.mujoco/mjkey.txt

# download OSX binaries
wget https://www.roboti.us/download/mjpro150_osx.zip
unzip mjpro150_osx.zip && mv mjpro150  ~/.mujoco  

# wget https://www.roboti.us/download/mjpro131_osx.zip
# unzip mjpro131_osx.zip && mv mjpro131  ~/.mujoco 
git clone https://github.com/openai/mujoco-py.git
    
# pwd: cd ~/openai
rm -rf mujoco-py
git clone https://github.com/openai/mujoco-py.git
git checkout master             # mujoco v1.50

cd mujoco-py && python setup.py install
```

And you can run this script in python to verify
```
import gym
env = gym.make('Hopper-v1')
print(env.observation_space)
print(env.action_space)
env.reset()
for _ in range(100):
    env.render()
    env.step(env.action_space.sample()) # take a random action        
```

tshrjn, 2017-11-27:
@mixuala I'm OSX 10.11.3 and I'm having the same error even after following your steps.

mixuala, 2017-11-27:
did you install `mujoco-py` from the cloned git repo?  I did not install using `pip`

tshrjn, 2017-11-27:
Yes, I did but that did not help. Finally resolved all problems by installing `0.5.7` via pip.

```
pip install mujoco-py==0.5.7
```

richardrl, 2018-01-27:
Although I can install .5.7, functions are missing. For example, 'load_model_from_path' which is only present in latest mujoco_py 1.5. Anyone get that working?

erwincoumans, 2018-02-01:
What are the up-to-date instructions to install mujoco_py on Mac OSX 
(my version is 10.13.4, but errors are the same: 
/cymj_macextensionbuilder.so, 2): Symbol not found: _GOMP_parallel

mixuala, 2018-02-02:
I just installed again on a new Mac-mini with the latest 10.13. It worked fine according to the script I posted above. But I had to use the 1.3.1 version—1.5.0 did not work.

nemanja-rakicevic, 2018-02-17:
check if this helps https://github.com/openai/mujoco-py/issues/80#issuecomment-366438347

f1recracker, 2018-03-08:
If it helps, I resolved the `Symbol not found: _GOMP_parallel)` problem by using Python 3.5.3 (I was actually using Python 3.6 all this while which is currently not supported). I was also able to install mujoco_py directly from pip (MacOS 10.13.3, conda).

azmosavi, 2018-04-05:
@mixuala Thanks it was really helpful.

billtubbs, 2018-05-02:
@mixuala's install script worked for me on  OS X 10.13.4 with python 3.6.5, gym 0.10.5, and mujoco 1.50.1.56.  Although the coursework I am doing says I have to install 1.3.1.

[> Note that we use version 1.31 and not
> the latest version 1.50. The python interface to mujoco bundled with OpenAI gym is also compatible
> with this version of mujoco.](http://rll.berkeley.edu/deeprlcourse/f17docs/hw1fall2017.pdf)

Does anyone know if this is still an issue or whether it will now work with 1.5.1?  I will try it...

zjzijielu, 2018-05-24:
Hi @billtubbs, I'm also struggling with the same issue. Did you get it to work with mujoco 1.5?

ammarhydr, 2019-03-04:
> It took me awhile, but the above comment helped me get it working for OSX. This is what I did:
>
> **openAI/gym install on OSX 10.12.6, with mujoco**
>
> ```
> # create conda env with python v3.5.2
> # [...]

After applying all these steps on MacOS 10.14 I am having error below: 

`[2019-03-03 19:12:17,075] Making new env: InvertedPendulum-v1
running build_ext
building 'mujoco_py.cymj' extension
x86_64-apple-darwin13.4.0-clang -DNDEBUG -fwrapv -O2 -Wall -march=core2 -mtune=haswell -mssse3 -ftree-vectorize -fPIC -fPIE -fstack-protector-strong -O2 -pipe -D_FORTIFY_SOURCE=2 -mmacosx-version-min=10.9 -DONMAC -Imujoco_py -I/Users/ammarhaydari/openai/mujoco-py/mujoco_py -I/Users/ammarhaydari/.mujoco/mjpro150/include -I/anaconda3/lib/python3.6/site-packages/numpy/core/include -I/anaconda3/include/python3.6m -c /Users/ammarhaydari/openai/mujoco-py/mujoco_py/cymj.c -o /Users/ammarhaydari/openai/mujoco-py/mujoco_py/generated/_pyxbld_1.50.1.68_36_macextensionbuilder/temp.macosx-10.9-x86_64-3.6/Users/ammarhaydari/openai/mujoco-py/mujoco_py/cymj.o -fopenmp -w
/Users/ammarhaydari/openai/mujoco-py/mujoco_py/cymj.c:632:10: fatal error: 'omp.h' file not found
#include <omp.h>
         ^~~~~~~
1 error generated.
Traceback (most recent call last):
  File "/anaconda3/lib/python3.6/distutils/unixccompiler.py", line 118, in _compile
    extra_postargs)
  File "/anaconda3/lib/python3.6/distutils/ccompiler.py", line 909, in spawn
    spawn(cmd, dry_run=self.dry_run)
  File "/anaconda3/lib/python3.6/distutils/spawn.py", line 36, in spawn
    _spawn_posix(cmd, search_path, dry_run=dry_run)
  File "/anaconda3/lib/python3.6/distutils/spawn.py", line 159, in _spawn_posix
    % (cmd, exit_status))
distutils.errors.DistutilsExecError: command 'x86_64-apple-darwin13.4.0-clang' failed with exit status 1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/ammarhaydari/Rl_Files/ddpg-aigym-master/main.py", line 72, in <module>
    main()    
  File "/Users/ammarhaydari/Rl_Files/ddpg-aigym-master/main.py", line 16, in main
    env= gym.make(experiment)
  File "/anaconda3/lib/python3.6/site-packages/gym/envs/registration.py", line 140, in make
    return registry.make(id)
  File "/anaconda3/lib/python3.6/site-packages/gym/envs/registration.py", line 105, in make
    return spec.make()
  File "/anaconda3/lib/python3.6/site-packages/gym/envs/registration.py", line 69, in make
    cls = load(self._entry_point)
  File "/anaconda3/lib/python3.6/site-packages/gym/envs/registration.py", line 18, in load
    result = entry_point.load(False)
  File "/anaconda3/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2322, in load
    return self.resolve()
  File "/anaconda3/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2328, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/anaconda3/lib/python3.6/site-packages/gym/envs/mujoco/__init__.py", line 1, in <module>
    from gym.envs.mujoco.mujoco_env import MujocoEnv 
  File "/anaconda3/lib/python3.6/site-packages/gym/envs/mujoco/mujoco_env.py", line 11, in <module>
    import mujoco_py
  File "/Users/ammarhaydari/openai/mujoco-py/mujoco_py/__init__.py", line 3, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/Users/ammarhaydari/openai/mujoco-py/mujoco_py/builder.py", line 503, in <module>
    cymj = load_cython_ext(mjpro_path)
  File "/Users/ammarhaydari/openai/mujoco-py/mujoco_py/builder.py", line 105, in load_cython_ext
    cext_so_path = builder.build()
  File "/Users/ammarhaydari/openai/mujoco-py/mujoco_py/builder.py", line 220, in build
    built_so_file_path = self._build_impl()
  File "/Users/ammarhaydari/openai/mujoco-py/mujoco_py/builder.py", line 329, in _build_impl
    so_file_path = super()._build_impl()
  File "/Users/ammarhaydari/openai/mujoco-py/mujoco_py/builder.py", line 243, in _build_impl
    dist.run_commands()
  File "/anaconda3/lib/python3.6/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/anaconda3/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/anaconda3/lib/python3.6/site-packages/Cython/Distutils/old_build_ext.py", line 186, in run
    _build_ext.build_ext.run(self)
  File "/anaconda3/lib/python3.6/distutils/command/build_ext.py", line 339, in run
    self.build_extensions()
  File "/Users/ammarhaydari/openai/mujoco-py/mujoco_py/builder.py", line 143, in build_extensions
    build_ext.build_extensions(self)
  File "/anaconda3/lib/python3.6/site-packages/Cython/Distutils/old_build_ext.py", line 194, in build_extensions
    self.build_extension(ext)
  File "/anaconda3/lib/python3.6/distutils/command/build_ext.py", line 533, in build_extension
    depends=ext.depends)
  File "/anaconda3/lib/python3.6/distutils/ccompiler.py", line 574, in compile
    self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
  File "/anaconda3/lib/python3.6/distutils/unixccompiler.py", line 120, in _compile
    raise CompileError(msg)
distutils.errors.CompileError: command 'x86_64-apple-darwin13.4.0-clang' failed with exit status 1`

When I use spider in this gym venv, I have another error which is `error NameError: name 'runfile' is not defined`. I am not sure is this a right place to ask this error but I wanted to include it in this comment. 

If anyone can help me I would appreicate