openai/mujoco-py #75: Ubuntu 16.04 Python3.5: ERROR: GLEW initalization error: Missing GL version

iretiayo, 2017-07-20:
I installed mujocopy on Ubuntu 16.04 machine but I've been having issues using it or even running the example codes of mujocopy. I keep getting the error:
**_ERROR: GLEW initalization error: Missing GL version_**

Could you advise on how to fix this? I've spent quite some time on it but had no success.

hmishra2250, 2017-07-20:
It's a work in progress.
Track it [here](https://github.com/openai/mujoco-py/issues/44).

Jerryxiaoyu, 2017-12-05:
Hi @iretiayo ,I'm getting the same problem...How can you fix it?

etendue, 2018-03-29:
Nope, still get this error. native mujoco runs correctly.

jeremyf21, 2018-04-24:
I'm posting a solution here from [this thread](https://github.com/openai/mujoco-py/pull/145)

`LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libGLEW.so:/usr/lib/nvidia-384/libGL.so python your_file.py` 

I will add that line to my .bashrc and hopefully that should solve it.

xlrobotics, 2018-05-23:
@jeremyf21 Your solution works perfectly, thanks a lot!

Jin0932, 2018-06-10:
the method is: https://blog.csdn.net/gsww404/article/details/80636676

yaogang2018, 2018-07-15:
I have no idea why my Ubuntu 18.04 has no /usr/lib/nvidia-384 folder,following @jeremyf21 and @Jin0932 I add `export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libGLEW.so:/usr/lib/x86_64-linux-gnu/libGL.so.1`to .bashrc,and this error is gone.

inspirepassion, 2018-09-06:
To update what's happening to my case for 
@jeremyf21 's reply https://github.com/openai/mujoco-py/issues/75#issuecomment-383982069
1 I don't have libGLEW.so under /usr/lib/x86_64-linux-gnu/ in my file system (Ubuntu 18.04)
Solution: `sudo apt-get install libglew-dev`, I read it from this issue report: https://github.com/raulmur/ORB_SLAM2/issues/210

2 I don't have libGL.so under /usr/lib/nvidia-39*/ (I don't even have this folder), but when I searched the whole file system, libGL.so is actually located in /usr/lib/x86_64-linux-gnu
Solution: 
1) open .bashrc `sudo gedit ~/.bashrc`
2) add this line at the end `export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libGLEW.so:/usr/lib/x86_64-linux-gnu/libGL.so` (this is just a modification to people's previous reply, and I think some people might encounter similar situation)
3) save it, and restart the terminal, and WOLA~

manchengfenxu, 2018-11-10:
@jeremyf21 ,Your solution works perfectly, thanks a lot!

beingkartik, 2019-05-08:
@jeremyf21, thanks! for me, just adding `export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libGLEW.so` to my bashrc file worked

JiyueWang, 2019-10-20:
I hope I can help someone too.
I found out that I have no libGLEW in /usr/lib/x86_64-linux-gnu/ so I install the lib by 
$ sudo apt-get install libglew2.0
after that I add this line to .bashrc
export LD_PRELOAD=/usr/lib/x86_64-Linux-gnu/libGLEW.so.2.0:/usr/lib/x86_64-Linux-gnu/libGL.so 
Make sure that you have both libs in the right version and file name in the path above. Good luck.

wkoa, 2019-12-24:
it's not work in my situation, 
- Ubuntu 18.04
- CUDA 10.1 with GPU: Nvidia 1060
- Mujoco-py 1.50.1.68
- Mujoco 1.5
I have added the line export LD_PRELOAD=/usr/lib/x86_64-Linux-gnu/libGLEW.so.2.0:/usr/lib/x86_64-Linux-gnu/libGL.so to .bashrc,
but it seems failed