openai/mujoco-py #79: Cannot open include file: 'mjmodel.h' error on Windows 10

ViktorM, 2017-07-22:
Hi,

I can't get mujoco-py working on Windows 10. After import mujoco_py I always get the same error:

`C:\Users\Viktor\Anaconda3\lib\site-packages\mujoco_py-1.50.1.10-py3.5.egg\mujoco_py\cymj.c(511): fatal error C1083: Cannot open include file: 'mjmodel.h': No such file or directory`

I tried different installation variants including pip install but got always the same result. Mujoco is installed on my PC and simulate of human model is working. Will be thankful for any advises how to make it finally working.

P. S. I've seen similar issues open here in past but no one of them was helpful.

machinaut, 2017-07-25:
Can you verify the headers are installed where the builder is looking for them.

If your home directory is `C:\Users\Viktor` then the header files should be in `C:\Users\Viktor\.mujoco\mjpro150\include`.  Is there a `mjmodel.h` file there?

ViktorM, 2017-07-26:
@machinaut thanks for advice. There was an error with folder placement. I was able to move further and run test script and body_interaction.py script from examples folder. So it is working at least. But got a model load errors for disco_fetch.py and tosser.py : 
Traceback (most recent call last):
```
  File "disco_fetch.py", line 9, in <module>
    model = load_model_from_path("xmls/fetch/main.xml")
  File "mujoco_py\cymj.pyx", line 113, in mujoco_py.cymj.load_model_from_path
Exception: Failed to load XML file: xmls/fetch/main.xml. mj_loadXML error: b'XML parse error at line 0, column 0:\nFailed to open file\n'
```

ViktorM, 2017-07-26:
Also found another source of the problem. I need Mujoco for locomotion experiments with gym, but currently it doesn't support version 1.5: https://github.com/openai/gym/issues/665

ViktorM, 2017-07-26:
Ok, I can run disco_fetch or tosser if I write the full path to the xml models like:
`model = load_model_from_path("D:/mujoco-py/xmls/fetch/main.xml")`

So this error isn't related to the mujoco-py

The only problem left is Gym compatibility with 1.5

machinaut, 2017-07-26:
updating gym is planned (tracked elsewhere).

closed 2017-07-26 by machinaut

abhigenie92, 2017-11-12:
@machinaut  Could you have a look https://github.com/openai/mujoco-py/issues/168?