openai/mujoco-py #80: No module named 'mujoco_py.mjlib'

yayitsamyzhang, 2017-07-24:
Looks like gym is not compatible with the latest mujoco-py version.  When will it be updated/what do I use to replace the mjlib dependency?

machinaut, 2017-07-25:
Updating the gym mujoco environments is on the roadmap, but for now gym's mujoco environments use the 0.5 version of mujoco-py.  (gym itself doesn't require mujoco-py, though)

https://github.com/openai/gym/blob/master/setup.py#L14 calls out the specific version needs.

It's recommended to use python environments to have separate versions of mujoco-py for different uses.

yayitsamyzhang, 2017-07-25:
I'm using the mujoco envs in gym which do use the mujoco-py bindings, and
I'd like to switch to the latest version so I can make use of the headless
rendering on a remote server.  Any idea if that's an easy change I can make
myself?

machinaut, 2017-07-25:
I think so.  The mujoco environment wrappers (in gym) would need to be updated, as internal state (e.g. `qpos`, `qvel`, etc) is accessed differently.

the XML files can stay the same I think.

yayitsamyzhang, 2017-07-27:
So I replaced mjlib and model.data with the new versions pymjmodel and
mjsim (I think) but am again getting glfw errors:
  File
"/opt/conda/envs/pytorch-py35/lib/python3.5/site-packages/mujoco_py-1.50.1.10-py3.5.egg/mujoco_py/mjviewer.py",
line 132, in __init__
    super().__init__(sim)
  File
"/opt/conda/envs/pytorch-py35/lib/python3.5/site-packages/mujoco_py-1.50.1.10-py3.5.egg/mujoco_py/mjviewer.py",
line 26, in __init__
    super().__init__(sim)
  File "mujoco_py/mjrendercontext.pyx", line 246, in
mujoco_py.cymj.MjRenderContextWindow.__init__
  File "mujoco_py/mjrendercontext.pyx", line 43, in
mujoco_py.cymj.MjRenderContext.__init__
  File "mujoco_py/mjrendercontext.pyx", line 92, in
mujoco_py.cymj.MjRenderContext._setup_opengl_context
  File "mujoco_py/opengl_context.pyx", line 44, in
mujoco_py.cymj.GlfwContext.__init__
  File "mujoco_py/opengl_context.pyx", line 64, in
mujoco_py.cymj.GlfwContext._init_glfw
mujoco_py.cymj.GlfwError: Failed to initialize GLFW

so I assume in the new mujoco_py model and sim got explicitly separated to
enable headless rendering, but I am not familiar enough with the code to
understand how I should change the gym code to support this...and I
probably just re-mucked it up. Any tips? I'm just going through
the mujoco/mujoco_env.py file and fixing all the parts that broke with new
mujoco_py...

adamlerer, 2017-09-05:
We got this working, we'll try to push a PR soon if anyone's interested.

emerrf, 2017-09-19:
@adamlerer great news, any update about the status of the PR?

DoraXingyu, 2017-09-26:
Any update? This would be very helpful to me

adamlerer, 2017-09-27:
I extracted the relevant changes into this branch (co-authored with @yayitsamyzhang ):

https://github.com/adamlerer/gym/tree/mujoco-1.0

I don't have any test cases except an internal environment that we are working on, so I don't know if it will be accepted as a PR yet, but you can fork that branch and try it out. Let me know if it works for you!

DoraXingyu, 2017-09-28:
@adamlerer @yayitsamyzhang That's great news. I tried the mujoco-1.0 branch, but got the follow error: 
2017-09-27 22:03:51,017] Making new env: Humanoid-v1
Traceback (most recent call last):
  File "test.py", line 2, in <module>
    env = gym.make('Humanoid-v1')
  File "/Users/Dora/Projects/gym/gym/envs/registration.py", line 164, in make
    return registry.make(id)
  File "/Users/Dora/Projects/gym/gym/envs/registration.py", line 122, in make
    env = spec.make()
  File "/Users/Dora/Projects/gym/gym/envs/registration.py", line 89, in make
    env = cls(**self._kwargs)
  File "/Users/Dora/Projects/gym/gym/envs/mujoco/humanoid.py", line 12, in __init__
    mujoco_env.MujocoEnv.__init__(self, 'humanoid.xml', 5)
  File "/Users/Dora/Projects/gym/gym/envs/mujoco/mujoco_env.py", line 44, in __init__
    observation, _reward, done, _info = self._step(np.zeros(self.model.nu))
  File "/Users/Dora/Projects/gym/gym/envs/mujoco/humanoid.py", line 25, in _step
    pos_before = mass_center(self.model)
  File "/Users/Dora/Projects/gym/gym/envs/mujoco/humanoid.py", line 7, in mass_center
    xpos = model.data.xipos
AttributeError: 'mujoco_py.cymj.PyMjModel' object has no attribute 'data'

jalexvig, 2017-10-27:
Any progress on this?

adamlerer, 2017-10-28:
@jalexvig progress on what?

@DoraXingyu I didn't update the individual envs; I think that `self.model.data` is now `self.sim.data`, so you'll have to fix up the envs to accomodate this. It should be pretty straightforward if you take a look at the code changes here https://github.com/openai/gym/pull/733/files

jalexvig, 2017-11-01:
@adamlerer Wondering if anyone had been testing this/ working on getting the PR merged

abhigenie92, 2017-11-12:
@adamlerer Sorry, not sure what you meant by got this working.

```
$ python
Python 3.6.1 |Anaconda custom (64-bit)| (default, May 11 2017, 13:09:58)
[GCC 4.4.7 20120313 (Red Hat 4.4.7-1)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import mujoco_py
>>> from os.path import dirname
>>> model = mujoco_py.load_model_from_path(dirname(dirname(mujoco_py.__file__))  +"/xmls/claw.xml")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "mujoco_py/cymj.pyx", line 132, in mujoco_py.cymj.load_model_from_path
Exception: Failed to load XML file: /nohome/jaan/abhishek/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.31-py3.6.egg/xmls/claw.xml. mj_loadXML error: b'XML parse error at line 0, column 0:\nFailed to open file\n'
```

abhigenie92, 2017-11-12:
@adamlerer Hi, I tried your gym version, but am getting the error. If possible could you suggest any directions.
```
python -c "import mujoco_py
                                   from os.path import dirname
                                   model = mujoco_py.load_model_from_path(dirname(dirname(mujoco_py.__file__))  +'/xmls/claw.xml')
                                   sim = mujoco_py.MjSim(model)"
Traceback (most recent call last):
  File "<string>", line 3, in <module>
  File "mujoco_py/cymj.pyx", line 132, in mujoco_py.cymj.load_model_from_path
Exception: Failed to load XML file: /nohome/jaan/abhishek/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.31-py3.6.egg/xmls/claw.xml. mj_loadXML error: b'XML parse error at line 0, column 0:\nFailed to open file\n'
 ✘  gonzo  ~/a/p/m/gym   master   python
Python 3.6.1 |Anaconda custom (64-bit)| (default, May 11 2017, 13:09:58)
[GCC 4.4.7 20120313 (Red Hat 4.4.7-1)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import gym
>>> exit()
 gonzo  ~/a/p/m/gym   master   ipython
Python 3.6.1 |Anaconda custom (64-bit)| (default, May 11 2017, 13:09:58)
Type "copyright", "credits" or "license" for more information.

IPython 5.3.0 -- An enhanced Interactive Python.
?         -> Introduction and overview of IPython's features.
%quickref -> Quick reference.
help      -> Python's own help system.
object?   -> Details about 'object', use 'object??' for extra details.

In [1]: import gym

In [2]: gym.make("Hopper-v1")
[2017-11-12 00:24:55,044] Making new env: Hopper-v1
---------------------------------------------------------------------------
ModuleNotFoundError                       Traceback (most recent call last)
/nohome/jaan/abhishek/packages/mujoco_diff/gym/gym/envs/mujoco/mujoco_env.py in <module>()
     11     import mujoco_py
---> 12     from mujoco_py.mjlib import mjlib
     13 except ImportError as e:

ModuleNotFoundError: No module named 'mujoco_py.mjlib'

During handling of the above exception, another exception occurred:

DependencyNotInstalled                    Traceback (most recent call last)
<ipython-input-2-15e33d66bf45> in <module>()
----> 1 gym.make("Hopper-v1")

/nohome/jaan/abhishek/packages/mujoco_diff/gym/gym/envs/registration.py in make(id)
    162
    163 def make(id):
--> 164     return registry.make(id)
    165
    166 def spec(id):

/nohome/jaan/abhishek/packages/mujoco_diff/gym/gym/envs/registration.py in make(self, id)
    120         logger.info('Making new env: %s', id)
    121         spec = self.spec(id)
--> 122         env = spec.make()
    123         if (env.spec.timestep_limit is not None) and not spec.tags.get('vnc'):
    124             from gym.wrappers.time_limit import TimeLimit

/nohome/jaan/abhishek/packages/mujoco_diff/gym/gym/envs/registration.py in make(self)
     86             env = self._entry_point()
     87         else:
---> 88             cls = load(self._entry_point)
     89             env = cls(**self._kwargs)
     90

/nohome/jaan/abhishek/packages/mujoco_diff/gym/gym/envs/registration.py in load(name)
     15 def load(name):
     16     entry_point = pkg_resources.EntryPoint.parse('x={}'.format(name))
---> 17     result = entry_point.load(False)
     18     return result
     19

/nohome/jaan/abhishek/anaconda3/lib/python3.6/site-packages/pkg_resources/__init__.py in load(self, require, *args, **kwargs)
   2314         if require:
   2315             self.require(*args, **kwargs)
-> 2316         return self.resolve()
   2317
   2318     def resolve(self):

/nohome/jaan/abhishek/anaconda3/lib/python3.6/site-packages/pkg_resources/__init__.py in resolve(self)
   2320         Resolve the entry point from its module and attrs.
   2321         """
-> 2322         module = __import__(self.module_name, fromlist=['__name__'], level=0)
   2323         try:
   2324             return functools.reduce(getattr, self.attrs, module)

/nohome/jaan/abhishek/packages/mujoco_diff/gym/gym/envs/mujoco/__init__.py in <module>()
----> 1 from gym.envs.mujoco.mujoco_env import MujocoEnv
      2 # ^^^^^ so that user gets the correct error
      3 # message if mujoco is not installed correctly
      4 from gym.envs.mujoco.ant import AntEnv
      5 from gym.envs.mujoco.half_cheetah import HalfCheetahEnv

/nohome/jaan/abhishek/packages/mujoco_diff/gym/gym/envs/mujoco/mujoco_env.py in <module>()
     12     from mujoco_py.mjlib import mjlib
     13 except ImportError as e:
---> 14     raise error.DependencyNotInstalled("{}. (HINT: you need to install mujoco_py, and also perform the setup instructions here: https://github.com/openai/mujoco-py/.)".format(e))
     15
     16 class MujocoEnv(gym.Env):

DependencyNotInstalled: No module named 'mujoco_py.mjlib'. (HINT: you need to install mujoco_py, and also perform the setup instructions here: https://github.com/openai/mujoco-py/.)
```

yayitsamyzhang, 2017-11-12:
Did you install mujoco_py?

stevenzyang, 2017-11-18:
Getting this error too
```
>>> import mujoco_py
>>> import gym
>>> env = gym.make("InvertedPendulum-v1")
Traceback (most recent call last):
  File "C:\Users\Steven\Anaconda3\lib\site-packages\gym\envs\mujoco\mujoco_env.py", line 12, in <module>
    from mujoco_py.mjlib import mjlib
ImportError: No module named 'mujoco_py.mjlib'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\Steven\Anaconda3\lib\site-packages\gym\envs\registration.py", line 164, in make
    return registry.make(id)
  File "C:\Users\Steven\Anaconda3\lib\site-packages\gym\envs\registration.py", line 122, in make
    env = spec.make()
  File "C:\Users\Steven\Anaconda3\lib\site-packages\gym\envs\registration.py", line 88, in make
    cls = load(self._entry_point)
  File "C:\Users\Steven\Anaconda3\lib\site-packages\gym\envs\registration.py", line 17, in load
    result = entry_point.load(False)
  File "C:\Users\Steven\Anaconda3\lib\site-packages\pkg_resources\__init__.py", line 2409, in load
    return self.resolve()
  File "C:\Users\Steven\Anaconda3\lib\site-packages\pkg_resources\__init__.py", line 2415, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "C:\Users\Steven\Anaconda3\lib\site-packages\gym\envs\mujoco\__init__.py", line 1, in <module>
    from gym.envs.mujoco.mujoco_env import MujocoEnv
  File "C:\Users\Steven\Anaconda3\lib\site-packages\gym\envs\mujoco\mujoco_env.py", line 14, in <module>
    raise error.DependencyNotInstalled("{}. (HINT: you need to install mujoco_py, and also perform the setup instructions here: https://github.com/openai/mujoco-py/.)".format(e))
gym.error.DependencyNotInstalled: No module named 'mujoco_py.mjlib'. (HINT: you need to install mujoco_py, and also perform the setup instructions here: https://github.com/openai/mujoco-py/.)
```

j8lp, 2017-12-07:
@abhigenie92  Not sure if you figured it out already, but just in case:
Since @adamlerer changes are still waiting for approval, you'll need to merge the changes in yourself.  
You can either create a fork of @adamlerer repository, merge it with the latest changes from the master branch, and then install gym from your forked repo.
Or you can manually modify the mujoco_env.py and pusher.py files in your local site-packages/gym folder.

sushrutt12, 2017-12-08:
@j8lp Still doesn't work I tried installing the entire repository from @adamlerer's Github link. Alternatively I tried to just replace the mujoco_env.py and pusher.py it still gives the same error `ModuleNotFoundError: No module named 'mujoco_py.mjlib'`

j8lp, 2017-12-08:
Can you post entire stack trace?  @adamlerer code doesn’t use municipal_py.mjlib so there must be an older version of one of the file floating around somewhere

junweima, 2017-12-15:
where do I download the old version of mujoco-py so that the gym will support it?

nemanja-rakicevic, 2018-02-17:
I am using MacOS Sierra 10.12.6 and Python 3.5.2 Anaconda environment

I had an identical issue like @abhigenie92 and previously like #74 I solved it by:

- updating gcc: `brew upgrade gcc`

- installed latest openai gym from source:
``
git clone https://github.com/openai/gym
cd gym 
pip install -e .[all]
``
- installed MuJoCo 1.50 (mjpro150_osx) and licence like here:
https://www.roboti.us/index.html
https://github.com/openai/mujoco-py/issues/74#issuecomment-342046444

- installed mujoco-py from source:
``
git clone https://github.com/openai/mujoco-py.git
cd mujoco-py
python setup.py install
``

You should be able to have this example working:
``
import gym
env = gym.make('Hopper-v2')
print(env.observation_space)
print(env.action_space)
env.reset()
for _ in range(100):
    env.render()
    env.step(env.action_space.sample()) # take a random action
``

geg58, 2018-02-19:
@nemanja-rakicevic this worked great! python setup.py install was the main fix

kirk86, 2018-03-13:
@nemanja-rakicevic thanks for the info. I tried it but for me it failed upon running the example:
```
>>> import gym
>>> env = gym.make('Hopper-v2')
Compiling /home/john/miniconda3/envs/tf/lib/python3.6/site-packages/mujoco_py-1.50.1.42-py3.6.egg/mujoco_py/cymj.pyx because it changed.
[1/1] Cythonizing /home/john/miniconda3/envs/tf/lib/python3.6/site-packages/mujoco_py-1.50.1.42-py3.6.egg/mujoco_py/cymj.pyx
warning: /home/john/miniconda3/envs/tf/lib/python3.6/site-packages/mujoco_py-1.50.1.42-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:99:4: 'mjtDisableBit' redeclared
warning: /home/john/miniconda3/envs/tf/lib/python3.6/site-packages/mujoco_py-1.50.1.42-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:114:4: 'mjtEnableBit' redeclared
warning: /home/john/miniconda3/envs/tf/lib/python3.6/site-packages/mujoco_py-1.50.1.42-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:121:4: 'mjtJoint' redeclared
warning: /home/john/miniconda3/envs/tf/lib/python3.6/site-packages/mujoco_py-1.50.1.42-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:127:4: 'mjtGeom' redeclared
warning: /home/john/miniconda3/envs/tf/lib/python3.6/site-packages/mujoco_py-1.50.1.42-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:149:4: 'mjtCamLight' redeclared
warning: /home/john/miniconda3/envs/tf/lib/python3.6/site-packages/mujoco_py-1.50.1.42-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:157:4: 'mjtTexture' redeclared
warning: /home/john/miniconda3/envs/tf/lib/python3.6/site-packages/mujoco_py-1.50.1.42-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:163:4: 'mjtIntegrator' redeclared
warning: /home/john/miniconda3/envs/tf/lib/python3.6/site-packages/mujoco_py-1.50.1.42-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:168:4: 'mjtCollision' redeclared
warning: /home/john/miniconda3/envs/tf/lib/python3.6/site-packages/mujoco_py-1.50.1.42-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:174:4: 'mjtCone' redeclared
warning: /home/john/miniconda3/envs/tf/lib/python3.6/site-packages/mujoco_py-1.50.1.42-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:179:4: 'mjtJacobian' redeclared
warning: /home/john/miniconda3/envs/tf/lib/python3.6/site-packages/mujoco_py-1.50.1.42-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:185:4: 'mjtSolver' redeclared
warning: /home/john/miniconda3/envs/tf/lib/python3.6/site-packages/mujoco_py-1.50.1.42-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:191:4: 'mjtImp' redeclared
warning: /home/john/miniconda3/envs/tf/lib/python3.6/site-packages/mujoco_py-1.50.1.42-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:198:4: 'mjtRef' redeclared
warning: /home/john/miniconda3/envs/tf/lib/python3.6/site-packages/mujoco_py-1.50.1.42-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:203:4: 'mjtEq' redeclared
warning: /home/john/miniconda3/envs/tf/lib/python3.6/site-packages/mujoco_py-1.50.1.42-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:211:4: 'mjtWrap' redeclared
warning: /home/john/miniconda3/envs/tf/lib/python3.6/site-packages/mujoco_py-1.50.1.42-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:220:4: 'mjtTrn' redeclared
warning: /home/john/miniconda3/envs/tf/lib/python3.6/site-packages/mujoco_py-1.50.1.42-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:230:4: 'mjtDyn' redeclared
warning: /home/john/miniconda3/envs/tf/lib/python3.6/site-packages/mujoco_py-1.50.1.42-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:237:4: 'mjtGain' redeclared
warning: /home/john/miniconda3/envs/tf/lib/python3.6/site-packages/mujoco_py-1.50.1.42-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:242:4: 'mjtBias' redeclared
warning: /home/john/miniconda3/envs/tf/lib/python3.6/site-packages/mujoco_py-1.50.1.42-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:248:4: 'mjtObj' redeclared
warning: /home/john/miniconda3/envs/tf/lib/python3.6/site-packages/mujoco_py-1.50.1.42-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:274:4: 'mjtConstraint' redeclared
warning: /home/john/miniconda3/envs/tf/lib/python3.6/site-packages/mujoco_py-1.50.1.42-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:285:4: 'mjtConstraintState' redeclared
warning: /home/john/miniconda3/envs/tf/lib/python3.6/site-packages/mujoco_py-1.50.1.42-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:294:4: 'mjtSensor' redeclared
warning: /home/john/miniconda3/envs/tf/lib/python3.6/site-packages/mujoco_py-1.50.1.42-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:338:4: 'mjtStage' redeclared
warning: /home/john/miniconda3/envs/tf/lib/python3.6/site-packages/mujoco_py-1.50.1.42-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:345:4: 'mjtDataType' redeclared
warning: /home/john/miniconda3/envs/tf/lib/python3.6/site-packages/mujoco_py-1.50.1.42-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:99:4: 'mjtDisableBit' redeclared
warning: /home/john/miniconda3/envs/tf/lib/python3.6/site-packages/mujoco_py-1.50.1.42-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:114:4: 'mjtEnableBit' redeclared
warning: /home/john/miniconda3/envs/tf/lib/python3.6/site-packages/mujoco_py-1.50.1.42-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:121:4: 'mjtJoint' redeclared
warning: /home/john/miniconda3/envs/tf/lib/python3.6/site-packages/mujoco_py-1.50.1.42-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:127:4: 'mjtGeom' redeclared
warning: /home/john/miniconda3/envs/tf/lib/python3.6/site-packages/mujoco_py-1.50.1.42-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:149:4: 'mjtCamLight' redeclared
[... cut ...]
"/home/user/miniconda3/envs/tf/lib/python3.6/distutils/command/build_ext.py",
line 533, in build_extension
    depends=ext.depends)
  File
"/home/user/miniconda3/envs/tf/lib/python3.6/site-packages/mujoco_py-1.50.1.42-py3.6.egg/mujoco_py/builder.py",
line 127, in caching_compile
    self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
  File "/home/user/miniconda3/envs/tf/lib/python3.6/distutils/unixccompiler.py",
line 120, in _compile
    raise CompileError(msg)
distutils.errors.CompileError: command 'gcc' failed with exit status 1
```

nemanja-rakicevic, 2018-03-14:
@kirk86 so one issue might be that you're using python 3.6, I think gym has problems with 3.6. Another issue could be related to the gcc update.
Not sure if I could be more helpful than this..

kirk86, 2018-03-14:
@nemanja-rakicevic thanks a lot. Appreciate it! I thought that by reading the docs they mentioned that they released PR in which they support python 3.6? 

> Another issue could be related to the gcc update.

I don't understand why that could be an issue. It's version 5.4.x ?

Anyways thanks again. ;)

AnandS09, 2018-04-19:
@kirk86 : Looks like libglew-dev is not installed. 

Use all the dependencies mentioned in this Dockerfile: https://github.com/openai/mujoco-py/blob/master/Dockerfile

YeeCY, 2020-04-29:
Has the mjlib module been deleted in mujoco-py? What if I want to use some built-in functions such as mj_fwdPosition?

XYZ-99, 2021-09-13:
> Has the mjlib module been deleted in mujoco-py? What if I want to use some built-in functions such as mj_fwdPosition?

Check [this example](https://github.com/openai/mujoco-py/blob/master/examples/internal_functions.py#L5).

Basically you can use `from mujoco_py import functions` to use these internal functions.