openai/mujoco-py #82: Installation Error

jaskarangrover92, 2017-07-25:
Hello I am trying to install mujoco-py on MacOSX 10.11 with Python 3.5.2 following the instructions on https://github.com/openai/mujoco-py.  However, I get the following error log when it tries to run setup.py install.    It would be really helpful if you could help me debug this issue. 

Installing collected packages: mujoco-py
  Running setup.py install for mujoco-py ... error
    Complete output from command /Library/Frameworks/Python.framework/Versions/3.5/bin/python3.5 -u -c "import setuptools, tokenize;__file__='/private/var/folders/b0/lv4f1rzd4vb9cz1r8582__1r0000gn/T/pip-build-qdi2s4hi/mujoco-py/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /var/folders/b0/lv4f1rzd4vb9cz1r8582__1r0000gn/T/pip-zegn3228-record/install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    Compiling /private/var/folders/b0/lv4f1rzd4vb9cz1r8582__1r0000gn/T/pip-build-qdi2s4hi/mujoco-py/mujoco_py/cymj.pyx because it changed.
    [1/1] Cythonizing /private/var/folders/b0/lv4f1rzd4vb9cz1r8582__1r0000gn/T/pip-build-qdi2s4hi/mujoco-py/mujoco_py/cymj.pyx
    running build_ext
    building 'mujoco_py.cymj' extension
    creating /private/var/folders/b0/lv4f1rzd4vb9cz1r8582__1r0000gn/T/pip-build-qdi2s4hi/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder
    creating /private/var/folders/b0/lv4f1rzd4vb9cz1r8582__1r0000gn/T/pip-build-qdi2s4hi/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-intel-3.5
    creating /private/var/folders/b0/lv4f1rzd4vb9cz1r8582__1r0000gn/T/pip-build-qdi2s4hi/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-intel-3.5/private
    creating /private/var/folders/b0/lv4f1rzd4vb9cz1r8582__1r0000gn/T/pip-build-qdi2s4hi/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-intel-3.5/private/var
    creating /private/var/folders/b0/lv4f1rzd4vb9cz1r8582__1r0000gn/T/pip-build-qdi2s4hi/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-intel-3.5/private/var/folders
    creating /private/var/folders/b0/lv4f1rzd4vb9cz1r8582__1r0000gn/T/pip-build-qdi2s4hi/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-intel-3.5/private/var/folders/b0
    creating /private/var/folders/b0/lv4f1rzd4vb9cz1r8582__1r0000gn/T/pip-build-qdi2s4hi/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-intel-3.5/private/var/folders/b0/lv4f1rzd4vb9cz1r8582__1r0000gn
    creating /private/var/folders/b0/lv4f1rzd4vb9cz1r8582__1r0000gn/T/pip-build-qdi2s4hi/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-intel-3.5/private/var/folders/b0/lv4f1rzd4vb9cz1r8582__1r0000gn/T
    creating /private/var/folders/b0/lv4f1rzd4vb9cz1r8582__1r0000gn/T/pip-build-qdi2s4hi/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-intel-3.5/private/var/folders/b0/lv4f1rzd4vb9cz1r8582__1r0000gn/T/pip-build-qdi2s4hi
    creating /private/var/folders/b0/lv4f1rzd4vb9cz1r8582__1r0000gn/T/pip-build-qdi2s4hi/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-intel-3.5/private/var/folders/b0/lv4f1rzd4vb9cz1r8582__1r0000gn/T/pip-build-qdi2s4hi/mujoco-py
    creating /private/var/folders/b0/lv4f1rzd4vb9cz1r8582__1r0000gn/T/pip-build-qdi2s4hi/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-intel-3.5/private/var/folders/b0/lv4f1rzd4vb9cz1r8582__1r0000gn/T/pip-build-qdi2s4hi/mujoco-py/mujoco_py
    creating /private/var/folders/b0/lv4f1rzd4vb9cz1r8582__1r0000gn/T/pip-build-qdi2s4hi/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-intel-3.5/private/var/folders/b0/lv4f1rzd4vb9cz1r8582__1r0000gn/T/pip-build-qdi2s4hi/mujoco-py/mujoco_py/gl
    /usr/local/bin/gcc-7 -fno-strict-aliasing -Wsign-compare -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -arch i386 -arch x86_64 -g -DONMAC -I/private/var/folders/b0/lv4f1rzd4vb9cz1r8582__1r0000gn/T/pip-build-qdi2s4hi/mujoco-py/mujoco_py -I/Users/jaskaran/.mujoco/mjpro150/include -I/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/numpy/core/include -I/Library/Frameworks/Python.framework/Versions/3.5/include/python3.5m -c /private/var/folders/b0/lv4f1rzd4vb9cz1r8582__1r0000gn/T/pip-build-qdi2s4hi/mujoco-py/mujoco_py/cymj.c -o /private/var/folders/b0/lv4f1rzd4vb9cz1r8582__1r0000gn/T/pip-build-qdi2s4hi/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.6-intel-3.5/private/var/folders/b0/lv4f1rzd4vb9cz1r8582__1r0000gn/T/pip-build-qdi2s4hi/mujoco-py/mujoco_py/cymj.o -fopenmp -w
    gcc-7: warning: x86_64 conflicts with i386 (arch flags ignored)
    /private/var/folders/b0/lv4f1rzd4vb9cz1r8582__1r0000gn/T/pip-build-qdi2s4hi/mujoco-py/mujoco_py/cymj.c:521:10: fatal error: mjmodel.h: No such file or directory
     #include "mjmodel.h"
              ^~~~~~~~~~~
    compilation terminated.
    error: command '/usr/local/bin/gcc-7' failed with exit status 1
    

----------------------------------------
Command "/Library/Frameworks/Python.framework/Versions/3.5/bin/python3.5 -u -c "import setuptools, tokenize;__file__='/private/var/folders/b0/lv4f1rzd4vb9cz1r8582__1r0000gn/T/pip-build-qdi2s4hi/mujoco-py/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /var/folders/b0/lv4f1rzd4vb9cz1r8582__1r0000gn/T/pip-zegn3228-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /private/var/folders/b0/lv4f1rzd4vb9cz1r8582__1r0000gn/T/pip-build-qdi2s4hi/mujoco-py/

machinaut, 2017-07-26:
It's looking for the header file in the directory `/Users/jaskaran/.mujoco/mjpro150/include`, can you confirm there's a `mjmodel.h` in that directory?

jaskarangrover92, 2017-07-26:
Hi Alex

Thanks for the reply. Yes the header file is indeed there.

machinaut, 2017-07-26:
what architecture are you building on?  `gcc-7: warning: x86_64 conflicts with i386 (arch flags ignored)` looks like there may be a mismatch

machinaut, 2017-07-26:
Also what version of gcc are you using?  `gcc --version`

jaskarangrover92, 2017-07-27:
Hey Alex, I have x86_64 and gcc version is 7.3.0 .

njbittner, 2017-09-17:
@jaskarangrover92 did you ever find a solution for this?

tshrjn, 2017-11-26:
I'm having a similar problem on macOS 10.11.6 with anaconda python 3.5.3:

````

Collecting mujoco-py
  Using cached mujoco-py-1.50.1.33.tar.gz
Requirement already satisfied: glfw>=1.4.0 in /Users/tshrjn/anaconda3/lib/python3.5/site-packages (from mujoco-py)
Requirement already satisfied: numpy>=1.11 in /Users/tshrjn/anaconda3/lib/python3.5/site-packages (from mujoco-py)
Requirement already satisfied: Cython>=0.27.2 in /Users/tshrjn/anaconda3/lib/python3.5/site-packages (from mujoco-py)
Requirement already satisfied: imageio>=2.1.2 in /Users/tshrjn/anaconda3/lib/python3.5/site-packages (from mujoco-py)
Requirement already satisfied: cffi>=1.10 in /Users/tshrjn/anaconda3/lib/python3.5/site-packages (from mujoco-py)
Requirement already satisfied: pycparser in /Users/tshrjn/anaconda3/lib/python3.5/site-packages (from cffi>=1.10->mujoco-py)
Building wheels for collected packages: mujoco-py
  Running setup.py bdist_wheel for mujoco-py ... error
  Complete output from command /Users/tshrjn/anaconda3/bin/python -u -c "import setuptools, tokenize;__file__='/private/var/folders/tn/8ztgmljj4334_jsp3qh1jqhw0000gn/T/pip-build-jv9jkell/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /var/folders/tn/8ztgmljj4334_jsp3qh1jqhw0000gn/T/tmpzgsj07g5pip-wheel- --python-tag cp35:
  running bdist_wheel
  running build
  Compiling /private/var/folders/tn/8ztgmljj4334_jsp3qh1jqhw0000gn/T/pip-build-jv9jkell/mujoco-py/mujoco_py/cymj.pyx because it depends on /Users/tshrjn/anaconda3/lib/python3.5/site-packages/Cython/Includes/numpy/__init__.pxd.
  [1/1] Cythonizing /private/var/folders/tn/8ztgmljj4334_jsp3qh1jqhw0000gn/T/pip-build-jv9jkell/mujoco-py/mujoco_py/cymj.pyx
  running build_ext
  building 'mujoco_py.cymj' extension
  creating /private/var/folders/tn/8ztgmljj4334_jsp3qh1jqhw0000gn/T/pip-build-jv9jkell/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder
  creating /private/var/folders/tn/8ztgmljj4334_jsp3qh1jqhw0000gn/T/pip-build-jv9jkell/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.7-x86_64-3.5
  creating /private/var/folders/tn/8ztgmljj4334_jsp3qh1jqhw0000gn/T/pip-build-jv9jkell/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.7-x86_64-3.5/private
  creating /private/var/folders/tn/8ztgmljj4334_jsp3qh1jqhw0000gn/T/pip-build-jv9jkell/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.7-x86_64-3.5/private/var
  creating /private/var/folders/tn/8ztgmljj4334_jsp3qh1jqhw0000gn/T/pip-build-jv9jkell/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.7-x86_64-3.5/private/var/folders
  creating /private/var/folders/tn/8ztgmljj4334_jsp3qh1jqhw0000gn/T/pip-build-jv9jkell/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.7-x86_64-3.5/private/var/folders/tn
  creating /private/var/folders/tn/8ztgmljj4334_jsp3qh1jqhw0000gn/T/pip-build-jv9jkell/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.7-x86_64-3.5/private/var/folders/tn/8ztgmljj4334_jsp3qh1jqhw0000gn
  creating /private/var/folders/tn/8ztgmljj4334_jsp3qh1jqhw0000gn/T/pip-build-jv9jkell/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.7-x86_64-3.5/private/var/folders/tn/8ztgmljj4334_jsp3qh1jqhw0000gn/T
  creating /private/var/folders/tn/8ztgmljj4334_jsp3qh1jqhw0000gn/T/pip-build-jv9jkell/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.7-x86_64-3.5/private/var/folders/tn/8ztgmljj4334_jsp3qh1jqhw0000gn/T/pip-build-jv9jkell
  creating /private/var/folders/tn/8ztgmljj4334_jsp3qh1jqhw0000gn/T/pip-build-jv9jkell/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.7-x86_64-3.5/private/var/folders/tn/8ztgmljj4334_jsp3qh1jqhw0000gn/T/pip-build-jv9jkell/mujoco-py
  creating /private/var/folders/tn/8ztgmljj4334_jsp3qh1jqhw0000gn/T/pip-build-jv9jkell/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.7-x86_64-3.5/private/var/folders/tn/8ztgmljj4334_jsp3qh1jqhw0000gn/T/pip-build-jv9jkell/mujoco-py/mujoco_py
  creating /private/var/folders/tn/8ztgmljj4334_jsp3qh1jqhw0000gn/T/pip-build-jv9jkell/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.7-x86_64-3.5/private/var/folders/tn/8ztgmljj4334_jsp3qh1jqhw0000gn/T/pip-build-jv9jkell/mujoco-py/mujoco_py/gl
  /usr/local/bin/gcc-7 -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -I/Users/tshrjn/anaconda3/include -arch x86_64 -DONMAC -Imujoco_py -I/private/var/folders/tn/8ztgmljj4334_jsp3qh1jqhw0000gn/T/pip-build-jv9jkell/mujoco-py/mujoco_py -I/Users/tshrjn/.mujoco/mjpro150/include -I/Users/tshrjn/anaconda3/lib/python3.5/site-packages/numpy/core/include -I/Users/tshrjn/anaconda3/include/python3.5m -c /private/var/folders/tn/8ztgmljj4334_jsp3qh1jqhw0000gn/T/pip-build-jv9jkell/mujoco-py/mujoco_py/cymj.c -o /private/var/folders/tn/8ztgmljj4334_jsp3qh1jqhw0000gn/T/pip-build-jv9jkell/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.7-x86_64-3.5/private/var/folders/tn/8ztgmljj4334_jsp3qh1jqhw0000gn/T/pip-build-jv9jkell/mujoco-py/mujoco_py/cymj.o -fopenmp -w
  /private/var/folders/tn/8ztgmljj4334_jsp3qh1jqhw0000gn/T/pip-build-jv9jkell/mujoco-py/mujoco_py/cymj.c:573:10: fatal error: mjmodel.h: No such file or directory
   #include "mjmodel.h"
            ^~~~~~~~~~~
  compilation terminated.
[... cut ...]
Command "/Users/tshrjn/anaconda3/bin/python -u -c "import setuptools, tokenize;__file__='/private/var/folders/tn/8ztgmljj4334_jsp3qh1jqhw0000gn/T/pip-build-jv9jkell/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /var/folders/tn/8ztgmljj4334_jsp3qh1jqhw0000gn/T/pip-n619ywyy-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /private/var/folders/tn/8ztgmljj4334_jsp3qh1jqhw0000gn/T/pip-build-jv9jkell/mujoco-py/

hanspond, 2018-02-16:
This is a bug due to the incompatibility of python2 pip(default in your anaconda) and the current version of mujoco-py.
Plz refer to the solution listed in here.[https://github.com/openai/mujoco-py/issues/47](url)