openai/mujoco-py #88: $import mujoco_py, then failed: command 'gcc' failed with exit status 1

xiachongkun, 2017-08-08:
I have installed Ubuntu16.04+Anaconda4.4+tensorflow(python3.5.3)+open AI gym[all], then I want to install mujoco. So , I followed by  https://github.com/openai/mujoco-py. But ,when I want :$import mujoco_py,then it have a mistake: distutils.errors.CompileError: command 'gcc' failed with exit status 1.

The code is described below:
(tensorflow) xia@ubuntu:~/mujoco-py-master$ python
Python 3.5.3 |Continuum Analytics, Inc.| (default, Mar  6 2017, 11:58:13) 
[GCC 4.4.7 20120313 (Red Hat 4.4.7-1)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import mujoco_py
Compiling /home/xia/mujoco-py-master/mujoco_py/cymj.pyx because it changed.
[1/1] Cythonizing /home/xia/mujoco-py-master/mujoco_py/cymj.pyx
running build_ext
building 'mujoco_py.cymj' extension
creating /home/xia/mujoco-py-master/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder
creating /home/xia/mujoco-py-master/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5
creating /home/xia/mujoco-py-master/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/home
creating /home/xia/mujoco-py-master/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/home/xia
creating /home/xia/mujoco-py-master/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/home/xia/mujoco-py-master
creating /home/xia/mujoco-py-master/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/home/xia/mujoco-py-master/mujoco_py
creating /home/xia/mujoco-py-master/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/home/xia/mujoco-py-master/mujoco_py/gl
gcc -pthread -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -I/home/xia/mujoco-py-master/mujoco_py -I/home/xia/.mujoco/mjpro150/include -I/home/xia/.local/lib/python3.5/site-packages/numpy/core/include -I/home/xia/anaconda3/envs/tensorflow/include/python3.5m -c /home/xia/mujoco-py-master/mujoco_py/cymj.c -o /home/xia/mujoco-py-master/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/home/xia/mujoco-py-master/mujoco_py/cymj.o -fopenmp -w
gcc -pthread -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -I/home/xia/mujoco-py-master/mujoco_py -I/home/xia/.mujoco/mjpro150/include -I/home/xia/.local/lib/python3.5/site-packages/numpy/core/include -I/home/xia/anaconda3/envs/tensorflow/include/python3.5m -c /home/xia/mujoco-py-master/mujoco_py/gl/osmesashim.c -o /home/xia/mujoco-py-master/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/home/xia/mujoco-py-master/mujoco_py/gl/osmesashim.o -fopenmp -w
/home/xia/mujoco-py-master/mujoco_py/gl/osmesashim.c:1:23: fatal error: GL/osmesa.h: 没有那个文件或目录
compilation terminated.
Traceback (most recent call last):
  File "/home/xia/anaconda3/envs/tensorflow/lib/python3.5/distutils/unixccompiler.py", line 118, in _compile
    extra_postargs)
  File "/home/xia/anaconda3/envs/tensorflow/lib/python3.5/distutils/ccompiler.py", line 909, in spawn
    spawn(cmd, dry_run=self.dry_run)
  File "/home/xia/anaconda3/envs/tensorflow/lib/python3.5/distutils/spawn.py", line 36, in spawn
    _spawn_posix(cmd, search_path, dry_run=dry_run)
  File "/home/xia/anaconda3/envs/tensorflow/lib/python3.5/distutils/spawn.py", line 159, in _spawn_posix
    % (cmd, exit_status))
distutils.errors.DistutilsExecError: command 'gcc' failed with exit status 1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/xia/mujoco-py-master/mujoco_py/__init__.py", line 1, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/home/xia/mujoco-py-master/mujoco_py/builder.py", line 296, in <module>
    cymj = load_cython_ext(mjpro_path)
  File "/home/xia/mujoco-py-master/mujoco_py/builder.py", line 56, in load_cython_ext
    cext_so_path = builder.build()
  File "/home/xia/mujoco-py-master/mujoco_py/builder.py", line 119, in build
    built_so_file_path = self._build_impl()
  File "/home/xia/mujoco-py-master/mujoco_py/builder.py", line 139, in _build_impl
    dist.run_commands()
  File "/home/xia/anaconda3/envs/tensorflow/lib/python3.5/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/home/xia/anaconda3/envs/tensorflow/lib/python3.5/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/home/xia/.local/lib/python3.5/site-packages/Cython/Distutils/old_build_ext.py", line 185, in run
    _build_ext.build_ext.run(self)
  File "/home/xia/anaconda3/envs/tensorflow/lib/python3.5/distutils/command/build_ext.py", line 339, in run
    self.build_extensions()
  File "/home/xia/mujoco-py-master/mujoco_py/builder.py", line 77, in build_extensions
    build_ext.build_extensions(self)
  File "/home/xia/.local/lib/python3.5/site-packages/Cython/Distutils/old_build_ext.py", line 193, in build_extensions
    self.build_extension(ext)
  File "/home/xia/anaconda3/envs/tensorflow/lib/python3.5/distutils/command/build_ext.py", line 533, in build_extension
    depends=ext.depends)
  File "/home/xia/anaconda3/envs/tensorflow/lib/python3.5/distutils/ccompiler.py", line 574, in compile
    self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
  File "/home/xia/anaconda3/envs/tensorflow/lib/python3.5/distutils/unixccompiler.py", line 120, in _compile
    raise CompileError(msg)
distutils.errors.CompileError: command 'gcc' failed with exit status 1
>>> 

(tensorflow) xia@ubuntu:~/mujoco-py-master$ 
(tensorflow) xia@ubuntu:~/mujoco-py-master$ gcc --version
gcc (Ubuntu 5.4.0-6ubuntu1~16.04.4) 5.4.0 20160609
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

For the mistake:  distutils.errors.CompileError: command 'gcc' failed with exit status 1
I have looked for many solutions  but they did't work.  I feel I wanna go die~~ I don't know how to solve it~~I need to help~~

ahartikainen, 2017-08-08:
Hi, try to install in fresh conda virtual env

    conda create -n py36 python=3.6 numpy cython
    source activate
    pip ....

There can be problems with Ubuntu default gcc and Anaconda (libstdc++ file in Anaconda/bin)

hangxu124, 2017-11-15:
@xiachongkun 
have u solved this problem?