openai/mujoco-py #90: fatal error: GL/osmesa.h: 没有那个文件或目录

HuachunWang, 2017-08-16:
pip3 install -U 'mujoco-py<1.50.2,>=1.50.1'
Collecting mujoco-py<1.50.2,>=1.50.1
  Using cached mujoco-py-1.50.1.16.tar.gz
Collecting glfw>=1.4.0 (from mujoco-py<1.50.2,>=1.50.1)
Collecting numpy>=1.11 (from mujoco-py<1.50.2,>=1.50.1)
  Using cached numpy-1.13.1-cp35-cp35m-manylinux1_x86_64.whl
Collecting Cython>=0.25.2 (from mujoco-py<1.50.2,>=1.50.1)
  Using cached Cython-0.26-cp35-cp35m-manylinux1_x86_64.whl
Collecting imageio>=2.1.2 (from mujoco-py<1.50.2,>=1.50.1)
Collecting pillow (from imageio>=2.1.2->mujoco-py<1.50.2,>=1.50.1)
  Using cached Pillow-4.2.1-cp35-cp35m-manylinux1_x86_64.whl
Collecting olefile (from pillow->imageio>=2.1.2->mujoco-py<1.50.2,>=1.50.1)
Building wheels for collected packages: mujoco-py
  Running setup.py bdist_wheel for mujoco-py ... error
  Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-ftgzdgl1/mujoco-py/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" bdist_wheel -d /tmp/tmp6pnyslm1pip-wheel- --python-tag cp35:
  running bdist_wheel
  running build
  Compiling /tmp/pip-build-ftgzdgl1/mujoco-py/mujoco_py/cymj.pyx because it depends on /home/maizi/.local/lib/python3.5/site-packages/Cython/Includes/numpy/__init__.pxd.
  [1/1] Cythonizing /tmp/pip-build-ftgzdgl1/mujoco-py/mujoco_py/cymj.pyx
  running build_ext
  building 'mujoco_py.cymj' extension
  creating /tmp/pip-build-ftgzdgl1/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder
  creating /tmp/pip-build-ftgzdgl1/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5
  creating /tmp/pip-build-ftgzdgl1/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp
  creating /tmp/pip-build-ftgzdgl1/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build-ftgzdgl1
  creating /tmp/pip-build-ftgzdgl1/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build-ftgzdgl1/mujoco-py
  creating /tmp/pip-build-ftgzdgl1/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build-ftgzdgl1/mujoco-py/mujoco_py
  creating /tmp/pip-build-ftgzdgl1/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build-ftgzdgl1/mujoco-py/mujoco_py/gl
  x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/tmp/pip-build-ftgzdgl1/mujoco-py/mujoco_py -I/home/maizi/.mujoco/mjpro150/include -I/home/maizi/.local/lib/python3.5/site-packages/numpy/core/include -I/usr/include/python3.5m -c /tmp/pip-build-ftgzdgl1/mujoco-py/mujoco_py/cymj.c -o /tmp/pip-build-ftgzdgl1/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build-ftgzdgl1/mujoco-py/mujoco_py/cymj.o -fopenmp -w
  x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/tmp/pip-build-ftgzdgl1/mujoco-py/mujoco_py -I/home/maizi/.mujoco/mjpro150/include -I/home/maizi/.local/lib/python3.5/site-packages/numpy/core/include -I/usr/include/python3.5m -c /tmp/pip-build-ftgzdgl1/mujoco-py/mujoco_py/gl/osmesashim.c -o /tmp/pip-build-ftgzdgl1/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build-ftgzdgl1/mujoco-py/mujoco_py/gl/osmesashim.o -fopenmp -w
  /tmp/pip-build-ftgzdgl1/mujoco-py/mujoco_py/gl/osmesashim.c:1:23: fatal error: GL/osmesa.h: 没有那个文件或目录
  compilation terminated.
  error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
  
  ----------------------------------------
  Failed building wheel for mujoco-py
  Running setup.py clean for mujoco-py
Failed to build mujoco-py
Installing collected packages: glfw, numpy, Cython, olefile, pillow, imageio, mujoco-py
  Running setup.py install for mujoco-py ... error
    Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-ftgzdgl1/mujoco-py/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-0yrgnfe4-record/install-record.txt --single-version-externally-managed --compile --user --prefix=:
    running install
    running build
    running build_ext
    building 'mujoco_py.cymj' extension
    x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/tmp/pip-build-ftgzdgl1/mujoco-py/mujoco_py -I/home/maizi/.mujoco/mjpro150/include -I/home/maizi/.local/lib/python3.5/site-packages/numpy/core/include -I/usr/include/python3.5m -c /tmp/pip-build-ftgzdgl1/mujoco-py/mujoco_py/cymj.c -o /tmp/pip-build-ftgzdgl1/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build-ftgzdgl1/mujoco-py/mujoco_py/cymj.o -fopenmp -w
    x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/tmp/pip-build-ftgzdgl1/mujoco-py/mujoco_py -I/home/maizi/.mujoco/mjpro150/include -I/home/maizi/.local/lib/python3.5/site-packages/numpy/core/include -I/usr/include/python3.5m -c /tmp/pip-build-ftgzdgl1/mujoco-py/mujoco_py/gl/osmesashim.c -o /tmp/pip-build-ftgzdgl1/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build-ftgzdgl1/mujoco-py/mujoco_py/gl/osmesashim.o -fopenmp -w
    /tmp/pip-build-ftgzdgl1/mujoco-py/mujoco_py/gl/osmesashim.c:1:23: fatal error: GL/osmesa.h: 没有那个文件或目录
    compilation terminated.
    error: command 'x86_64-linux-gnu-gcc' failed with exit status 1

HuachunWang, 2017-08-16:
apt-file update
apt-file search "GL/osmesa.h
Then you will find libosmesa6-dev: /usr/include/GL/osmesa.h
sudo apt install libosmesa5-dev
done

closed 2017-08-16 by HuachunWang

YuehChuan, 2018-03-05:
It's sudo apt install libosmesa6-dev

inspirepassion, 2018-09-02:
@HuachunWang  Hi, I tried your solution. but it returns error like this:
libosmesa6-dev : Depends: libosmesa6 (= 18.0.0~rc5-1ubuntu1) but it is not going to be installed

my system is Ubuntu 18.04

HuachunWang, 2018-09-02:
My OS is Ubuntu 16.04 and I try it again, it works! It seems that your problem is just about dependency, you could try "sudo apt install libosmesa6" first, then "sudo apt install libosmesa6-dev".

jangirrishabh, 2019-05-06:
Hello @HuachunWang my 18.04 has the file
' apt-file search "GL/osmesa.h"
libosmesa6-dev: /usr/include/GL/osmesa.h '
But mujoco_py still cannot find the headers, any clue on how to direct the installation to the headers directory? Thanks

HuachunWang, 2019-05-06:
> Hello @HuachunWang my 18.04 has the file
> ' apt-file search "GL/osmesa.h"
> libosmesa6-dev: /usr/include/GL/osmesa.h '
> But mujoco_py still cannot find the headers, any clue on how to direct the installation to the headers directory? Thanks

After apt-file search "GL/osmesa.h", you need run this command: sudo apt install libosmesa6-dev.
If it didn't work, tell me the details of error.

jangirrishabh, 2019-05-06:
I ran the install command and have the GL library installed in my system. But when importing mujoco_py I get the "GL/osmesa.h" package not found error. 
 I am running an anaconda environment so that might be a problem? cause my system cannot find the header file.
`echo $LD_PRELOAD :/usr/lib/x86_64-linux-gnu/libGL.so:/usr/lib/x86_64-linux-gnu/libGLEW.so`
`echo $LD_LIBRARY_PATH :/usr/local/cuda-9.2/lib64:/home/rjangir/.mujoco/mujoco200/bin`

How do I link the location of my header files

ERROR:

`>>> import mujoco_py
running build_ext
building 'mujoco_py.cymj' extension
/home/rjangir/softwareInstalled/anaconda3/envs/dnn/bin/x86_64-conda_cos6-linux-gnu-cc -DNDEBUG -fwrapv -O2 -Wall -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -pipe -DNDEBUG -D_FORTIFY_SOURCE=2 -O2 -fPIC -Imujoco_py -I/home/rjangir/workSpace/Mujoco/mujoco-py/mujoco_py -I/home/rjangir/.mujoco/mujoco200/include -I/home/rjangir/softwareInstalled/anaconda3/envs/dnn/lib/python3.6/site-packages/numpy/core/include -I/home/rjangir/softwareInstalled/anaconda3/envs/dnn/include/python3.6m -c /home/rjangir/workSpace/Mujoco/mujoco-py/mujoco_py/cymj.c -o /home/rjangir/workSpace/Mujoco/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_36_linuxcpuextensionbuilder/temp.linux-x86_64-3.6/home/rjangir/workSpace/Mujoco/mujoco-py/mujoco_py/cymj.o -fopenmp -w
/home/rjangir/softwareInstalled/anaconda3/envs/dnn/bin/x86_64-conda_cos6-linux-gnu-cc -DNDEBUG -fwrapv -O2 -Wall -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -pipe -DNDEBUG -D_FORTIFY_SOURCE=2 -O2 -fPIC -Imujoco_py -I/home/rjangir/workSpace/Mujoco/mujoco-py/mujoco_py -I/home/rjangir/.mujoco/mujoco200/include -I/home/rjangir/softwareInstalled/anaconda3/envs/dnn/lib/python3.6/site-packages/numpy/core/include -I/home/rjangir/softwareInstalled/anaconda3/envs/dnn/include/python3.6m -c /home/rjangir/workSpace/Mujoco/mujoco-py/mujoco_py/gl/osmesashim.c -o /home/rjangir/workSpace/Mujoco/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_36_linuxcpuextensionbuilder/temp.linux-x86_64-3.6/home/rjangir/workSpace/Mujoco/mujoco-py/mujoco_py/gl/osmesashim.o -fopenmp -w
/home/rjangir/workSpace/Mujoco/mujoco-py/mujoco_py/gl/osmesashim.c:1:10: fatal error: GL/osmesa.h: No such file or directory
 #include <GL/osmesa.h>
          ^~~~~~~~~~~~~
compilation terminated.
Traceback (most recent call last):
  File "/home/rjangir/softwareInstalled/anaconda3/envs/dnn/lib/python3.6/distutils/unixccompiler.py", line 118, in _compile
    extra_postargs)
  File "/home/rjangir/softwareInstalled/anaconda3/envs/dnn/lib/python3.6/distutils/ccompiler.py", line 909, in spawn
    spawn(cmd, dry_run=self.dry_run)
  File "/home/rjangir/softwareInstalled/anaconda3/envs/dnn/lib/python3.6/distutils/spawn.py", line 36, in spawn
    _spawn_posix(cmd, search_path, dry_run=dry_run)
  File "/home/rjangir/softwareInstalled/anaconda3/envs/dnn/lib/python3.6/distutils/spawn.py", line 159, in _spawn_posix
    % (cmd, exit_status))
distutils.errors.DistutilsExecError: command '/home/rjangir/softwareInstalled/anaconda3/envs/dnn/bin/x86_64-conda_cos6-linux-gnu-cc' failed with exit status 1`

HuachunWang, 2019-05-13:
> I ran the install command and have the GL library installed in my system. But when importing mujoco_py I get the "GL/osmesa.h" package not found error.
> I am running an anaconda environment so that [...]

If possible, you can try to change anaconda3 environment to pure python3.5. I haven't used anaconda3.

jangirrishabh, 2019-05-14:
So my system was using Anaconda's gcc which was causing problems. In that case even though osmesa drivers were installed the compiler could not find them (Anaconda compiler). The solution is to source the native python environment and use system gcc. Thanks @HuachunWang !

GoingMyWay, 2019-09-18:
If you install openai's baselines, you should `python setup.py install` rather than `pip install -e .`

huiwenzhang, 2019-11-19:
@jangirrishabh  Hi, I encountered the same problem. What do you mean by sourcing the native python? source the system python interpreter instead of the the conda version? But if I just want to install the mujoco_py inside a conda virtual env, what should I do? Is there any way to modify the gcc's behavior for a specific conda env?

HankerSia, 2020-11-27:
> @HuachunWang Hi, I tried your solution. but it returns error like this:
> libosmesa6-dev : Depends: libosmesa6 (= 18.0.0~rc5-1ubuntu1) but it is not going to be installed
> 
> my system is Ubuntu 18.04

how you solve this problem?