openai/mujoco-py #96: /tmp/pip-build-92_avjgn/mujoco-py/mujoco_py/gl/osmesashim.c:1:23: fatal error: GL/osmesa.h: No such file or directory

harrypann, 2017-08-24:
I follow the instructions to install MuJoCo, and got the following error:
        /tmp/pip-build-92_avjgn/mujoco-py/mujoco_py/gl/osmesashim.c:1:23: fatal error: GL/osmesa.h: No such file or directory
So I tried to install with "sudo apt-get install libosmesa6-dev", and it prompts that I should install a list of dependencies, so I tried to install via "sudo apt-get install libdrm2" and it propmts an error message says
        E: Sub-process /usr/bin/dpkg returned an error code (1)
The computer is a Ubuntu 14.04, with GPU installed.

andrewliao11, 2017-10-03:
I encounter this issue, too. 
Also Ubuntu 14.04, with GPU installed.
Using python3.5.2

ViktorM, 2017-10-09:
Just have the same error on Ubuntu 16.04, Python 3.5.2, laptop with GPU - GTX 1070:
```gcc -pthread -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -Imujoco_py -I/tmp/pip-build-20t9hi5v/mujoco-py/mujoco_py -I/home/viktor/.mujoco/mjpro150/include -I/home/viktor/.local/lib/python3.5/site-packages/numpy/core/include -I/home/viktor/anaconda3/envs/mujoco/include/python3.5m -c /tmp/pip-build-20t9hi5v/mujoco-py/mujoco_py/cymj.c -o /tmp/pip-build-20t9hi5v/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build-20t9hi5v/mujoco-py/mujoco_py/cymj.o -fopenmp -w
  gcc -pthread -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -Imujoco_py -I/tmp/pip-build-20t9hi5v/mujoco-py/mujoco_py -I/home/viktor/.mujoco/mjpro150/include -I/home/viktor/.local/lib/python3.5/site-packages/numpy/core/include -I/home/viktor/anaconda3/envs/mujoco/include/python3.5m -c /tmp/pip-build-20t9hi5v/mujoco-py/mujoco_py/gl/osmesashim.c -o /tmp/pip-build-20t9hi5v/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build-20t9hi5v/mujoco-py/mujoco_py/gl/osmesashim.o -fopenmp -w
  /tmp/pip-build-20t9hi5v/mujoco-py/mujoco_py/gl/osmesashim.c:1:23: fatal error: GL/osmesa.h: No such file or directory
  compilation terminated.
  error: command 'gcc' failed with exit status 1```

yenchenlin, 2017-10-11:
same issue here.

keven425, 2017-10-14:
same issue

UIDILR, 2017-10-17:
sudo apt-get install libosmesa6-dev 
worked for me

gntoni, 2017-10-17:
I had the same problem. Solved it by installing the last version of mesa (17.0.2) 
https://www.mesa3d.org/relnotes/17.0.2.html

nrontsis, 2017-11-23:
A potentially unorthodox **conda-only** solution that worked for me is:
```
conda install -c clinicalgraphics osmesa 
```
and then update the `C_LIBRARY_PATH` variable accordingly:
```
export C_LIBRARY_PATH="YOUR-ANACONDA-ENVIRONMENT/include:$C_LIBRARY_PATH"
```

JoyChopra1298, 2018-02-02:
I have the same problem running it on a CentOS6 cluster. But I cannot find a .el6 rpm for latest version of mesa and the .el7(supported for CentOS7 gives [error ](https://superuser.com/questions/850276/how-to-resolve-rpm-dependencies) similar to this requesting for glib_2.14). 
Will mesa 11.0.7 work for any version of mujoco-py or does it strictly needs mesa 17

lmzintgraf, 2018-04-11:
Solved it for me on Fedora26:
`sudo dnf install mesa-libOSMesa-devel`

inspirepassion, 2018-09-02:
Still cannot solve it. Does anyone know how to deal with it?

(temp36) zhixun@zhixun-MS-7B48:~/Downloads/mujoco-py$ sudo apt-get install libosmesa6-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libosmesa6-dev : Depends: libosmesa6 (= 18.0.0~rc5-1ubuntu1) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

zhangchuheng123, 2019-04-11:
Anyone solve this problem on CentOS?

mrahtz, 2019-05-17:
On CentOS 7, the following worked for me to install the necessary packages without root:

```bash 
cd $(mktemp -d)
for url in http://mirror.centos.org/centos/7/os/x86_64/Packages/mesa-libOSMesa-18.0.5-3.el7.x86_64.rpm http://mirror.centos.org/centos/7/os/x86_64/Packages/mesa-libOSMesa-devel-18.0.5-3.el7.x86_64.rpm http://download-ib01.fedoraproject.org/pub/epel/7/x86_64/Packages/p/patchelf-0.9-10.el7.x86_64.rpm; do
    curl $url | rpm2cpio | cpio -id
done
mkdir ~/.local
cp -a usr/* ~/.local/
```

then adding the following lines to my `.bashrc`:
```bash
export PATH="$PATH:$HOME/.local/bin"
export LD_LIBRARY_PATH="$LD_LIBRARY_PATH:$HOME/.local/lib:$HOME/.local/lib64"
export LDFLAGS="-L$HOME/.local/lib -L$HOME/.local/lib64"
export CPATH="$CPATH:$HOME/.local/include"
```

`pip install mujoco-py` then worked fine.

samarjeet, 2019-05-21:
Installing https://pkgs.org/download/mesa-libOSMesa on CentOS worked for me.

GoingMyWay, 2019-09-18:
If you install openai's baselines, you should `python setup.py install` rather than `pip install -e .`

vinamrabenara, 2019-10-16:
try this for conda 

`conda install -c menpo osmesa`

muratgursu, 2019-10-29:
https://github.com/openai/mujoco-py/issues/90
^This solved it for me.

Duuuuuu, 2020-03-20:
> On CentOS 7, the following worked for me to install the necessary packages without root:
> 
> ```shell
> cd $(mktemp -d)
> for url in http://mirror.centos.org/centos/7/os/x86_64/Packages/mesa-libOSMesa-18.0.5-3.el7.x86_64.rpm http://mirror.centos.org/centos/7/os/x86_64/Packages/mesa-libOSMesa-devel-18.0.5-3.el7.x86_64.rpm http://download-ib01.fedoraproject.org/pub/epel/7/x86_64/Packages/p/patchelf-0.9-10.el7.x86_64.rpm; do
>     curl $url | rpm2cpio | cpio -id
> done
> mkdir ~/.local
> cp -a usr/* ~/.local/
> ```
> 
> then adding the following lines to my `.bashrc`:
> 
> ```shell
> export PATH="$PATH:$HOME/.local/bin"
> export LD_LIBRARY_PATH="$LD_LIBRARY_PATH:$HOME/.local/lib:$HOME/.local/lib64"
> export LDFLAGS="-L$HOME/.local/lib -L$HOME/.local/lib64"
> export CPATH="$CPATH:$HOME/.local/include"
> ```
> 
> `pip install mujoco-py` then worked fine.

For people who still have this problem on centos 7 with root access, it looks like the rpm links are not up-to-date anymore. We should use this:

```shell
cd $(mktemp -d)
for url in http://mirror.centos.org/centos/7/os/x86_64/Packages/mesa-libOSMesa-18.3.4-5.el7.x86_64.rpm http://mirror.centos.org/centos/7/os/x86_64/Packages/mesa-libOSMesa-devel-18.3.4-5.el7.x86_64.rpm; do
    curl $url | rpm2cpio | cpio -id
done
mkdir ~/.local
cp -a usr/* ~/.local/
```
always check the latest version of libOSMeas on https://pkgs.org/download/mesa-libOSMesa!

gfreund123, 2020-11-08:
> Based on https://stackoverflow.com/a/52567731/11063709 and the previous comments, for CentOS without root access, the following should work without having to manually update the package URLs:
> 
> ```
> mkdir -p ~/rpm
> yumdownloader --destdir ~/rpm --resolve mesa-libOSMesa.x86_64 mesa-libOSMesa-devel.x86_64 patchelf.x86_64
> cd ~/rpm
> for rpm in `ls`; do rpm2cpio $rpm | cpio -id ; done
> ```
> 
> ```
> export PATH="$PATH:$HOME/rpm/usr/bin"
> export LD_LIBRARY_PATH="$LD_LIBRARY_PATH:$HOME/rpm/usr/lib:$HOME/rpm/usr/lib64"
> export LDFLAGS="-L$HOME/rpm/usr/lib -L$HOME/rpm/usr/lib64"
> export CPATH="$CPATH:$HOME/rpm/usr/include"
> ```

@RaghuSpaceRajan  Hey, I had the original error and your solution worked for me in the base environment (I use Anaconda), but now when I've installed mujoco-py in another environment I get the following similar error when importing it: 

> distutils.errors.CompileError: command '/home/dsi/myname/anaconda3/envs/spinningup/bin/x86_64-conda_cos6-linux-gnu-cc' failed with exit status 1

Sorry for the noob question, but what do I need to add now to my .bashrc so that mujoco-py will also work in this environment?

RaghuSpaceRajan, 2020-11-19:
Hi @gfreund123 , unfortunately I don't think I have seen this error before. If it's related to your environments only, then you might want to either activate the environment in `.bashrc` or run the export commands again after activating the environment.