openai/mujoco-py #99: Stuck at Running setup.py bdist_wheel for mujoco-py ...

JiankaiSun-SJTU-MVIG-training, 2017-08-27:
System:
- Ubuntu 16.04 64-bit
- Python 3.5

In a `virtualenv` python 3.5 environment, after excuting `pip install mujoco-py` or `pip3 install -U 'mujoco-py<1.50.2,>=1.50.1'`, it gets stuck while building wheels (no response), and does not even raise an error.
The ~/.mujoco directory is prepared with a valid mjkey.txt. and the mujoco150/bin/simulate works well.

The details:
```
(python3) jack@jack:~$ pip install mujoco-py
Collecting mujoco-py
  Downloading mujoco-py-1.50.1.19.tar.gz (659kB)
    100% |████████████████████████████████| 665kB 1.1MB/s 
Requirement already satisfied: glfw>=1.4.0 in ./Applications/pythonEnv/python3/lib/python3.5/site-packages (from mujoco-py)
Requirement already satisfied: numpy>=1.11 in ./Applications/pythonEnv/python3/lib/python3.5/site-packages (from mujoco-py)
Requirement already satisfied: Cython>=0.25.2 in ./Applications/pythonEnv/python3/lib/python3.5/site-packages (from mujoco-py)
Requirement already satisfied: imageio>=2.1.2 in ./Applications/pythonEnv/python3/lib/python3.5/site-packages (from mujoco-py)
Requirement already satisfied: pillow in ./Applications/pythonEnv/python3/lib/python3.5/site-packages (from imageio>=2.1.2->mujoco-py)
Requirement already satisfied: olefile in ./Applications/pythonEnv/python3/lib/python3.5/site-packages (from pillow->imageio>=2.1.2->mujoco-py)
Building wheels for collected packages: mujoco-py
  Running setup.py bdist_wheel for mujoco-py ... -
```
All the packages I have installed in this virtual environment:
```
(python3) jack@jack:~$ pip list
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
ale-python-interface (0.0.1)
atari-py (0.1.1)
azure (1.0.3)
azure-common (1.1.8)
azure-mgmt (0.20.2)
azure-mgmt-common (0.20.0)
azure-mgmt-compute (0.20.1)
azure-mgmt-network (0.20.1)
azure-mgmt-nspkg (2.0.0)
azure-mgmt-resource (0.20.1)
azure-mgmt-storage (0.20.0)
azure-nspkg (2.0.0)
azure-servicebus (0.20.1)
azure-servicemanagement-legacy (0.20.2)
azure-storage (0.20.3)
bleach (1.5.0)
Box2D-kengz (2.3.3)
certifi (2017.7.27.1)
chardet (3.0.4)
Cython (0.26)
decorator (4.1.2)
dill (0.2.7.1)
entrypoints (0.2.3)
glfw (1.4.0)
graphviz (0.8)
gym (0.9.2)
html5lib (0.9999999)
idna (2.6)
imageio (2.2.0)
ipykernel (4.6.1)
ipython (6.1.0)
ipython-genutils (0.2.0)
ipywidgets (7.0.0)
jedi (0.10.2)
Jinja2 (2.9.6)
joblib (0.11)
jsonschema (2.6.0)
jupyter (1.0.0)
jupyter-client (5.1.0)
jupyter-console (5.2.0)
jupyter-core (4.3.0)
Keras (2.0.6)
Markdown (2.6.9)
MarkupSafe (1.0)
mistune (0.7.4)
mpi4py (2.0.0)
mxnet (0.11.0)
nbconvert (5.2.1)
nbformat (4.4.0)
notebook (5.0.0)
numpy (1.13.1)
olefile (0.44)
opencv-contrib-python (3.3.0.9)
pachi-py (0.0.21)
pandocfilters (1.4.2)
pexpect (4.2.1)
pickleshare (0.7.4)
Pillow (4.2.1)
pip (9.0.1)
progressbar2 (3.34.2)
prompt-toolkit (1.0.15)
protobuf (3.4.0)
ptyprocess (0.5.2)
pyglet (1.2.4)
Pygments (2.2.0)
PyOpenGL (3.1.0)
python-dateutil (2.6.1)
python-utils (2.2.0)
PyYAML (3.12)
pyzmq (16.0.2)
qtconsole (4.3.1)
requests (2.18.4)
scipy (0.19.1)
setuptools (36.2.7)
simplegeneric (0.8.1)
six (1.10.0)
tensorflow (1.3.0)
tensorflow-tensorboard (0.1.4)
terminado (0.6)
testpath (0.3.1)
Theano (0.9.0)
tornado (4.5.1)
tqdm (4.15.0)
traitlets (4.3.2)
urllib3 (1.22)
wcwidth (0.1.7)
Werkzeug (0.12.2)
wheel (0.29.0)
widgetsnbextension (3.0.2)
zmq (0.0.0)
```

Do you have any ideas to solve it? Thank you.

ThyrixYang, 2017-08-28:
Same issue, waiting for a solution.

Kukanani, 2017-09-03:
Instead of installing mujoco-py from pip, try cloning the source and installing it with `pip install -e .`. I still am having trouble getting everything to work, but at least I could make it past the install step.

JiankaiSun-SJTU-MVIG-training, 2017-09-04:
@Kukanani  Yes, I successfully installed mujoco-py in the way you provided. Thank you!

inksci, 2017-10-13:
@Kukanani Perfect solution!

sushrutt12, 2017-12-07:
I downloaded [mjpro150_win64](https://www.roboti.us/download/mjpro150_win64.zip)
and followed these [steps](https://github.com/openai/mujoco-py/blob/master/.appveyor.yml#L16-L32) ,
downloaded [mujoco-py](https://github.com/openai/mujoco-py/releases/tag/1.50.1.0)  prior to this tried version 1.50.1.34 which didn't workout don't know why.
then used `pip install mujoco-py-1.50.1.0.zip`.
After this you can import  mujoco_py
you may have to download [xmls](https://github.com/openai/mujoco-py) and paste it in "C:\ProgramData\Anaconda3\Lib\site-packages" to run the code
Using Windows 10. Python 3.6.3, mujoco-py-1.50.1.0 and mjpro150_win64. It finally works

enjoykit, 2017-12-26:
@Kukanani  I know this might be a stupid question. But may I ask what is the 'source'? Thanks.

Kukanani, 2017-12-28:
The source code of this repository.

auzyze, 2018-01-04:
@Kukanani Thanks for you method, "pip install -e ." works for my case. (ubuntu14.04, virtualenv with python 3.5.4)

azmosavi, 2018-04-05:
I have similar issue with installing mujoco as well.
For me every things is installed and located under the python virtual environment.
The pip install mujoco-py==0.5.7 works very well. But when is the turn for the
pip3 install -U 'mujoco-py<1.50.2,>=1.50.1'
I get the error of:
Exception: Please add path to mujoco library to your .bashrc:
export DYLD_LIBRARY_PATH=$DYLD_LIBRARY_PATH:/Users/user/.mujoco/mjpro150/bin
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/Users/user.mujoco/mjpro150/bin
While I have in my bashrc the path to the virtual py:
export DYLD_LIBRARY_PATH=$DYLD_LIBRARY_PATH:/Users/user/.pyenv/versions/3.6.1/envs/py_env_azi/lib/python3.6/site-packages/gym/envs/mujoco/mjpro150/bin
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/Users/user/.pyenv/versions/3.6.1/envs/py_env_azi/lib/python3.6/site-packages/gym/envs/mujoco/mjpro150/bin

So, I dont know why it can not figure it out the location of binary files although I have the correct path under my bashrc!