openai/mujoco-py #110: ImportError: No module named 'glfw'

watsonyanghx, 2017-09-23:
``` python
running install
    running build
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-hqdkt5rb/mujoco-py/setup.py", line 52, in <module>
        cmdclass={'build': Build},
      File "/home/yang/anaconda3/lib/python3.5/distutils/core.py", line 148, in setup
        dist.run_commands()
      File "/home/yang/anaconda3/lib/python3.5/distutils/dist.py", line 955, in run_commands
        self.run_command(cmd)
      File "/home/yang/anaconda3/lib/python3.5/distutils/dist.py", line 974, in run_command
        cmd_obj.run()
      File "/home/yang/anaconda3/lib/python3.5/site-packages/setuptools-27.2.0-py3.5.egg/setuptools/command/install.py", line 61, in run
      File "/home/yang/anaconda3/lib/python3.5/distutils/command/install.py", line 539, in run
        self.run_command('build')
      File "/home/yang/anaconda3/lib/python3.5/distutils/cmd.py", line 313, in run_command
        self.distribution.run_command(command)
      File "/home/yang/anaconda3/lib/python3.5/distutils/dist.py", line 974, in run_command
        cmd_obj.run()
      File "/tmp/pip-build-hqdkt5rb/mujoco-py/setup.py", line 19, in run
        spec.loader.exec_module(builder)
      File "<frozen importlib._bootstrap_external>", line 678, in exec_module
      File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
      File "/tmp/pip-build-hqdkt5rb/mujoco-py/mujoco_py/builder.py", line 17, in <module>
        from mujoco_py.utils import discover_mujoco
      File "/tmp/pip-build-hqdkt5rb/mujoco-py/mujoco_py/__init__.py", line 4, in <module>
        from mujoco_py.mjviewer import MjViewer, MjViewerBasic
      File "/tmp/pip-build-hqdkt5rb/mujoco-py/mujoco_py/mjviewer.py", line 2, in <module>
        import glfw
      File "/home/yang/anaconda3/lib/python3.5/site-packages/glfw.py", line 194, in <module>
        raise ImportError("Failed to load GLFW3 shared library.")
    ImportError: Failed to load GLFW3 shared library.
``` 

Ubuntu 16.04, Anaconda 3.5

I have installed pyopengl. I google it, but find nothing about GLFW.

Is there anyone konw how to fix this? Thanks.

watsonyanghx, 2017-09-23:
What did I do, I just submit one issue, why are there so many? What is wrong therr?

watsonyanghx, 2017-09-23:
I'm so sorry, I closed other issues. Sorry again.

machinaut, 2017-09-28:
Not exactly sure in this case, but this can happen if the MuJoCo-distributed GLFW library is not in your path.  This only is an issue during installation, as mujoco-py is able to fixup the library path during normal use.

Can you try running the install with `LD_LIBRARY_PATH=$HOME/.mujoco/mjpro150/bin pip install mujoco-py`?

ViktorM, 2017-10-09:
Have the same error and ```LD_LIBRARY_PATH=$HOME/.mujoco/mjpro150/bin pip install mujoco-py``` didn't help.

machinaut, 2017-10-12:
Is there a glfw library in the `$HOME/.mujoco/mjpro150/bin` directory for python to load?

ViktorM, 2017-10-13:
Yes it was. After following recommendations from readme I've got this error: https://github.com/openai/mujoco-py/issues/75 or what is the same: https://github.com/openai/mujoco-py/issues/44 Then I rried older 0.5.7 version and it started working for me out of the box with just ```pip install mujoco-py==0.5.7```

keven425, 2017-10-14:
Running into the same issue. Adding `LD_LIBRARY_PATH=$HOME/.mujoco/mjpro150/bin` didn't help. 

Here's the content of the bin directory:
```
$ ls $HOME/.mujoco/mjpro150/bin
basic    derivative     libglewosmesa.so  libglfw.so.3         libmujoco150.so  MUJOCO_LOG.TXT  simulate
compile  libglewegl.so  libglew.so        libmujoco150nogl.so  mjkey.txt        record          test
```

xyc2690, 2017-11-17:
ImportError: Failed to load GLFW3 shared library.
I got the same problem.
Solved by :
`sudo cp mjpro150/bin/libglfw.so.3 /usr/local/lib`

keven425, 2017-11-20:
OK this solved my issue on ubuntu:

```
sudo apt-get install libglfw3-dev
```

kmotohas, 2018-03-21:
This works for me.
```
# $HOME/.bashrc
export MUJOCO_HOME=$HOME/.mujoco/mjpro150
exoprt LD_LIBRARY_PATH=$MUJOCO_HOME/bin:$LD_LIBRARY_PATH
```

HareshMiriyala, 2019-01-05:
Still doesnt work for me

zheyuanWang, 2020-01-11:
uninstall them(the packages with "glfw" in their name), 
then reinstall it with `pip3 install glfw` solved my problem of `ImportError: Failed to load GLFW3 shared library.`