openai/mujoco-py #131: AttributeError: type object 'mujoco_py.cymj.PyMjVisual_headlight' has no attribute '__reduce_cython__'

MichaelSnowden, 2017-09-24:
I get this error when installing mujoco-py 1.50.1.21 on macOS with Python 3.5.2: ```AttributeError: type object 'mujoco_py.cymj.PyMjVisual_headlight' has no attribute '__reduce_cython__'```. I followed the [instructions here](https://github.com/openai/mujoco-py)

Full stack trace:

```
(3.5) $ python --version
Python 3.5.2
(3.5) $ pip install mujoco-py
Collecting mujoco-py
  Using cached mujoco-py-1.50.1.21.tar.gz
Requirement already satisfied: glfw>=1.4.0 in ./.virtualenvs/3.5/lib/python3.5/site-packages (from mujoco-py)
Requirement already satisfied: numpy>=1.11 in ./.virtualenvs/3.5/lib/python3.5/site-packages (from mujoco-py)
Requirement already satisfied: Cython>=0.25.2 in ./.virtualenvs/3.5/lib/python3.5/site-packages (from mujoco-py)
Requirement already satisfied: imageio>=2.1.2 in ./.virtualenvs/3.5/lib/python3.5/site-packages (from mujoco-py)
Requirement already satisfied: pillow in ./.virtualenvs/3.5/lib/python3.5/site-packages (from imageio>=2.1.2->mujoco-py)
Requirement already satisfied: olefile in ./.virtualenvs/3.5/lib/python3.5/site-packages (from pillow->imageio>=2.1.2->mujoco-py)
Building wheels for collected packages: mujoco-py
  Running setup.py bdist_wheel for mujoco-py ... error
  Complete output from command /Users/michael.snowden/.virtualenvs/3.5/bin/python -u -c "import setuptools, tokenize;__file__='/private/var/folders/nn/3dq7tlzd7s705505x7w8h1400000gn/T/pip-build-2et2y95k/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /var/folders/nn/3dq7tlzd7s705505x7w8h1400000gn/T/tmpdrx4rdyipip-wheel- --python-tag cp35:
  running bdist_wheel
  running build
  Compiling /private/var/folders/nn/3dq7tlzd7s705505x7w8h1400000gn/T/pip-build-2et2y95k/mujoco-py/mujoco_py/cymj.pyx because it changed.
  [1/1] Cythonizing /private/var/folders/nn/3dq7tlzd7s705505x7w8h1400000gn/T/pip-build-2et2y95k/mujoco-py/mujoco_py/cymj.pyx
  running build_ext
  building 'mujoco_py.cymj' extension
  creating /private/var/folders/nn/3dq7tlzd7s705505x7w8h1400000gn/T/pip-build-2et2y95k/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder
  creating /private/var/folders/nn/3dq7tlzd7s705505x7w8h1400000gn/T/pip-build-2et2y95k/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.12-x86_64-3.5
  creating /private/var/folders/nn/3dq7tlzd7s705505x7w8h1400000gn/T/pip-build-2et2y95k/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.12-x86_64-3.5/private
  creating /private/var/folders/nn/3dq7tlzd7s705505x7w8h1400000gn/T/pip-build-2et2y95k/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.12-x86_64-3.5/private/var
  creating /private/var/folders/nn/3dq7tlzd7s705505x7w8h1400000gn/T/pip-build-2et2y95k/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.12-x86_64-3.5/private/var/folders
  creating /private/var/folders/nn/3dq7tlzd7s705505x7w8h1400000gn/T/pip-build-2et2y95k/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.12-x86_64-3.5/private/var/folders/nn
  creating /private/var/folders/nn/3dq7tlzd7s705505x7w8h1400000gn/T/pip-build-2et2y95k/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.12-x86_64-3.5/private/var/folders/nn/3dq7tlzd7s705505x7w8h1400000gn
  creating /private/var/folders/nn/3dq7tlzd7s705505x7w8h1400000gn/T/pip-build-2et2y95k/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.12-x86_64-3.5/private/var/folders/nn/3dq7tlzd7s705505x7w8h1400000gn/T
  creating /private/var/folders/nn/3dq7tlzd7s705505x7w8h1400000gn/T/pip-build-2et2y95k/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.12-x86_64-3.5/private/var/folders/nn/3dq7tlzd7s705505x7w8h1400000gn/T/pip-build-2et2y95k
  creating /private/var/folders/nn/3dq7tlzd7s705505x7w8h1400000gn/T/pip-build-2et2y95k/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.12-x86_64-3.5/private/var/folders/nn/3dq7tlzd7s705505x7w8h1400000gn/T/pip-build-2et2y95k/mujoco-py
  creating /private/var/folders/nn/3dq7tlzd7s705505x7w8h1400000gn/T/pip-build-2et2y95k/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.12-x86_64-3.5/private/var/folders/nn/3dq7tlzd7s705505x7w8h1400000gn/T/pip-build-2et2y95k/mujoco-py/mujoco_py
  creating /private/var/folders/nn/3dq7tlzd7s705505x7w8h1400000gn/T/pip-build-2et2y95k/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.12-x86_64-3.5/private/var/folders/nn/3dq7tlzd7s705505x7w8h1400000gn/T/pip-build-2et2y95k/mujoco-py/mujoco_py/gl
  /usr/local/bin/gcc-7 -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -DONMAC -Imujoco_py -I/private/var/folders/nn/3dq7tlzd7s705505x7w8h1400000gn/T/pip-build-2et2y95k/mujoco-py/mujoco_py -I/Users/michael.snowden/.mujoco/mjpro150/include -I/Users/michael.snowden/.virtualenvs/3.5/lib/python3.5/site-packages/numpy/core/include -I/Users/michael.snowden/.pyenv/versions/3.5.2/include/python3.5m -c /private/var/folders/nn/3dq7tlzd7s705505x7w8h1400000gn/T/pip-build-2et2y95k/mujoco-py/mujoco_py/cymj.c -o /private/var/folders/nn/3dq7tlzd7s705505x7w8h1400000gn/T/pip-build-2et2y95k/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.12-x86_64-3.5/private/var/folders/nn/3dq7tlzd7s705505x7w8h1400000gn/T/pip-build-2et2y95k/mujoco-py/mujoco_py/cymj.o -fopenmp -w
  /usr/local/bin/gcc-7 -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -DONMAC -Imujoco_py -I/private/var/folders/nn/3dq7tlzd7s705505x7w8h1400000gn/T/pip-build-2et2y95k/mujoco-py/mujoco_py -I/Users/michael.snowden/.mujoco/mjpro150/include -I/Users/michael.snowden/.virtualenvs/3.5/lib/python3.5/site-packages/numpy/core/include -I/Users/michael.snowden/.pyenv/versions/3.5.2/include/python3.5m -c /private/var/folders/nn/3dq7tlzd7s705505x7w8h1400000gn/T/pip-build-2et2y95k/mujoco-py/mujoco_py/gl/dummyshim.c -o /private/var/folders/nn/3dq7tlzd7s705505x7w8h1400000gn/T/pip-build-2et2y95k/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.12-x86_64-3.5/private/var/folders/nn/3dq7tlzd7s705505x7w8h1400000gn/T/pip-build-2et2y95k/mujoco-py/mujoco_py/gl/dummyshim.o -fopenmp -w
  creating /private/var/folders/nn/3dq7tlzd7s705505x7w8h1400000gn/T/pip-build-2et2y95k/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/lib.macosx-10.12-x86_64-3.5
  creating /private/var/folders/nn/3dq7tlzd7s705505x7w8h1400000gn/T/pip-build-2et2y95k/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/lib.macosx-10.12-x86_64-3.5/mujoco_py
  /usr/local/bin/gcc-7 -bundle -undefined dynamic_lookup -L/usr/local/opt/readline/lib -L/usr/local/opt/readline/lib -L/usr/local/opt/openssl/lib -L/Users/michael.snowden/.pyenv/versions/3.5.2/lib /private/var/folders/nn/3dq7tlzd7s705505x7w8h1400000gn/T/pip-build-2et2y95k/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.12-x86_64-3.5/private/var/folders/nn/3dq7tlzd7s705505x7w8h1400000gn/T/pip-build-2et2y95k/mujoco-py/mujoco_py/cymj.o /private/var/folders/nn/3dq7tlzd7s705505x7w8h1400000gn/T/pip-build-2et2y95k/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/temp.macosx-10.12-x86_64-3.5/private/var/folders/nn/3dq7tlzd7s705505x7w8h1400000gn/T/pip-build-2et2y95k/mujoco-py/mujoco_py/gl/dummyshim.o -L/Users/michael.snowden/.mujoco/mjpro150/bin -L/Users/michael.snowden/.mujoco/mjpro150/bin -lmujoco150 -lglfw.3 -o /private/var/folders/nn/3dq7tlzd7s705505x7w8h1400000gn/T/pip-build-2et2y95k/mujoco-py/mujoco_py/generated/_pyxbld_MacExtensionBuilder/lib.macosx-10.12-x86_64-3.5/mujoco_py/cymj.cpython-35m-darwin.so -fopenmp
  Traceback (most recent call last):
    File "<string>", line 1, in <module>
    File "/private/var/folders/nn/3dq7tlzd7s705505x7w8h1400000gn/T/pip-build-2et2y95k/mujoco-py/setup.py", line 52, in <module>
      cmdclass={'build': Build},
    File "/Users/michael.snowden/.pyenv/versions/3.5.2/lib/python3.5/distutils/core.py", line 148, in setup
      dist.run_commands()
    File "/Users/michael.snowden/.pyenv/versions/3.5.2/lib/python3.5/distutils/dist.py", line 955, in run_commands
      self.run_command(cmd)
    File "/Users/michael.snowden/.pyenv/versions/3.5.2/lib/python3.5/distutils/dist.py", line 974, in run_command
      cmd_obj.run()
    File "/Users/michael.snowden/.virtualenvs/3.5/lib/python3.5/site-packages/wheel/bdist_wheel.py", line 204, in run
      self.run_command('build')
[... cut ...]
    File "<frozen importlib._bootstrap>", line 693, in _load
    File "<frozen importlib._bootstrap>", line 673, in _load_unlocked
    File "<frozen importlib._bootstrap_external>", line 913, in exec_module
    File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
    File "mujoco_py/generated/wrappers.pxi", line 11, in init mujoco_py.cymj
      cdef class PyMjVisual_headlight(object):
  AttributeError: type object 'mujoco_py.cymj.PyMjVisual_headlight' has no attribute '__reduce_cython__'
```

Kelym, 2017-09-25:
I have exactly the same error and went with [this suggestion](https://github.com/openai/mujoco-py/issues/66#issuecomment-327960438). Basically: 
```git clone the mujoco-py repo```
 ```cd mujoco-py```
```python3 setup.py install```
```python3``` 
```import mujoco-py``` 
Seemed to work. I didn't play with it much and ended up going back with mujoco-py 0.57 though because of OpenAI gym.

MichaelSnowden, 2017-09-25:
Yeah, I gotta use gym unfortunately, so I ended up just running `pip install mujoco-py==0.5.7`. I tried virtually every `1.x.y` version and none of them worked--seems like a pretty major issue.

machinaut, 2017-09-26:
I can replicate this.  It looks like there's an issue with the latest (0.27) version of Cython and mujoco-py.

If you downgrade to Cython 0.26 does it work for you?  @MichaelSnowden

machinaut, 2017-09-26:
Also installing from source (like @Kelym recommended) works for me with Cython=0.27.

jonasschneider, 2017-09-27:
A workaround is to install `Cython==0.25.2` strictly before attempting to install `mujoco-py` -- we'll look into what the root cause is with the new Cython eventually.

scoder, 2017-10-27:
This is fixed in Cython 0.27.1 and 0.27.2.

PR #151 (2017-11-03): bump cython to 0.27.2
It looks like the recent patches fixed the issue we had with newer cython.

Fixes #131
closed 2017-11-03 by machinaut