openai/mujoco-py #132: error: command 'x86_64-linux-gnu-gcc' failed with exit status 1

donamin, 2017-09-26:
Hi
I get this weird error while trying to install mujoco-py on Linux Mint.
Any idea how to solve this?
Thanks

```
Collecting mujoco-py
  Downloading mujoco-py-1.50.1.21.tar.gz (110kB)
    100% |████████████████████████████████| 112kB 4.1MB/s 
Requirement already satisfied: glfw>=1.4.0 in /usr/local/lib/python3.5/dist-packages (from mujoco-py)
Requirement already satisfied: numpy>=1.11 in /usr/local/lib/python3.5/dist-packages (from mujoco-py)
Requirement already satisfied: Cython>=0.25.2 in /usr/local/lib/python3.5/dist-packages (from mujoco-py)
Requirement already satisfied: imageio>=2.1.2 in /usr/local/lib/python3.5/dist-packages (from mujoco-py)
Requirement already satisfied: pillow in /usr/lib/python3/dist-packages (from imageio>=2.1.2->mujoco-py)
Building wheels for collected packages: mujoco-py
  Running setup.py bdist_wheel for mujoco-py ... error
  Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-bx14fko6/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmpxgstypt0pip-wheel- --python-tag cp35:
  running bdist_wheel
  running build
  Compiling /tmp/pip-build-bx14fko6/mujoco-py/mujoco_py/cymj.pyx because it changed.
  [1/1] Cythonizing /tmp/pip-build-bx14fko6/mujoco-py/mujoco_py/cymj.pyx
  running build_ext
  building 'mujoco_py.cymj' extension
  creating /tmp/pip-build-bx14fko6/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder
  creating /tmp/pip-build-bx14fko6/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5
  creating /tmp/pip-build-bx14fko6/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp
  creating /tmp/pip-build-bx14fko6/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build-bx14fko6
  creating /tmp/pip-build-bx14fko6/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build-bx14fko6/mujoco-py
  creating /tmp/pip-build-bx14fko6/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build-bx14fko6/mujoco-py/mujoco_py
  creating /tmp/pip-build-bx14fko6/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build-bx14fko6/mujoco-py/mujoco_py/gl
  x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -Imujoco_py -I/tmp/pip-build-bx14fko6/mujoco-py/mujoco_py -I/root/.mujoco/mjpro150/include -I/usr/local/lib/python3.5/dist-packages/numpy/core/include -I/usr/include/python3.5m -c /tmp/pip-build-bx14fko6/mujoco-py/mujoco_py/cymj.c -o /tmp/pip-build-bx14fko6/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build-bx14fko6/mujoco-py/mujoco_py/cymj.o -fopenmp -w
  /tmp/pip-build-bx14fko6/mujoco-py/mujoco_py/cymj.c:48:20: fatal error: Python.h: No such file or directory
  compilation terminated.
  error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
  
  ----------------------------------------
  Failed building wheel for mujoco-py
  Running setup.py clean for mujoco-py
Failed to build mujoco-py
Installing collected packages: mujoco-py
  Running setup.py install for mujoco-py ... error
    Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-bx14fko6/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-5mcyuabw-record/install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    running build_ext
    building 'mujoco_py.cymj' extension
    x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -Imujoco_py -I/tmp/pip-build-bx14fko6/mujoco-py/mujoco_py -I/root/.mujoco/mjpro150/include -I/usr/local/lib/python3.5/dist-packages/numpy/core/include -I/usr/include/python3.5m -c /tmp/pip-build-bx14fko6/mujoco-py/mujoco_py/cymj.c -o /tmp/pip-build-bx14fko6/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/tmp/pip-build-bx14fko6/mujoco-py/mujoco_py/cymj.o -fopenmp -w
    /tmp/pip-build-bx14fko6/mujoco-py/mujoco_py/cymj.c:48:20: fatal error: Python.h: No such file or directory
    compilation terminated.
    error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
    
    ----------------------------------------
Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-bx14fko6/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-5mcyuabw-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-bx14fko6/mujoco-py/
```

donamin, 2017-09-26:
OK I managed to install mujoco-py from source code using `pip3 install -e .`
But now when I import mujoco-py into my code, I get the same error:

```
running build_ext
building 'mujoco_py.cymj' extension
x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -Imujoco_py -I/usr/local/mujoco-py/mujoco_py -I/home/amin/.mujoco/mjpro150/include -I/usr/local/lib/python3.5/dist-packages/numpy/core/include -I/usr/include/python3.5m -c /usr/local/mujoco-py/mujoco_py/cymj.c -o /usr/local/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/usr/local/mujoco-py/mujoco_py/cymj.o -fopenmp -w
/usr/local/mujoco-py/mujoco_py/cymj.c:560:21: fatal error: mjmodel.h: No such file or directory
compilation terminated.
Traceback (most recent call last):
  File "/usr/lib/python3.5/distutils/unixccompiler.py", line 118, in _compile
    extra_postargs)
  File "/usr/lib/python3.5/distutils/ccompiler.py", line 909, in spawn
    spawn(cmd, dry_run=self.dry_run)
  File "/usr/lib/python3.5/distutils/spawn.py", line 36, in spawn
    _spawn_posix(cmd, search_path, dry_run=dry_run)
  File "/usr/lib/python3.5/distutils/spawn.py", line 159, in _spawn_posix
    % (cmd, exit_status))
distutils.errors.DistutilsExecError: command 'x86_64-linux-gnu-gcc' failed with exit status 1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/mujoco-py/mujoco_py/__init__.py", line 1, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/usr/local/mujoco-py/mujoco_py/builder.py", line 296, in <module>
    cymj = load_cython_ext(mjpro_path)
  File "/usr/local/mujoco-py/mujoco_py/builder.py", line 56, in load_cython_ext
    cext_so_path = builder.build()
  File "/usr/local/mujoco-py/mujoco_py/builder.py", line 119, in build
    built_so_file_path = self._build_impl()
  File "/usr/local/mujoco-py/mujoco_py/builder.py", line 139, in _build_impl
    dist.run_commands()
  File "/usr/lib/python3.5/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/usr/lib/python3.5/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/usr/local/lib/python3.5/dist-packages/Cython/Distutils/old_build_ext.py", line 185, in run
    _build_ext.build_ext.run(self)
  File "/usr/lib/python3.5/distutils/command/build_ext.py", line 338, in run
    self.build_extensions()
  File "/usr/local/mujoco-py/mujoco_py/builder.py", line 77, in build_extensions
    build_ext.build_extensions(self)
  File "/usr/local/lib/python3.5/dist-packages/Cython/Distutils/old_build_ext.py", line 193, in build_extensions
    self.build_extension(ext)
  File "/usr/lib/python3.5/distutils/command/build_ext.py", line 532, in build_extension
    depends=ext.depends)
  File "/usr/lib/python3.5/distutils/ccompiler.py", line 574, in compile
    self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
  File "/usr/lib/python3.5/distutils/unixccompiler.py", line 120, in _compile
    raise CompileError(msg)
distutils.errors.CompileError: command 'x86_64-linux-gnu-gcc' failed with exit status 1
```

donamin, 2017-09-27:
I found what I did wrong.
I put the mjpro150 folder in /.mujoco instead of ~/.mujoco
The latter is the right address. Maybe someone will find that useful.

closed 2017-09-27 by donamin

HassamSheikh, 2018-02-06:
getting the same error. You meant making a ./mujoco folfer in the home directory? right?

MariaEckstein, 2019-05-06:
I am having the same error, and moving mjpro150 to ~./mujoco did not help. Any other ideas?

christianjcc, 2020-05-16:
am experiencing the same issue I have my mjkey.txt file in the following directory,

~/.mujoco/mjpro150/mjkey.txt

Was anyone able to resolve the issue? Any insight would be appreciated.

TheSpaceSheep, 2020-06-18:
Here is what I did wrong : when unzipping the installation folder, it created a new directory so that my path was `~/.mujoco/mujoco200/mujoco200_linux` instead of `~/.mujoco/mujoco200`.

Copy-pasting all the files to the parent directory solved the issue.

ashutoshtiwari13, 2020-07-20:
Installing the following developmental header worked for me , 
`$ sudo apt-get install libosmesa6-dev`

MartinaRuocco, 2021-03-11:
for me, it worked following [this](https://github.com/openai/mujoco-py/issues/265#issuecomment-401480621) alas installing python3.x-dev (in my case, 3.7)