openai/mujoco-py #133: `sim.render` causes view to freeze

lobachevzky, 2017-09-28:
If I add `sim.render(255, 255)` to the `body_interaction.py`--
```
while True:
    sim.data.ctrl[0] = math.cos(t / 10.) * 0.01
    sim.data.ctrl[1] = math.sin(t / 10.) * 0.01
    t += 1
    sim.step()
    viewer.render()
    array = sim.render(255, 255)
```

-- it causes the main camera to freeze.

This is a minimal example. In my actual program, attempting to capture a camera image as a numpy array is causing a different problem -- it displays the camera view superimposed on top of the main camera view, darkens the main camera view, and makes objects transparent:

[image]

Why is this happening and what is the correct way to capture a camera image as a numpy array?

machinaut, 2017-10-12:
this looks like what might happen if for some reason the GL buffers are being used for two things at once (and being drawn to without blanking from two different places).

The viewer has a screenshot ability, I think.  Would that work for your purposes (simultaneous viewer and render)?

lobachevzky, 2017-10-12:
Yes, it might. By the way, I'm on a new machine and it is now having _this_ problem: https://github.com/openai/mujoco-py/issues/44

I'll close this one as soon as I have rendering up and running on the new machine and can confirm that your solution works.

lobachevzky, 2017-11-01:
The fix for this problem is in this pull request: https://github.com/openai/mujoco-py/pull/145
If you are simultaneously rendering on and off screen, you need to call mjr_setBuffer in between renders (every timestep).

closed 2017-11-01 by lobachevzky