openai/mujoco-py #137: Within docker: ERROR: Invalid activation key

inksci, 2017-10-11:
What is the problem?
`root@ba8d364eb3cc:/mujoco_py# python3
Python 3.5.2 (default, Sep 14 2017, 22:51:06) 
[GCC 5.4.0 20160609] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import mujoco_py
Compiling /mujoco_py/mujoco_py/cymj.pyx because it changed.
[1/1] Cythonizing /mujoco_py/mujoco_py/cymj.pyx
running build_ext
building 'mujoco_py.cymj' extension
creating /mujoco_py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder
creating /mujoco_py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5
creating /mujoco_py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/mujoco_py
creating /mujoco_py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/mujoco_py/mujoco_py
creating /mujoco_py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/mujoco_py/mujoco_py/gl
x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -Imujoco_py -I/mujoco_py/mujoco_py -I/root/.mujoco/mjpro150/include -I/usr/lib/python3/dist-packages/numpy/core/include -I/usr/include/python3.5m -c /mujoco_py/mujoco_py/cymj.c -o /mujoco_py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/mujoco_py/mujoco_py/cymj.o -fopenmp -w
x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -Imujoco_py -I/mujoco_py/mujoco_py -I/root/.mujoco/mjpro150/include -I/usr/lib/python3/dist-packages/numpy/core/include -I/usr/include/python3.5m -c /mujoco_py/mujoco_py/gl/osmesashim.c -o /mujoco_py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/mujoco_py/mujoco_py/gl/osmesashim.o -fopenmp -w
creating /mujoco_py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/lib.linux-x86_64-3.5
creating /mujoco_py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/lib.linux-x86_64-3.5/mujoco_py
x86_64-linux-gnu-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 /mujoco_py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/mujoco_py/mujoco_py/cymj.o /mujoco_py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/mujoco_py/mujoco_py/gl/osmesashim.o -L/root/.mujoco/mjpro150/bin -Wl,--enable-new-dtags,-R/root/.mujoco/mjpro150/bin -lmujoco150 -lglewosmesa -lOSMesa -lGL -o /mujoco_py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/lib.linux-x86_64-3.5/mujoco_py/cymj.cpython-35m-x86_64-linux-gnu.so -fopenmp
ERROR: Invalid activation key

Press Enter to exit ...`

nejyeah, 2017-10-13:
Same problem.  It seems the key has something with the hardware of machine.

inksci, 2017-10-14:
I received an email from Roboti LLC.

The hardware-locking mechanism in MuJoCo does not work with Docker.  The only way to use MuJoCo with Docker is to have an institutional license, which comes with an unlocked activation key.

But what is the institutional license?

machinaut, 2017-11-03:
License questions are better asked on the [MuJoCo Forum](http://www.mujoco.org/forum/index.php).

We don't handle licenses (or even mujoco releases) -- this is just a library which provides python bindings.

closed 2017-11-03 by machinaut