openai/mujoco-py #141: ubuntu mujoco-py segmentation fault

taochenshh, 2017-10-12:
I am using ubuntu 14.04, nvidia-375, anaconda 2 with python 3.5. I have successfully installed mujoco_py. But when I ran the example `body_interaction.py`, the program just crashed. The error is:
```bash
GLFW error (code %d): %s 65542 b'EGL: Failed to get EGL display: Success'
Creating window glfw
Segmentation fault (core dumped)
```
I tried to use gdb to debug, it showed:
```bash
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
GLFW error (code %d): %s 65542 b'EGL: Failed to get EGL display: Success'
Creating window glfw

Program received signal SIGSEGV, Segmentation fault.
0x00007fffefccd28e in glGetString ()
   from /usr/lib/x86_64-linux-gnu/libOSMesa.so.6
```

taochenshh, 2017-10-13:
I have found solved this problem partly. For those who also have this problem, please check https://github.com/openai/mujoco-py/pull/138/files and https://github.com/openai/mujoco-py/issues/47. The major problem is that there is a hard-coded path in mujoco_py/builder.py: `/usr/local/nvidia/lib64`. In my case, it should be `/usr/lib/nvidia-375/lib64`. Also my `sys.platform` outputs `linux2`, so I changed this line to `sys.platform.startswith("linux")`

inksci, 2017-10-13:
I cannot find the relative pathes (used in mujoco_py/builder.py) in my ubuntu:

```
exbot@exbot:~$ cd /usr/local/nvidia/lib64/
bash: cd: /usr/local/nvidia/lib64/: No such file or directory
exbot@exbot:~$ cd /usr/local/bin/gcc-6
bash: cd: /usr/local/bin/gcc-6: No such file or directory
exbot@exbot:~$ cd /usr/local/bin/gcc-7
bash: cd: /usr/local/bin/gcc-7: No such file or directory
```

inksci, 2017-10-14:
# return "/usr/lib/x86_64-linux-gnu"
    return "/usr/lib/i386-linux-gnu"

closed 2017-10-14 by taochenshh

rojas70, 2021-04-21:
I had this happen, but with changes listed in https://github.com/openai/mujoco-py/pull/145 got mine to disappear.