openai/mujoco-py #147: Installation/Importting Error

foolyc, 2017-10-19:
`sudo pip3 install -e .`

then `sudo ./examples/simpool.py`, it comes with
`
running build_ext
building 'mujoco_py.cymj' extension
x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -Imujoco_py -I/home/foolyc/local_install/mujoco-py/mujoco_py -I/home/foolyc/.mujoco/mjpro150/include -I/usr/local/lib/python3.5/dist-packages/numpy/core/include -I/home/foolyc/local_install/mujoco-py/mujoco_py/vendor/egl -I/usr/include/python3.5m -c /home/foolyc/local_install/mujoco-py/mujoco_py/cymj.c -o /home/foolyc/local_install/mujoco-py/mujoco_py/generated/_pyxbld_LinuxGPUExtensionBuilder/temp.linux-x86_64-3.5/home/foolyc/local_install/mujoco-py/mujoco_py/cymj.o -fopenmp -w

x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -Imujoco_py -I/home/foolyc/local_install/mujoco-py/mujoco_py -I/home/foolyc/.mujoco/mjpro150/include -I/usr/local/lib/python3.5/dist-packages/numpy/core/include -I/home/foolyc/local_install/mujoco-py/mujoco_py/vendor/egl -I/usr/include/python3.5m -c /home/foolyc/local_install/mujoco-py/mujoco_py/gl/eglshim.c -o /home/foolyc/local_install/mujoco-py/mujoco_py/generated/_pyxbld_LinuxGPUExtensionBuilder/temp.linux-x86_64-3.5/home/foolyc/local_install/mujoco-py/mujoco_py/gl/eglshim.o -fopenmp -w
x86_64-linux-gnu-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 /home/foolyc/local_install/mujoco-py/mujoco_py/generated/_pyxbld_LinuxGPUExtensionBuilder/temp.linux-x86_64-3.5/home/foolyc/local_install/mujoco-py/mujoco_py/cymj.o /home/foolyc/local_install/mujoco-py/mujoco_py/generated/_pyxbld_LinuxGPUExtensionBuilder/temp.linux-x86_64-3.5/home/foolyc/local_install/mujoco-py/mujoco_py/gl/eglshim.o -L/home/foolyc/.mujoco/mjpro150/bin -Wl,--enable-new-dtags,-R/home/foolyc/.mujoco/mjpro150/bin -lmujoco150 -lglewegl -o /home/foolyc/local_install/mujoco-py/mujoco_py/generated/_pyxbld_LinuxGPUExtensionBuilder/lib.linux-x86_64-3.5/mujoco_py/cymj.cpython-35m-x86_64-linux-gnu.so -fopenmp
stat: No such file or directory
Traceback (most recent call last):
  File "./examples/simpool.py", line 4, in <module>
    from mujoco_py import load_model_from_xml, MjSim, MjSimPool
  File "/home/foolyc/local_install/mujoco-py/mujoco_py/__init__.py", line 1, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/home/foolyc/local_install/mujoco-py/mujoco_py/builder.py", line 439, in <module>
    cymj = load_cython_ext(mjpro_path)
  File "/home/foolyc/local_install/mujoco-py/mujoco_py/builder.py", line 80, in load_cython_ext
    cext_so_path = builder.build()
  File "/home/foolyc/local_install/mujoco-py/mujoco_py/builder.py", line 187, in build
    built_so_file_path = self._build_impl()
  File "/home/foolyc/local_install/mujoco-py/mujoco_py/builder.py", line 249, in _build_impl
    join(get_nvidia_lib_dir(), 'libOpenGL.so.0'))
  File "/home/foolyc/local_install/mujoco-py/mujoco_py/builder.py", line 122, in fix_shared_library
    so_file])
  File "/usr/lib/python3.5/subprocess.py", line 581, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['patchelf', '--add-needed', '/usr/lib/nvidia-375/libOpenGL.so.0', '/home/foolyc/local_install/mujoco-py/mujoco_py/generated/_pyxbld_LinuxGPUExtensionBuilder/lib.linux-x86_64-3.5/mujoco_py/cymj.cpython-35m-x86_64-linux-gnu.so']' returned non-zero exit status 1
`
Any one

foolyc, 2017-10-20:
I have solved this bug:

- install the mujoco_py from source code , `sudo pip3 install -e .`
- intstall patchelf 0.9 from source code [patchelf]( https://nixos.org/releases/patchelf/patchelf-0.9/patchelf-0.9.tar.gz)
- `cd mjpro150/bin` and `sudo cp *.so /usr/local/lib/`, so that glfw.py can load shared lib
- then exec `sudo python3 ./examples/simpool.py` to build the mujoco_py.cymj

closed 2017-10-26 by machinaut

abhigenie92, 2017-11-08:
Hi I was getting sort of similar error: https://github.com/openai/mujoco-py/issues/164.
A few questions
1. From source you mean? Would that be python setup.py install?
2. How did you install patchelf?

Is there a reason to using sudo?

josyulakrishna, 2017-11-10:
Hi @foolyc 
I have followed all the steps you suggested but i still get this error

`running build_ext
stat: No such file or directory
Traceback (most recent call last):
  File "./examples/simpool.py", line 4, in <module>
    from mujoco_py import load_model_from_xml, MjSim, MjSimPool
  File "/home/josyula/mujoco-py/mujoco_py/__init__.py", line 1, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/home/josyula/mujoco-py/mujoco_py/builder.py", line 502, in <module>
    cymj = load_cython_ext(mjpro_path)
  File "/home/josyula/mujoco-py/mujoco_py/builder.py", line 81, in load_cython_ext
    cext_so_path = builder.build()
  File "/home/josyula/mujoco-py/mujoco_py/builder.py", line 238, in build
    built_so_file_path = self._build_impl()
  File "/home/josyula/mujoco-py/mujoco_py/builder.py", line 312, in _build_impl
    join(get_nvidia_lib_dir(), 'libOpenGL.so.0'))
  File "/home/josyula/mujoco-py/mujoco_py/builder.py", line 173, in fix_shared_library
    so_file])
  File "/usr/lib/python3.5/subprocess.py", line 581, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['patchelf', '--add-needed', '/usr/lib/nvidia-384/libOpenGL.so.0', '/home/josyula/mujoco-py/mujoco_py/generated/_pyxbld_LinuxExtensionBuilder/lib.linux-x86_64-3.5/mujoco_py/cymj.cpython-35m-x86_64-linux-gnu.so']' returned non-zero exit status 1
`

abhigenie92, 2017-11-10:
@machinaut @foolyc It looks like there is a bug. 

To get it to work I had to comment out `--add-needed` line 122 in `../lib/python3.6/site-packages/mujoco_py-1.50.1.28-py3.6.egg/mujoco_py/builder.py`

Now I am getting this new error:

```
gonzo% python -c "import mujoco_py"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/nohome/jaan/abhishek/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.28-py3.6.egg/mujoco_py/__init__.py", line 1, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/nohome/jaan/abhishek/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.28-py3.6.egg/mujoco_py/builder.py", line 439, in <module>
    cymj = load_cython_ext(mjpro_path)
  File "/nohome/jaan/abhishek/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.28-py3.6.egg/mujoco_py/builder.py", line 81, in load_cython_ext
    return load_dynamic_ext('cymj', cext_so_path)
  File "/nohome/jaan/abhishek/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.28-py3.6.egg/mujoco_py/builder.py", line 87, in load_dynamic_ext
    return loader.load_module()
ImportError: /nohome/jaan/.mujoco/mjpro150/bin/libglewegl.so: undefined symbol: eglGetProcAddress
```

ryanjulian, 2018-01-29:
I also encountered this on Ubuntu 16.04 with GPU. Fixed by installed patchelf 0.9+ from a PPA:

```shell
sudo add-apt-repository ppa:jamesh/snap-support
sudo apt-get update
sudo apt install patchelf
```

JoyChopra1298, 2018-03-01:
If using anaconda virtual environment, then this one line does the job
`conda install -c anaconda patchelf `

astier, 2018-07-06:
For me it was enough to execute:
> conda install patchelf

emigmo, 2018-10-13:
FOR CONDA-ONLY user
1:  conda install -c menpo osmesa
2: install patchelf from source or conda install patchelf
if install from source, we should update the $PATH variable

JamieShelley, 2019-01-01:
> I also encountered this on Ubuntu 16.04 with GPU. Fixed by installed patchelf 0.9+ from a PPA:
> 
> ```shell
> sudo add-apt-repository ppa:jamesh/snap-support
> sudo apt-get update
> sudo apt install patchelf
> ```

Not all hero's wear capes

gaoyuankidult, 2019-01-13:
My problem was that my version of patchelf was 8.4. It was the default version in Ubuntu 16.04 repo.  You should down 0.9 version and install https://nixos.org/releases/patchelf/patchelf-0.9/patchelf-0.9.tar.gz.   Source: http://zktang.farbox.com/post/install-mujoco-py

MrDadaGuy, 2019-02-28:
I did not need to add the other repository, I was able to get it just using 

`sudo apt install patchelf`

thanks

gautams3, 2019-05-23:
> I did not need to add the other repository, I was able to get it just using
> 
> `sudo apt install patchelf`
> 
> thanks

This worked for me on Ubuntu 18.04 with a GPU. FYI it installed 0.9-1, not the latest 0.10
> patchelf/bionic,now 0.9-1 amd64

wang88256187, 2019-09-11:
> > I did not need to add the other repository, I was able to get it just using
> > `sudo apt install patchelf`
> > thanks
> 
> This worked for me on Ubuntu 18.04 with a GPU. FYI it installed 0.9-1, not the latest 0.10
> 
> > patchelf/bionic,now 0.9-1 amd64

thanks，are also interest in RL? Can I be friends with you? so that we can talk about together and  move on ?

alifahsanul, 2019-10-01:
this line do the job
`sudo apt install patchelf`

SimeonOA, 2021-11-09:
> 
> 
> I did not need to add the other repository, I was able to get it just using
> 
> `sudo apt install patchelf`
> 
> thanks

This worked for me. Thank you.

yulonglin, 2021-12-18:
Similar to the conda install solutions, I was able to solve this using pip in my conda environment:
`pip install patchelf`