openai/mujoco-py #148: Support for dynamics modding and randomization (DynamicsModder)

matwilso, 2017-10-23:
In [*Sim-to-Real Transfer of Robotic Control with Dynamics Randomization*](https://arxiv.org/pdf/1710.06537.pdf), they randomize model properties of the Mujoco simulation like mass of links and damping of joints.

Is there any planned support for this dynamics modding and randomization as there is for texture modding ([TextureModder](https://github.com/openai/mujoco-py/blob/master/mujoco_py/modder.py#L198-L431))?

I am guessing dynamics modding is not as straightforward, due to how Mujoco handles model loading.

It looks like the [current workaround](https://github.com/openai/gym/issues/234) is altering an xml file and reloading the model from that.  If this is the best way, would it make sense to add a nicer API for it?

machinaut, 2017-11-02:
Closing this out for now because there doesn't seem to be anything else concretely actionable for it right now.

closed 2017-11-02 by machinaut

HareshMiriyala, 2019-02-20:
Is the community looking at a way to change parameters online ?