openai/mujoco-py #149: Task space control

arpit15, 2017-10-25:
I am trying to control the baxter in task space using by connecting a mocap object to my end -effector and then setting the mocap model.data.mocap_pos and model.data.mocap_quat. 
However the robot ends at very different pose then commanded. 
('ee position', array([ 0.69456633, -0.45786777, -0.03939242]), array([ 0.25345693,  0.71674097, -0.31498301,  0.56817926]))
('box position', array([  6.00000009e-01,  -1.18464247e-08,   1.00001152e-03]), array([  1.00000000e+00,   1.91855285e-07,  -2.27786234e-10,
         8.46836952e-09]))
[This ](https://drive.google.com/file/d/0B62cBQ948cfVQXFpLV9oXzU0SXM/view?usp=sharing) is the model file for baxter. [This](https://drive.google.com/file/d/0B62cBQ948cfVWGdHTjdzem5sNnM/view?usp=sharing) zip contains everything together with model files.
Any suggestions how to move the robot to a given end-effector pose.

machinaut, 2017-10-26:
In my experience some common reasons for a mocap welded body not achieving mocap position are:
- out of range (e.g. beyond the reach of the robot arm)
- self collision (e.g. robot is getting in the way of itself -- turn on contact visualization to see these)
- path is blocked (there is no path planning for mocap welded body)

If you change the shape of the mocap body to the shape of the welded body, and make them both transparent, do they overlap exactly?

arpit15, 2017-10-26:
I am commanding the end effector to go to the box in front of it. I am able to find a valid IK solution using orocos KDL library. The following is the image where I have contact switched on and transparent body. I am not able to check if the welded objects exactly overlap. Even though I changed the geom for both to be the same. 
[image]

In the picture it seems that the 2 left joints angles are in collision because of a geom body. However I am not sure of why. 
Regarding your points, I was wondering if there is way around path blocked. However for the simple case like I have in which I command it to go to a very comfortable goal position, I was wondering if there is a simpler solution.

machinaut, 2017-10-27:
What's the shape of the contact geometry?

In many robots in mujoco, high-quality meshes are used for visualization, but simple objects are used to model contacts.

Alternatively, if meshes are used for contact geometry, MuJoCo will convexify them.  This could possibly lead to your geoms being much larger than you think, and intersecting each other.

In any case this seems to be more of a mujoco simulation thing than a python binding thing, I'd suggest also asking over in the MuJoCo forums http://mujoco.org/forum/index.php

arpit15, 2017-10-27:
In my baxter model, in all the bodies I have 2 geom- mesh and a cylinder. In the convex hull, I don't see any enlargement of parts which might create problems with collision. 
[image]. I am trying to debug the contacts. Kindly let me have any other suggestions. Can you share an example script and xml where you can use mocap to move the robot end-effector to a given pose. 
I have posted an issue on the [mujoco forum](http://mujoco.org/forum/index.php?threads/commanding-to-a-joint-angle-configuration.3520/) as well.

machinaut, 2017-10-30:
closing this here since discussion moved to mujoco forums

closed 2017-10-30 by machinaut