openai/mujoco-py #152: Getting render from specific camera in mujoco 0.5.7

silverjoda, 2017-10-31:
Hi, I tried asking this question on the mujoco forum and gome help, however I didn't resolve my issue. 

How can I render a specific camera (not the default GUI camera) that I have attached to a body in my xml file? In mujoco 1.5 you can pass a camera name into the render method of the Mjsim class but in mujoco 0.5.7 there is none of that. The viewer is attached to the default camera and I don't know how to change it.

I can't run mujoco 1.5 on my system (Cannot initialize GLEW error which I can't find a fix for).

Thanks in advance for the help.

machinaut, 2017-11-02:
Could you file an issue for the 1.5 GLEW initialization error?  That's something we are still developing.

We're not using 0.5.7 much, so it doesn't get updated anymore.

silverjoda, 2017-11-02:
I'm not sure what you mean by filing an issue for it since it's already been filed in #44, as you mentioned.