openai/mujoco-py #157: Failure to pip install due to various errors

avaziri, 2017-11-05:
**Summary:**
Even after installing all the dependencies in the dockerfile, I was unable to successfully pip install mujoco-py. See the detailed description for the errors and attempted workarounds. I hope this might be of use to people with similar errors.

**System Info:**
```
$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 16.10
Release:        16.10
Codename:       yakkety

```
```
$ gcc --version
gcc (Ubuntu 6.2.0-5ubuntu12) 6.2.0 20161005
```

**Detailed Description:**
I was trying to install mujoco-py I first got the following error about a libffi file missing. :
[failure_to_install_error.txt](https://github.com/openai/mujoco-py/files/1443675/failure_to_install_error.txt)
I went to the dockerfile and installed all the packages listed in the first `apt-get upgrade` but continued to get the same error.
I already had libffi6 on my machine, but I also needed:
`$ apt-get install libffi-dev`

That let me get a bit further, where I got the following error about "mjmodel.h" missing:
[failure_to_install_error_1.txt](https://github.com/openai/mujoco-py/files/1443676/failure_to_install_error_1.txt)
However that file does exist on my system:
```
$ find ~/.mujoco/ -name 'mjmodel.h'
~/.mujoco/mjpro150/include/mjmodel.h
```
I has been using `sudo -H pip3 install -U 'mujoco-py<1.50.2,>=1.50.1'` which differed from the directions. I tried installing without `sudo -H` which gave me the very common error related to GLFW3.
[failure_to_install_error_2.txt](https://github.com/openai/mujoco-py/files/1443694/failure_to_install_error_2.txt)
I adjusted my LD_LIBRARY_PATH and tried again with `LD_LIBRARY_PATH=$HOME/.mujoco/mjpro150/bin pip3 install -U 'mujoco-py<1.50.2,>=1.50.1'`
Now it has spent more than 20 minutes on:
```$ LD_LIBRARY_PATH=$HOME/.mujoco/mjpro150/bin pip3 install -U 'mujoco-py<1.50.2,>=1.50.1'
Collecting mujoco-py<1.50.2,>=1.50.1
  Using cached mujoco-py-1.50.1.26.tar.gz
Requirement already up-to-date: glfw>=1.4.0 in ./miniconda3/envs/robosumo/lib/python3.5/site-packages (from mujoco-py<1.50.2,>=1.50.1)
Requirement already up-to-date: numpy>=1.11 in ./miniconda3/envs/robosumo/lib/python3.5/site-packages (from mujoco-py<1.50.2,>=1.50.1)
Requirement already up-to-date: Cython>=0.25.2 in ./miniconda3/envs/robosumo/lib/python3.5/site-packages (from mujoco-py<1.50.2,>=1.50.1)
Requirement already up-to-date: imageio>=2.1.2 in ./miniconda3/envs/robosumo/lib/python3.5/site-packages (from mujoco-py<1.50.2,>=1.50.1)
Requirement already up-to-date: cffi>=1.10 in ./miniconda3/envs/robosumo/lib/python3.5/site-packages (from mujoco-py<1.50.2,>=1.50.1)
Requirement already up-to-date: pillow in ./miniconda3/envs/robosumo/lib/python3.5/site-packages (from imageio>=2.1.2->mujoco-py<1.50.2,>=1.50.1)
Requirement already up-to-date: pycparser in ./miniconda3/envs/robosumo/lib/python3.5/site-packages (from cffi>=1.10->mujoco-py<1.50.2,>=1.50.1)
Requirement already up-to-date: olefile in ./miniconda3/envs/robosumo/lib/python3.5/site-packages (from pillow->imageio>=2.1.2->mujoco-py<1.50.2,>=1.50.1)
Building wheels for collected packages: mujoco-py
  Running setup.py bdist_wheel for mujoco-py ... -
```
While waiting for more than 20 minutes I read around some other relevant bug reports. In particular I seem to have [this](https://github.com/openai/mujoco-py/issues/66) issue. As described in that thread I had to press enter to get the rest of the failure log:
[failure_to_install_error_3.txt](https://github.com/openai/mujoco-py/files/1443701/failure_to_install_error_3.txt)
This error log contains strange ascii characters and a spurious error about an invalid license. I do have a license.
```
$ find ~/ -name "mjkey*"
/home/andrew/.mujoco/mjkey.txt

```
I am using a conda environment, which seems to be the common thread among people who get this error. [This](https://github.com/openai/mujoco-py/issues/66)  thread suggests cloning the repo and installing from it directly. I will try that now.

avaziri, 2017-11-05:
Installing from repo:
```
git clone git@github.com:openai/mujoco-py.git
python3 setup.py install
python3
import mujoco_py
```
Failure due to missing GLWF3:
```
$ python
Python 3.5.4 |Anaconda, Inc.| (default, Nov  3 2017, 20:01:27) 
[GCC 7.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import mujoco_p
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named 'mujoco_p'
>>> import mujoco_py
Compiling /home/andrew/mujoco-py/mujoco_py/cymj.pyx because it changed.
[1/1] Cythonizing /home/andrew/mujoco-py/mujoco_py/cymj.pyx
running build_ext
building 'mujoco_py.cymj' extension
creating /home/andrew/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder
creating /home/andrew/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5
creating /home/andrew/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/home
creating /home/andrew/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/home/andrew
creating /home/andrew/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/home/andrew/mujoco-py
creating /home/andrew/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/home/andrew/mujoco-py/mujoco_py
creating /home/andrew/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/home/andrew/mujoco-py/mujoco_py/gl
gcc -pthread -B /home/andrew/miniconda3/envs/robosumo/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -Imujoco_py -I/home/andrew/mujoco-py/mujoco_py -I/home/andrew/.mujoco/mjpro150/include -I/home/andrew/miniconda3/envs/robosumo/lib/python3.5/site-packages/numpy/core/include -I/home/andrew/miniconda3/envs/robosumo/include/python3.5m -c /home/andrew/mujoco-py/mujoco_py/cymj.c -o /home/andrew/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/home/andrew/mujoco-py/mujoco_py/cymj.o -fopenmp -w
gcc -pthread -B /home/andrew/miniconda3/envs/robosumo/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -Imujoco_py -I/home/andrew/mujoco-py/mujoco_py -I/home/andrew/.mujoco/mjpro150/include -I/home/andrew/miniconda3/envs/robosumo/lib/python3.5/site-packages/numpy/core/include -I/home/andrew/miniconda3/envs/robosumo/include/python3.5m -c /home/andrew/mujoco-py/mujoco_py/gl/osmesashim.c -o /home/andrew/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/home/andrew/mujoco-py/mujoco_py/gl/osmesashim.o -fopenmp -w
creating /home/andrew/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/lib.linux-x86_64-3.5
creating /home/andrew/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/lib.linux-x86_64-3.5/mujoco_py
gcc -pthread -shared -L/home/andrew/miniconda3/envs/robosumo/lib -B /home/andrew/miniconda3/envs/robosumo/compiler_compat -Wl,-rpath=/home/andrew/miniconda3/envs/robosumo/lib -Wl,--no-as-needed -Wl,--sysroot=/ /home/andrew/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/home/andrew/mujoco-py/mujoco_py/cymj.o /home/andrew/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/temp.linux-x86_64-3.5/home/andrew/mujoco-py/mujoco_py/gl/osmesashim.o -L/home/andrew/.mujoco/mjpro150/bin -L/home/andrew/miniconda3/envs/robosumo/lib -Wl,--enable-new-dtags,-R/home/andrew/.mujoco/mjpro150/bin -lmujoco150 -lglewosmesa -lOSMesa -lGL -lpython3.5m -o /home/andrew/mujoco-py/mujoco_py/generated/_pyxbld_LinuxCPUExtensionBuilder/lib.linux-x86_64-3.5/mujoco_py/cymj.cpython-35m-x86_64-linux-gnu.so -fopenmp
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/andrew/mujoco-py/mujoco_py/__init__.py", line 4, in <module>
    from mujoco_py.mjviewer import MjViewer, MjViewerBasic
  File "/home/andrew/mujoco-py/mujoco_py/mjviewer.py", line 2, in <module>
    import glfw
  File "/home/andrew/miniconda3/envs/robosumo/lib/python3.5/site-packages/glfw.py", line 194, in <module>
    raise ImportError("Failed to load GLFW3 shared library.")
ImportError: Failed to load GLFW3 shared library.
```
It might not be clear from the above error, but with subsequent testing it seems that the Cythonizing actually succeeded, but was immediately followed by the GLFW3 error when the module tried to load. This can be seen because running the same commands give the same error, but without any messages about Cythonizing:
```
$ python3
Python 3.5.4 |Anaconda, Inc.| (default, Nov  3 2017, 20:01:27) 
[GCC 7.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.   
>>> import mujoco_py
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/andrew/mujoco-py/mujoco_py/__init__.py", line 4, in <module>
    from mujoco_py.mjviewer import MjViewer, MjViewerBasic
  File "/home/andrew/mujoco-py/mujoco_py/mjviewer.py", line 2, in <module>
    import glfw
  File "/home/andrew/miniconda3/envs/robosumo/lib/python3.5/site-packages/glfw.py", line 194, in <module>
    raise ImportError("Failed to load GLFW3 shared library.")
ImportError: Failed to load GLFW3 shared library.
```
Whereas running  `LD_LIBRARY_PATH=$HOME/.mujoco/mjpro150/bin python3` works, again without any messages about cythonizing.
```
$ LD_LIBRARY_PATH=$HOME/.mujoco/mjpro150/bin python3
[... cut ...]
>>> import mujoco_py
>>> from os.path import dirname
>>> model = mujoco_py.load_model_from_path(dirname(dirname(mujoco_py.__file__))  +"/xmls/claw.xml")
>>> sim = mujoco_py.MjSim(model)
>>> print(sim.data.qpos)
[ 0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.]
>>> sim.step()
>>> print(sim.data.qpos)
[  2.09217903e-06  -1.82329050e-12  -1.16711384e-07  -4.69613872e-11
  -1.43931860e-05   4.73350204e-10  -3.23749942e-05  -1.19854057e-13
  -2.39251380e-08  -4.46750545e-07   1.78771599e-09  -1.04232280e-08]
```

avaziri, 2017-11-05:
If someone wants to look into these errors, feel free to ask me to follow up. However, since there are workarounds feel free to close the ticket.

mt-, 2018-02-28:
Just a note that I ran into the above issue with a non-conda/virtualenv on ubuntu 16.04.  The failure_to_install_error_3.txt issue appears to be a blocker in terms of pip install.  Installing from checked out source resolves the issue for myself also.