openai/mujoco-py #158: pip install returns invalid activation key error when user uses conda env or virtualenv

avaziri, 2017-11-05:
**Summary:**
Many users have reported that they get an error message with strange ascii characters and an invalid activation key warning when using 

```
���[��� �[V: not found
sh: 2: [
: not found
���[��� �[V: not found
sh: 2: [
: not found
���[��� �[V: not found
sh: 2: Syntax error: "(" unexpected
���[��� �[V: not found
sh: 2: Syntax error: "(" unexpected
ERROR: Invalid activation key
```

avaziri, 2017-11-05:
On second thought it might be a bad idea to suggest this method. Some users report that installing in this way prevents interoperability with [gym](https://github.com/openai/mujoco-py/issues/66#issuecomment-327960438), or [causes segfaults. ](https://github.com/openai/mujoco-py/issues/47#issuecomment-312556635)

vastsoun, 2017-12-26:
Hi,

I too am having this issue. What is the recommended way to install then? I also want to use this with gym.

makdoudN, 2018-01-19:
Hi,

Is there any solution to this problem ?

gntoni, 2018-01-30:
@avaziri 

> Some users report that installing in this way prevents interoperability with gym, or causes segfaults. 

Those interoperability issues were before the last version of gym was released, now it should work with mujoco 1.50  (If we could get past this ASCII error )

danijar, 2018-04-06:
Install the `mujoco-py` package from source:

```
github clone https://github.com/openai/mujoco-py.git
cd mujoco-py
pip3 install -e .
```

Help installer find MuJoCo and OpenGL dependencies:

```
export PATH=$PATH:~/.mujoco/mjpro150
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:~/.mujoco/mjpro150/bin
locate libGL  # To find the right path.
sudo ln -s <path>/libGL.so.1 /usr/lib/libGL.so
```

Compile bindings by importing the package once:

```
python3 -c 'import mujoco_py'
```

youkaichao, 2020-03-22:
I met the same problem.
Using `pip install https://github.com/openai/mujoco-py/archive/1.50.1.1.zip` fails.
But if I download the zip file, and then use `python setup.py install`, it works.

GPrathap, 2020-06-23:
I used what @youkaichao  suggested and then got a permission [this](https://github.com/openai/mujoco-py/issues/291 ) issue. To avoid this issue, I had to give permission  to  /usr/local/lib/python3.5/dist-packages/mujoco_py directory (`sudo chmod 777 -R /usr/local/lib/python3.5/dist-packages/mujoco_py/*` ) , after that its work fine.

Schopenhauer-loves-Hegel, 2023-08-13:
pip install cython==0.29.36
try this