openai/mujoco-py #162: ERROR: Could not open activation key file mjkey.txt

abhigenie92, 2017-11-07:
```
~/.m/m/bin » ./simulate ../model/humanoid.xml                             (ppo_latest) 20:45:08
MuJoCo Pro library version 1.50
ERROR: Could not open activation key file mjkey.txt

Press Enter to exit ...

~/.m/m/bin » pwd                                                          (ppo_latest) 20:45:14
/Users/abhishek/.mujoco/mjpro150/bin

~/.m/m/bin » ls ../../                                                    (ppo_latest) 20:45:16
__MACOSX         mjkey.txt        mjpro150         mjpro150_osx.zip

```

Johnzdh, 2018-01-30:
I also had this problem. I thought there must be something wrong with the README.md. I solved this problem by placing my license key at `~/.mujoco/mjpro150/bin/mjkey.txt` instead of `~/.mujoco/mjkey.txt`.
Hope that can help you.

sudoankit, 2018-03-08:
Here's what worked for me: (note this is just to run `./simulate`, for your custom files/code you'll have to follow a similar way )

1. Open `simulate.cpp` and search for `mj_activate( )` function. ( `line number 1217` ).
2. Pass the full path of your `mjkey.txt`. For me it was, `mj_activate("/Users/sudoankit/.mujoco/mjpro150/sample/mjkey.txt");` 
3. Once done, `make` in the same directory. 
4. Now try running `./simulate`, which is in your `mujoco/bin` directory.
5. A GUI window should open, drag any model from the `model` directory to the window.

emmazfj, 2019-04-19:
> ```
> ~/.m/m/bin » ./simulate ../model/humanoid.xml                             (ppo_latest) 20:45:08
> MuJoCo Pro library version 1.50
> ERROR: Could not open activation key file mjkey.txt
>
> Press Enter [...]

problem solved