openai/mujoco-py #166: Fail to set the pos of a boy dynamically

foolyc, 2017-11-09:
Any one knows how to set the position of a boy simulation ?
I have tried `model.body_pos[i] = ` and `sim.data.body_xpos[i] = `, but it does not work.

iretiayo, 2017-12-14:
Is it possible to do this with the old version of mujocopy?
The env.model.body_pos variable is read only; is there a way to manually set the position of a body?

In essence, I want to dynamically set the location of a body that will be fixed once set. If I put extra slide joints, I loose the ability to make the body fixed.

foolyc, 2017-12-19:
@iretiayo  You can define your body with no freejoint, then you can dynamically set the position of the body (``model.body_pos[i]``)and it is fixed to the worldbody. This works in mujoco150 + mujoco_py(master)

foolyc, 2017-12-19:
@iretiayo setting extra slide joint solves my problem. Thank you!