openai/mujoco-py #168: XML parse error at line 0, column 0:\nFailed to open file\n

abhigenie92, 2017-11-12:
```
$ python
Python 3.6.1 |Anaconda custom (64-bit)| (default, May 11 2017, 13:09:58)
[GCC 4.4.7 20120313 (Red Hat 4.4.7-1)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import mujoco_py
>>> from os.path import dirname
>>> model = mujoco_py.load_model_from_path(dirname(dirname(mujoco_py.__file__))  +"/xmls/claw.xml")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "mujoco_py/cymj.pyx", line 132, in mujoco_py.cymj.load_model_from_path
Exception: Failed to load XML file: /nohome/jaan/abhishek/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.31-py3.6.egg/xmls/claw.xml. mj_loadXML error: b'XML parse error at line 0, column 0:\nFailed to open file\n'
```

Mujoco_py installed correctly and imports fine. I get the above error.

JoyChopra1298, 2018-01-23:
I was getting the same error. I wrote full path of my xml file in model line and it worked.
model = load_model_from_path("/home/joy/.mujoco/mjpro150/mujoco-py-1.50.1.0/xmls/fetch/main.xml")

Hansry, 2018-07-17:
I was getting the same error too!

GbengaOdesanmi, 2018-10-25:
orflow) gbenga@gbenga-Lenovo:~/Downloads/mujoco-py-master/examples$ python3 disco_fetch.py
Traceback (most recent call last):
  File "disco_fetch.py", line 9, in <module>
    model = load_model_from_path("xmls/fetch/main.xml")
  File "cymj.pyx", line 131, in mujoco_py.cymj.load_model_from_path
Exception: Failed to load XML file: xmls/fetch/main.xml. mj_loadXML error: b'XML parse error at line 0, column 0:\nFailed to open file\n'

machinaut, 2018-10-25:
@abhigenie92 if you update to latest mujoco-py, xmls are now distributed with the package.

Ameyapores, 2018-11-02:
@GbengaOdesanmi  I am getting the exact error, did you manage to solve it?

machinaut, 2018-11-02:
The XMLs are included with the python package, but don't seem to be getting installed to that location.

Does it work if you download the files or checkout the repo and load the model from there?

machinaut, 2018-11-02:
I just updated the readme to use a humanoid xml as an example instead.

https://github.com/openai/mujoco-py#install-and-use-mujoco-py

GbengaOdesanmi, 2018-11-03:
@Ameyapores then it was path problem, and i was able to resolve it, yesterday, I also had the same problem. try to check the updated readme. thanks @machinaut

cyrilzakka, 2018-11-19:
I had a similar error earlier today - as it turns out the message spewed by the compiler isn't very descriptive of the underlying error. In my case, the XML file I was trying to load had other XML dependencies it couldn't access or find. Fixed it by moving around some files. Hope this helps someone out!

66Lau, 2024-01-19:
> I was getting the same error. I wrote full path of my xml file in model line and it worked. model = load_model_from_path("/home/joy/.mujoco/mjpro150/mujoco-py-1.50.1.0/xmls/fetch/main.xml")

or using `load_model_from_path("../xmls/fetch/main.xml")` works